arendst/Tasmota #8464: Define Template in "user_config_override.h"

Cerothen, 2020-05-17:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
Currently there is no way to specify a customized PIN/module configuration from the config override file. 

This situation has been requested multiple times in the past though not specifically using templating as the solution:
https://github.com/arendst/Tasmota/issues/5068
https://github.com/arendst/Tasmota/issues/3747
https://github.com/arendst/Tasmota/issues/3258
**Describe the solution you'd like**  
If the ability to define the default template was added to this file it would be a single option to configure:
```
#undef TEMPLATESTR
#define TEMPLATESTR ""
{\"NAME\":\"Generic\",\"GPIO\":[255,255,255,255,255,255,255,255,255,255,255,255,255],\"FLAG\":15,\"BASE\":18}
```
Then individuals compiling the firmware could set the module to 255 to enable that template as the default when initially flashing or when resetting the configuration.
```
#undef MODULE
#define MODULE 255
```
**Describe alternatives you've considered**  
I have reviewed the suggestions in similar requests and the proposed solutions such as "creating something separate to automatically set the settings via serial or otherwise; Setting up a device then backup, restore to each device; Flash one device, set it up and read the flash back to then write to other devices as the firmware. None of these are as elegant as specifying a template in the user_config_override file and setting the template as the active module.

**Additional context**  
The main reason for this is to create firmware where the device is fully configured for that use case to offset issues with an unexpected reset (either by someone holding the button for too long [kids...], or if the device was to spontaneously reset (as was indicated in one of those linked issues)

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2020-05-18:
I know it's not exactly what you requested but may be [decode-config](https://github.com/tasmota/decode-config) would help?

Create a JSON text file e.g. `tempate.json` only containing the related settings:
```
{
  "user_template": {
    "base": 18, 
    "flag": {
      "adc0": 15
    }, 
    "gpio": [255,255,255,255,255,255,255,255,255,255,255,255,255], 
    "name": "mytemplate"
  }, 
  "last_module": 255
}
```

then load this config part using
`decode-config -d <yourdevice> -i template.json`
into each device you want.

Cerothen, 2020-05-18:
Not a bad solution however it would be less effort to edit the template and check off "activate" within the "Configure Other" menu when setting the display name. The main goal is for the device to function in its normal operating state should it get accidentally reset or spontaneously reset. I had noticed that it had been requested a few times but I think those were all before templates existed. If its a lot of work then it certainly isn't a necessity it would just really give people that are compiling firmware more options for firmware customization. Especially considering the number of devices on the blackadder site.

arendst, 2020-06-17:
Will have a look...

arendst, 2020-06-17:
> The main goal is for the device to function in its normal operating state should it get accidentally reset or spontaneously reset. 

Indeed currently it will be reset to a Sonoff Basic on multiple power drops and/or restarts.

As implementing a fixed template in user_config_override.h will take some additional code I think it's more realistic to make the auto fallback to Sonoff Basic optional. I think I'll implement an option to switch between either Sonoff Basic (Module 1) OR the active template (Module 0). This way no fixed template needs to be compiled while there is a solution for the case where the Sonoff Basic module won't work as fallback module (like an ESP32 ethernet device ;-)

Cerothen, 2020-06-17:
That would be good, For devices that don't have the same pinout of the Sonoff Basic it would allow the user to still be able to interact with the button to get the multiple click actions

meingraham, 2020-06-17:
For the choice between module (1 - basic) or the active template, would module 18 - Generic be the safest fallback?

arendst, 2020-06-17:
Good question. For now the default is SONOFF_BASIC for ESP8266 and WEMOS for ESP32. It can be changed in my_user_config.h with define FALLBACK_MODULE

closed 2020-06-19 by ascillato2

ramanraja, 2021-04-05:
I have a custom hardware module for which I configure a custom template derived from Generic 18. 
I enter the template in user_config_override.h and compile. It works well.
However, *after* I have upload the firmware, I have to activate the template manually either by checking the 'activate' box on the web interface, or issue the command 'module 0' through serial port.
Is there a way to avoid this extra step ? I just want to define the template in source code and simultaneously activate it, all at compile time.

adelafug, 2021-12-18:
@ramanraja you solves problem for activate user template in reset?

barbudor, 2021-12-18:
#define MODULE USER_MODULE

adelafug, 2021-12-18:
Thanks, works fine!!

rklasen, 2022-02-22:
I like that. So I can define the template at compile time and have it be activated automatically after I uploaded the firmware? Just to be sure, would adding this to `user_config_override.h` work?

```
#undef USER_TEMPLATE
#define USER_TEMPLATE "{\"NAME\":\"ESP01-Dimmer\",\"GPIO\":[1,1,416,1,1,1,1,1,1,1,1,1,1,1],\"FLAG\":0,\"BASE\":18}"
```

rklasen, 2022-02-22:
Thank you for the help and the quick reply!

Server4hassio, 2023-02-08:
> 

Hello its working on my 8266 things. 
but not working zigbee bridge (unofficial). After reset 1 command. its selecting {"Module":{"1":"ESP32-DevKit"}} and module2  {"Module2":{"1":"ESP32-DevKit"}}.

i added this lines

```
#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

#ifdef MODULE
#undef MODULE
#endif
#define MODULE                 USER_MODULE   // [Module] Select default model (the list is kModuleNiceList() in file tasmota_template.h) USER_MODULE is the TEMPLATE

#ifdef FALLBACK_MODULE
#undef FALLBACK_MODULE
#endif
#define FALLBACK_MODULE        USER_MODULE   // to Select the default model as FALLBACK when the user does a RESET 1

#ifdef USER_TEMPLATE
#undef USER_TEMPLATE
#endif
#define USER_TEMPLATE          "{\"NAME\":\"Zig_Bridge32\",\"GPIO\":[1,1,3552,1,3584,1,1,1,5793,5792,320,544,5536,1,5600,1,0,1,1,5568,0,1,1,1,0,0,0,0,608,640,32,1,1,0,0,1],\"FLAG\":0,\"BASE\":1}" // [Template] Set JSON template

#undef  MQTT_HOST
#define MQTT_HOST         "****" // [MqttHost]

#undef  MQTT_USER
#define MQTT_USER         "****"         // [MqttUser] Optional user

#define STA_SSID2              "****"                // [Ssid2] Optional alternate AP Wi-Fi SSID
#define STA_PASS2              "****"                // [Password2] Optional alternate AP Wi-Fi password

#define APP_TIMEZONE           *

#define WEB_PASSWORD           "****"                // [WebPassword] Web server Admin mode Password for WEB_USERNAME (empty string = Disable)

#define APP_DISABLE_POWERCYCLE true             // [SetOption65] Disable fast power cycle detection for device reset
#define ZIGBEE_DISTINCT_TOPICS true             // [SetOption89]

#define USE_SONOFF_SPM                           // Add support for ESP32 based Sonoff Smart Stackable Power Meter (+11k code)
#define USE_DISPLAY_TM1621_SONOFF                // Add support for TM1621 dsiplay driver used by Sonoff POWR3xxD and THR3xxD

#undef  SERIAL_LOG_LEVEL
#define SERIAL_LOG_LEVEL LOG_LEVEL_NONE

#define USE_ZIGBEE
#undef  USE_ZIGBEE_ZNP
#define USE_ZIGBEE_EZSP
#define USE_UFILESYS
#define USE_ZIGBEE_EEPROM // T24C512A
#define USE_TCP_BRIDGE
#undef  USE_ZIGBEE_CHANNEL
#define USE_ZIGBEE_CHANNEL 11 // (11-26)

#define USE_ETHERNET
#undef  ETH_TYPE
#define ETH_TYPE 0 // ETH_PHY_LAN8720
#undef  ETH_CLKMODE
#define ETH_CLKMODE 3 // ETH_CLOCK_GPIO17_OUT
#undef  ETH_ADDRESS
#define ETH_ADDRESS 1 // PHY1

#endif  // _USER_CONFIG_OVERRIDE_H_

```
then platformio run -e tasmota32-zbbrdgpro