arendst/Tasmota #8467: Add support for Adafruit STEMMA Soil Sensor using Seesaw code

alfista2600, 2020-05-17:
**Have you looked for this feature in other issues and in the docs?**  
to the best of my limited ability - looking at product schematics and trying to search for related keywords

**Is your feature request related to a problem? Please describe.**  
No

**Describe the solution you'd like**  
Integrate the Adafruit STEMMA Soil Sensor
https://www.adafruit.com/product/4026
https://github.com/adafruit/Adafruit_CircuitPython_seesaw
**Describe alternatives you've considered**  
I am using the sample code running on a Arduino Nano, sending data over serial to a NodeMCU but my implementation is flaky...I'm an amateur...

**Additional context**  
Happy to ship someone a device to test with.

**(Please, remember to close the issue when the problem has been addressed)**  
I will!

Jason2866, 2020-05-17:
There is already support for Moisture sensors. -> CHIRP - Chirp!-sensor and I2C-soil-moisture-sensors

alfista2600, 2020-05-17:
Thanks, Jason, for the reply. Yes, I should have mentioned that I investigated those and have several but I read that they have a lot of timing issues if you try to use more than one. Also, when I compiled a binary with support enabled, it appears to have indicated that a chirp was connected even when it wasn't, when any I2C device was on the bus. Surely I could log a bug, but these other sensors seem to have better code support and are half the cost...

alfista2600, 2020-05-18:
I spent some more time working with my Chirps last night and got the I2C working better, but testing with an official one and an ebay clone I couldn't get either to change darkness or moisture settings. I suspect it is something I'm doing wrong, but couldn't solve in two hours of futzing with different hardware and setting combinations...

arendst, 2020-05-18:
The chirps are a waste of time as they are no real I2C devices. They break the I2C protocol for real I2C devices.

Any decent I2C alternative is much appreciated.

Just see this issue concerning the same https://github.com/arendst/Tasmota/issues/8429#issuecomment-630184188

Jason2866, 2020-05-18:
@arendst  I2C-soil-moisture-sensors is not Chirp. Do the behave as bad as Chirp?

arendst, 2020-05-18:
I don't know. All I (and the owner of the issue 8429) see that the chirp driver breaks the I2C bus simply because the chirp hardware is bad I2C hardware.

Any decent I2C-soil-moisture-sensor not using chirp code is more than welcome.

Jason2866, 2020-05-18:
@alfista2600  Any source where the Adafruit STEMMA Soil Sensor is available? 
All my searches ended in "out of stock"

alfista2600, 2020-05-18:
I'd be happy/honored to mail you and/or Theo one.

Adafruit is in the heart of NYC so have been largely closed down for two months. They are starting to open back up but prioritizing health care supplies... They had a number of projects based on this sensor so I don't think it is likely to be discontinued in the short term, but I'm happy to inquire if that further justifies the effort.

alfista2600, 2020-05-28:
also, ideally, the implementation would support use of devices on all four addresses

alfista2600, 2020-05-28:
at the same time

alfista2600, 2020-06-16:
Sensors are back in stock at Adafruit

https://www.adafruit.com/product/4026

FrankH1, 2020-06-29:
Hi there,
I can report, that they are not compatible with the Chirp-implementation. I managed to compile my own build using the change in "user_config_override.h" as explained here - https://tasmota.github.io/docs/Moisture-Sensor-and-Chirp%21-Sensor/.
Have connected the sensor to an ESP8266-01S flashed with the own build. Both devices are powered with 3.3V. Have changed the configuration to use GPIO0 and GPIO2.
Once booted the ESP sometimes (not always) recognizes the sensor (00:00:00 I2C: CHIRP gefunden bei 0x36). But the sensor readings are wrong (18:55:26 RSL: tele/tasmota/SENSOR = {"Time":"2020-06-29T18:55:26","CHIRP0":{"Moisture":65535,"Temperature":0.0,"Darkness":0},"TempUnit":"C"}). Readings do not change if I put the sensor into water.
So we can just hope that someone has a solution - because the sensors look quite good to me.

alfista2600, 2020-07-25:
I'm still interested in seeing this implemented, Mr. Stale Bot...

alfista2600, 2020-08-09:
Happy to send a sensor to anyone who wants to work on it

wdross, 2020-08-12:
I found this post as I'm looking to see if Tasmota supports this soil sensor I happened to buy from Adafruit.
Looks like there's interest if I can get it running for more than just myself!

FrankH1, 2020-08-12:
Yes, there will be a bunch of happy gardeners and smart homes.... Unfortunately i'm only able to test as my programming skills are not sufficient. Cheers!

wdross, 2020-08-17:
Progress report: I've added a pair of source files from Adafruit's seesaw library, which is enough to communicate with the soil sensor.
 - I was also going to leave in their .github folder and documents.
 - BUT: Their **_library.properties_** is there, and if I leave that there it will be confusing, as I'm keeping only **part** of their library.
 - Should I rewrite their routines to leverage Tasmota's **_support.ino_**'s provided **_I2cValidRead/Write*()_** routines?  Might make for the smallest footprint!

Right now I'm getting both temperature and capacitance readings from the device and they are displaying on the web page.
I have a single soil sensor and have tested it on a couple different addresses, showing the code will search the 4 possible addresses; I don't have a way to test that indeed the code will communicate with 2 or more units.
Have yet to configure any MQTT and see how that comes across.

Had a couple questions:
- I found other devices would report "Moisture", so I chose that here
   - I don't see any other devices reporting Capacitance
   - So the web page reports around 350% for 'dry' and 1015% for 'wet'
   - Is there a different category I should use?  Some kind of "raw"?
- Once I finalize my testing and I'm satisfied, is that when I submit a PR to arendst/Tasmota?
   - If someone has the ability to test multiple devices, would that be done on arendst/Tasmota or my version of /Tasmota?

Thanks -- This is my first significant contribution I'm working towards and want to get as much right the first time as I can!

Jason2866, 2020-08-17:
Thx, for your work on this!
Smallest footprint which fits nice in Tasmota existing code is the most liked way.
For the representation of the values use the one you think it makes the best sense.
For test(s0 use the latest Tasmota development with your added driver.
If everything works you think it is fine submit a PR.
Theo or other contributers will review your code

alfista2600, 2020-08-17:
Excellent!

Personally, I'd prefer a value of 0 to 100% - the raw values never did much for me.

I can deploy up to four sensors for testing multiple addresses. I just need to figure out how to do that (until a PR is accepted, I guess). Still learning coding, GitHub, etc...

alfista2600, 2020-08-25:
@wdross happy to test multiple devices if you can show me how to compile/test (or just send me a bin file)...

lensherm, 2020-08-25:
I think I'd prefer both a 0 to 100 and a raw reading, in case the sensor drifts with age. Maybe even a way to set a  +/- offset, since I'm sure there are variances between the sensors. As an extra, ability to calibrate it by holding the sensor in the air vs. submerging it in water.

lensherm, 2020-08-28:
Just got two sensors, so I can be a QA guinea pig, if needed.

wdross, 2020-08-30:
Making progress, sending MQTT message that my driver creates.
Home Assistant is now showing me there's a device there (darn SetOption19 anyway!!)

Please tell me if this is the correct place, but I need help understanding the format of the JSON items in the MQTT message: Home Assistant is able to recognize the **_Moisture_**, but isn't parsing out the **_Temperature_**.  I'm sure the fault is in my formatting of those two data elements.
And when I "fake out" the driver to think it has 2 sensors, HA doesn't parse anything recognizable.  I've got serial logs in the details below.

```
01:57:45 MQT: homeassistant/sensor/D266C1_seesaw_soil_Moisture/config = {"name":"Tasmota seesaw_soil Moisture","stat_t":"tele/Garduino/SENSOR","avty_t":"tele/Garduino/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D266C1_seesaw_soil_Moisture","dev":{"ids":["D266C1"]},"unit_of_meas":"%","ic":"mdi:cup-water","frc_upd":true,"val_tpl":"{{value_json['seesaw_soil']['Moisture']}}"} (retained)
       :
01:57:48 MQT: tele/Garduino/SENSOR = {"Time":"2020-08-30T01:57:48","seesaw_soil":{"Temperature":77.4},"seesaw_soil":{"Moisture":2},"TempUnit":"F"}
```

Then in my Home Assistant interface I'm given (two screens pasted together; I get the "MQTT Info" associated with the device and scroll down to the Temperature Entity):

[image]

Then, moving on the "two devices" problem, I present the Serial data and a screen shot (two screens together) to show that issue)

```
00:00:08 MQT: homeassistant/sensor/D266C1_seesaw_soil_37_Moistur/config = {"name":"Tasmota seesaw_soil_37 Moisture","stat_t":"tele/Garduino/SENSOR","avty_t":"tele/Garduino/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D266C1_seesaw_soil_37_Moistur","dev":{"ids":["D266C1"]},"unit_of_meas":"%","ic":"mdi:cup-water","frc_upd":true,"val_tpl":"{{value_json['seesaw_soil_37']['Moisture']}}"} (retained)
00:00:08 MQT: homeassistant/sensor/D266C1_seesaw_soil_38_Moistur/config = {"name":"Tasmota seesaw_soil_38 Moisture","stat_t":"tele/Garduino/SENSOR","avty_t":"tele/Garduino/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"D266C1_seesaw_soil_38_Moistur","dev":{"ids":["D266C1"]},"unit_of_meas":"%","ic":"mdi:cup-water","frc_upd":true,"val_tpl":"{{value_json['seesaw_soil_38']['Moisture']}}"} (retained)
       :
00:00:11 MQT: tele/Garduino/SENSOR = {"Time":"1970-01-01T00:00:11","seesaw_soil-37":{"Temperature":77.0},"seesaw_soil-37":{"Moisture":3},"seesaw_soil-38":{"Temperature":94.8},"seesaw_soil-38":{"Moisture":5},"TempUnit":"F"}
```
[image]

My bottom line?
I need help to understand what format change I must make to have an acceptable parsing by Home Assistant.  I'm assuming if I can make it happy then other MQTT "users" should likewise be happy.

BTW: the web interface works just fine as far as I can tell (single sensor presented here):
[image]

lensherm, 2020-08-30:
@wdross 
Hi Wayne,

Thanks for the excellent work! Looks like you're just about there, and since most of us just want the moisture reading anyhow, it already gives a great starting point. Is your code here on github? Wanted to give it a try, but didn't seem to find a fork with recent updates.

On an unrelated topic how is the moisture percentage determined in your implementation? I'm interested if there will be discrepancies between soil types and sensors with these Adafruit sensors like there are on the Chirp ones. Not sure if you saw the Calibration section at the bottom of this page this here [https://github.com/Miceuz/i2c-moisture-sensor](https://github.com/Miceuz/i2c-moisture-sensor)

Maybe this is enough reason to implement a command such as MoistOffset, similar to HumOffset and TempOffset, or to potentially utilize those exact commands for your implementation, and just report moisture as "humidity"? Not sure what the internals of the code look like, but just a thought.
[Commands](https://tasmota.github.io/docs/Commands/)

wdross, 2020-08-30:
Sorry 'bout that.  My work is in my fork, and I put my commits in a branch 'seesaw': [https://github.com/wdross/Tasmota/tree/seesaw](https://github.com/wdross/Tasmota/tree/seesaw)  At the time of this posting I think I've managed to change my default branch to be seesaw, perhaps easier to find.

To be clear: I can get Moisture readings only if there is a single sensor.  I think I need MQTT formatting help to get multiple sensors to properly report into HA.

I took the lowest reading that I got from my capacitance sensor (330) to be a base 'dry', then divided that by the apparent range (which appears to be 10-bits, or 1023 for max 'wet').  So I built a (at this point hard coded) expression of:
`          moisture = (raw - min) / (max - min)`
with protection for underflow

That calibration looks pretty intensive.  And the result over there was non-linear so a simple offset wouldn't be enough to fully characterize it.  But could be simple to implement.  I'll take a look at the offset concept: it's intriguing.

Maybe someone can suggest a correction to my MQTT format such that multiple sensors can be reported (and probably also fixes Temperature!).  Looks like arendst and effelle might have done much of the commits in **_xdrv_12_home_assistant.ino_**

lensherm, 2020-08-31:
Got it. So I poked around and looking at what the designers of the sensors are saying the range is likely 11 bits

> We use the built in capacitive touch measurement system built into the ATSAMD10 chip, which will give you a reading ranging from about 200 (very dry) to 2000 (very wet). 
[Overview](https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor)

Wouldn't be surprised that although the chip itself has these values, you'll never hit the limits in real life, with the sensor attached. Probably worthwhile to try multiple sensors just to see what the discrepancies will look like. I think we have enough folks here to get a half decent sample size.

By the way,  do you happen to have your latest binary available somewhere?

wdross, 2020-08-31:
I've [pushed the binary](https://github.com/wdross/Tasmota/blob/seesaw/.pioenvs/tasmota/firmware.bin) into my fork as well (Tasmota/.pioenvs/tasmota/)

- I think the web page should work with the maximum 4 device addresses that Adafruit supports.
- I think you'll only get valid Moisture data in HA with a single sensor. 
- If you put 2 or more devices then it seems my bad MQTT format will provide NO Moisture data on ANY sensor

alfista2600, 2020-08-31:
I will test on my device tonight!

lensherm, 2020-08-31:
Going to test your bin and another one I compiled on Gitpod from your branch. I have two sensors a D1 Mini and a Node MCU to test on. Will try different combinations and report back how it goes, likely by the end of the day EST.

lensherm, 2020-08-31:
So far I got one sensor working on a NodeMCU with your bin. At least in my case, I'd definitely need to increase the range, and probably the resolution. It's showing 98% when I put the sensor in a plant I haven't watered in a week and know by touch could use a bit of watering.

lensherm, 2020-08-31:
Got a second one on a D1 Mini with pretty same results. This thing is definitely in need of some water, but 
[image]

[image]

wdross, 2020-09-01:
I've updated my source code ([and binary](https://github.com/wdross/Tasmota/blob/seesaw/.pioenvs/tasmota/firmware.bin)) to also post the raw capacitance on the web page as "Analog0".
This will allow anyone to see the raw, unaltered value coming from the sensor itself over the I2C interface.
It's apparent there's still work to be done to properly convert that capacitance into a moisture level.

lensherm, 2020-09-01:
Thanks @wdross. Updating firmware on both now. Will report my findings closer toward late afternoon EST.

lensherm, 2020-09-01:
On an unrelated note, since @wdross is now an i2c expert, getting support for this [little guy](https://circuitpython.readthedocs.io/projects/mcp9600/en/latest/) would be cool for grilling/smoking purposes.

lensherm, 2020-09-01:
Still haven't watered the poor Money Tree for the sake of science and thought enough was enough

Before watering:

[image]

And finally right after plenty of watering:

[image]

A bit anticlimactic. If this is truly the raw reading, maybe these sensors aren't sensitive enough, or don't have the required range for what we need them for.

I will try a few more plants for this experiment. Maybe something that lives in dryer soil, like a cactus will provide more insight.

**One important piece of advice. Try to keep things above the water line on the sensor dry.** Numbers went really funny on me, until I dried things up.

wdross, 2020-09-01:
Well that is disappointing.  I haven't even put mine in the dirt yet -- pretty silly!

Ont the alternate topic, I recently purchased this little ["cousin" of his](https://www.adafruit.com/product/1782) and Tasmota already supports him!  It was a matter of configuration of the I2C lines (just like is needed for the seesaw) and I could get readings from one of them out of the stock build!  The driver (and hardware) supports up to 8 different addresses off one unit Arduino, so you could daisy-chain them together.
Just use the [list of supported Tasmota I2C devices ](https://tasmota.github.io/docs/I2CDevices/)as to what's "allowed" on your shopping list!

lensherm, 2020-09-01:
Based on the docs, it looks like like the "cousin" doesn't support temperatures above 257 F, which is too low for getting temps of the grill itself.

alfista2600, 2020-09-01:
haven't had a chance to test yet (life has been crazy) but one thing based on looking at these screen shots - please make sure the code supports SetOption8 to report back in °F for those of us who don't speak Celsius :-(

lensherm, 2020-09-02:
@wdross , color me utterly confused. The first one is in a cup with water, and the second is in a potted plant, which is slightly moist. 

[image]

lensherm, 2020-09-09:
Hi @wdross 

I played around some more and even compiled and ran the example Arduino sketch for comparison with your Tasmota version. The raw readings are consistent with each other. 

If I ignore the readings I've seen by having the sensor in a glass of water (probably not made for this anyhow), my observations for the reported soil moisture limits are similar to yours. The raw readings I'm getting are around 328-ish for open indoor air with the indoor air being 70F and 50% humidity. At the upper end I'm seeing exactly 1016, in well-watered and tightly-packed soil.

It could very well be that the plants I had in above experiments were over-watered, even though I haven't watered them in a week. I'd say it is worthwhile continuing on with the work, if you still have the time.

@arendst  
 
Do you mind please taking a peek at Wayne's code to see if you can spot where the issue might be lurking in his JSON formatting, for temperature that he's trying to pass over MQTT? I would really appreciate it, as I'm sure would other owners of this sensor.

Thanks!

alfista2600, 2020-09-20:
Plants are thirsty!

alfista2600, 2020-09-28:
Any update? We seem so close, and getting ready to winterize some plants!

wdross, 2020-10-05:
Been a little busy of late - too much overtime at work (need to keep the employer happy!)
It'll be a bit before I get back to it, but my plan is to look at data coming out of other multiple sensor solutions (I've got several of those temperature sensors) and see if that helps me figure out what I've got wrong with my MQTT format.

In northern Michigan my growing season is well in the rear-view mirror.

Separately (and completely unrelated to this topic):

- I've got a rechargeable battery for my project
- Have the deep-sleep configuration figured out within Tasmota
- BUT I've figured out the HUZZAH won't connect to my WiFi from my garden, so I'll have to figure out a better antenna

alfista2600, 2020-11-07:
Still interested, StaleBot...

wdross, 2020-12-05:
Sorry it's taken me so long to make progress on this StaleBot, but I intend to have some time in the next month to revisit the proper formatting of multi-device MQTT messages.

FrankH1, 2020-12-05:
Looking forward, hope you manage it. Cheers

alfista2600, 2020-12-05:
"You can't rush perfection." 

Thanks for the efforts!

lensherm, 2020-12-18:
Fingers crossed

closed 2020-12-27 by ascillato2