arendst/Tasmota #8472: PowerCal Command Error

funboxbolzer, 2020-05-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Wenn ich PowerCal in der Konsole eintrage kommt...
`04:52:19 CMD: PowerCal
04:52:19 MQT: stat/tasmota_031FBB/RESULT = {"Command":"Error"}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
`04:52:19 CMD: PowerCal
04:52:19 MQT: stat/tasmota_031FBB/RESULT = {"Command":"Error"}`

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-05-18:
Without the information requested there is no way to know if you have the device configured properly as a power monitoring device or not.

closed 2020-05-18 by meingraham

funboxbolzer, 2020-05-18:
Hi and sorry... 
i flashed it like this:
[https://www.tasmota.info/gosund-sp112-flash/](url)

and the device setting is...
`{"NAME":"SHP5","GPIO":[57,145,56,146,0,22,0,0,0,0,21,0,17],"FLAG":0,"BASE":18}`

If i use the device "Gosund SP1 v23 Modul" the Console Command PowerCal is working. But then i can´t use the USB Ports from my device

meingraham, 2020-05-18:
Please start over and, in the new issue, complete the template using the commands as requested. Even with the device template you provided, there are still other settings that are needed in order to know whether your device is properly configured.

If you shortcut the process of completing the issue template, then the assistance you can get will also be "shortcutted".

ascillato, 2020-05-18:
Please, search in issues. This has been already addressed.

The correct command is POWERSET

Thanks.