arendst/Tasmota #8478: Beware of new MCU hardware found on Digoo NX-SP202 plugs

Lefuneste83, 2020-05-18:
### PROBLEM DESCRIPTION
For the benefit of the community. I recently purchased a Digoo NX-SP202 on aliexpress here
https://www.aliexpress.com/item/32952717922.html
It seems to be the original manufacturer store.

After opening the device I realized that the embedded MCU on this board is not documented in various posts about this device. It is a WR5E Module which is supported by Tuya ecosystem but may not be ESP8266 compatible. The MCU board has 7+8 pins instead of 6+6 configuration widely documented on previous generations of this device.

Documentation can be found here.
https://docs.tuya.com/zh/iot/device-development/module/wifi-module/wr5e

I tried to connect to the UART pins GPIO 18 (RX) and 23 (TX), 3V3 and GND.
Device powers up and is detected by host computer.

Upload from Visual Studio fails as I am unable to change the boot mode of the device, no actions seem to affect it at startup. Tried to pull up and pull down GPIO 0, press switches, nothing worked.

Tried Tuya convert. Once the serial connection is removed (desoldered) but just 3V3 and GND are used, I managed to enter pairing mode (blue LED blinking) by pressing 6 seconds any of both buttons. Note that this does not occur when serial connection is physically attached.

I see that there are no similar case documented with very little experience for this WR5E module.

### REQUESTED INFORMATION
For those facing similar issue I'd be glad to help run some tests. The MCU seems working fine and is documented but may not be Tasmota compatible. 

Can someone confirm this or has similar Tuya style MCU been successfully flashed ?

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

digiblur, 2020-05-18:
RTL based chipset - https://fccid.io/2ANDL-WR5E/Internal-Photos/Internal-Photos-4314008

blakadder, 2020-05-18:
Please address that to https://templates.blakadder.com/new-unsupported.html. This issue will be closed and will not be visible directly

closed 2020-05-18 by blakadder

Lefuneste83, 2020-05-18:
Yes indeed it is a RTL8710BX chip... Bad luck on this one...