arendst/Tasmota #8482: Integration of a meter for recording Wh or kWh

Teckware, 2020-05-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

No, no problem. Just a feature that would be nice. The counters for power monitoring already show the values: current, power and voltage.
These could be used to build a counter that shows the power / time in kWh in a predefined time interval.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-05-19:
Tasmota is designed to be used in conjunction with other home automation software which can easily do what you request. See https://tasmota.github.io/docs/Integrations/

Teckware, 2020-05-19:
I have not yet managed to migrate the data from an INA219. I have converted this so that it can measure 32A of my solar system. I have the possibility to do this via the integration of NodeRed but I don't have the necessary programming skills.

Jason2866, 2020-05-19:
Sorry, this will probably not happen, since this would need to many flash writes.
A flash chip is not designed for this use case.

closed 2020-05-19 by Teckware