arendst/Tasmota #8483: Device Name has a minor issue

wongnam, 2020-05-19:
### PROBLEM DESCRIPTION
Device Name has a minor issue that the Friendly Name info is displayed in Device Name description area.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S22
- [x] Tasmota binary firmware version number used: ver 8.3.1 - 2.7.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSC + PlatformIO IDE
- [x] Flashing tools used: device web UI
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
 16:51:57 MQT: stat/smartplug/RESULT = {"NAME":"Sonoff S22","GPIO":[17,255,0,255,255,0,0,0,21,56,255,0,0],"FLAG":0,"BASE":4}
16:51:57 MQT: stat/smartplug/RESULT = {"Module":{"0":"Sonoff S22"}}
16:51:58 MQT: stat/smartplug/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"2":"AM2301"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
16:53:17 MQT: stat/smartplug/STATUS = {"Status":{"Module":0,"DeviceName":"Device Name 1","FriendlyName":["Smart Plug 1"],"Topic":"smartplug","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:53:17 MQT: stat/smartplug/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"my_smarthome","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:36","StartupUTC":"2020-05-19T09:51:41","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-05-19T16:21:16","SaveCount":23,"SaveAddress":"F5000"}}
16:53:17 MQT: stat/smartplug/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-19T16:16:05","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"421/699"}}
16:53:17 MQT: stat/smartplug/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-2","Wong"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
16:53:17 MQT: stat/smartplug/STATUS4 = {"StatusMEM":{"ProgramSize":499,"Free":504,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","8F08A793","04100001","000000C4","000000C0","00000001","00000000"],"Drivers":"1,2,3,4,9,10,20,21,24","Sensors":"2,3,5,6"}}
16:53:17 MQT: stat/smartplug/STATUS5 = {"StatusNET":{"Hostname":"smartplug-1737","IPAddress":"192.168.12.154","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"2C:3A:E8:0F:86:C9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:53:17 MQT: stat/smartplug/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0F86C9","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:53:17 MQT: stat/smartplug/STATUS7 = {"StatusTIM":{"UTC":"2020-05-19T09:53:17","Local":"2020-05-19T16:53:17","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+07:00","Sunrise":"05:29","Sunset":"18:08"}}
16:53:17 MQT: stat/smartplug/STATUS10 = {"StatusSNS":{"Time":"2020-05-19T16:53:17","AM2301":{"Temperature":31.3,"Humidity":96.6,"DewPoint":30.7},"TempUnit":"C"}}
16:53:17 MQT: stat/smartplug/STATUS11 = {"StatusSTS":{"Time":"2020-05-19T16:53:17","Uptime":"0T00:01:36","UptimeSec":96,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"40:E3:D6:BD:DF:E0","Channel":6,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```

### TO REPRODUCE
 Set a device name and friendly name

### EXPECTED BEHAVIOUR
It should be in its proper place

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
 
[image]

closed 2020-05-19 by close-issue-app[bot]

wongnam, 2020-05-19:
bot close. ok, no time to re-open. it a minor issue. Thanks

arendst, 2020-05-19:
It's as designed.

The default DeviceName is ALWAYS equal to FriendlyName1. By entering a 1 in the field it will replace it by the FriendlyName1

wongnam, 2020-05-19:
It makes me a little confused and unnatural. maybe only I feel that way.
[image]

arendst, 2020-05-19:
If you look at it as being just the default value it makes sense (I thought ;-)