arendst/Tasmota #8484: Sonoff Mini external switch not working with 8.3.1

matthyne, 2020-05-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After upgrade to 8.3.1 the external switch/button (GPIO4) is no longer detected on a Sonoff Mini.  

(I also add that downgrade to 8.2.0 is failing in 8.3.1)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [Y ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ Y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [Y ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ Y] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ Y] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [Y ] Device used (e.g., Sonoff Basic): _____ Sonoff Mini
- [Y ] Tasmota binary firmware version number used: _____ 8.3.1
  - [ Y] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ Y] Flashing tools used: _____ OTA
- [ Y] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
06:31:48 CMD: Backlog Template; Module; GPIO 255
06:31:48 MQT: garage_lights/stat/RESULT = {"NAME":"Sonoff Mini","GPIO":[17,255,255,255,17,0,0,0,21,56,255,0,0],"FLAG":0,"BASE":1}
06:31:48 MQT: garage_lights/stat/RESULT = {"Module":{"0":"Sonoff Mini"}}
06:31:48 MQT: garage_lights/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"17":"Button1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ N] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

No rules

```
- [ Y] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
06:26:41 MQT: garage_lights/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Garage Lights","FriendlyName":["Garage Lights"],"Topic":"garage_lights","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:26:41 MQT: garage_lights/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:09","StartupUTC":"2020-05-19T05:23:32","Sleep":50,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2020-04-05T13:51:48","SaveCount":1754,"SaveAddress":"FA000"}}
06:26:41 MQT: garage_lights/stat/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"398/699"}}
06:26:41 MQT: garage_lights/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["cleve.net",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
06:26:41 MQT: garage_lights/stat/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
06:26:41 MQT: garage_lights/stat/STATUS5 = {"StatusNET":{"Hostname":"garage_lights-7708","IPAddress":"10.1.100.230","Gateway":"10.1.100.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.4.4","Mac":"D8:F1:5B:C8:3E:1C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
06:26:41 MQT: garage_lights/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.100.222","MqttPort":1883,"MqttClientMask":"GARAGE_LIGHT","MqttClient":"GARAGE_LIGHT","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
06:26:41 MQT: garage_lights/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-05-19T05:26:41","Local":"2020-05-19T06:26:41","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:02","Sunset":"20:30"}}
06:26:41 MQT: garage_lights/stat/STATUS10 = {"StatusSNS":{"Time":"2020-05-19T06:26:41"}}
06:26:42 MQT: garage_lights/stat/STATUS11 = {"StatusSTS":{"Time":"2020-05-19T06:26:41","Uptime":"0T00:03:09","UptimeSec":189,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"cleve.net","BSSI

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
11:23:53 CMD: weblog 4
11:23:53 MQT: garage_lights/stat/RESULT = {"WebLog":4}
11:23:54 CFG: Saved to flash at F5, Count 1759, Bytes 4096
11:23:58 WIF: Checking connection...
11:24:18 WIF: Checking connection...
11:24:38 WIF: Checking connection...
11:24:58 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

GPIO0 is configured as Button1.
GPIO4 is configured as Button1.
GPIO12 is relay1

Sonoff Mini has the ability to connect an external switch (eg Pushbutton) on GPIO4.  On 8.2.0 pressing this button correctly switches the relay (as does GPIO0) however on 8.3.1 this no longer works.  No changes have been made to the config or template.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

GPIO0 or GPIO4 should trigger GPIO12 - but GPIO4 does nothing.   I am unable to test GPIO0 as the unit is installed in a wall cavity

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-05-19:
[image]

Please configure only one button1

arendst, 2020-05-19:
Indeed. Only One Button1 was supposed to work. It may be more enforced now in the latest version.

To solve your issue you need to use Switch1 for GPIO4.

matthyne, 2020-05-19:
Update - It appears that 8.3.1 does not like having two GPIOs configured as Button1.  

If I remove the Button1 from GPIO0 and only have GPIO4 configured then the external switch connection works.

Seems to be a change from 8.2

arendst, 2020-05-19:
Read the docs and learn how to configure a switch as momentary button.

matthyne, 2020-05-19:
I also suggest documenting that only one ButtonX is permitted per GPIO (assume this applies to switches and other peripherals as well).  

I will need to change all my other Sonoff mini configs before I update to 8.3.

Sidebar:  There does seem to be issues with downgrading 8.3 to 8.2 as well.  I will investigate them and post in another ticket.

blakadder, 2020-05-19:
i thought it was obvious that you can have only one button1 and only one button2, same as a relay or a pwm

ascillato2, 2020-05-19:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-19 by ascillato2

raffaeler, 2022-02-03:
@arendst Suggestion: enforce a validation on assigning twice the same component. It's easy to make this mistake when configuring the device with json.

arendst, 2022-02-04:
Was in the code some time ago. Keeping it up-to-date was a nightmare. Removed it and until you came along no-one had issues with it.

No offence but won't be re-implemented.

raffaeler, 2022-02-04:
No worries at all @arendst 
I landed here because I could not understand from the documentation what is the number after the component.
In other words I was thinking that Button1 and Button2 were referred to the I/Os and not to a unique identifier of the component.
In the components page of the docs, there is no mention at all about their unique assignment to the I/O. I understand that it is straightforward once you understand the logic of the project, but most of the people lands here just because they want to see the light turning on :)

I was thinking that a simple validation could both clarify their meaning and avoid stupid errors like mine.