arendst/Tasmota #8485: After firmware upgrade Shelly 2.5 does not recognise double click

mraczka, 2020-05-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I upgraded Shelly 2.5 and EXS Raly v5 to tasmota 8.3.1. After that devices stopped recognizing double presses on button1. Single and long press works correctly on both buttons but double click on button one triggers "doble click button 2" and double click on button2 does nothing.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____ EXS Relay V5, shelly 2.5
- [ ] Tasmota binary firmware version number used: _____ 8.3.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____GUI firmware upgrade, mqtt command, manual upgrade via file upload
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
14:08:48 MQT: stat/sypialnia/sufit/RESULT = {"NAME":"","GPIO":[52,255,92,255,21,127,0,0,6,126,5,22,156],"FLAG":2,"BASE":18}
14:08:49 MQT: stat/sypialnia/sufit/RESULT = {"Module":{"0":""}}
14:08:49 MQT: stat/sypialnia/sufit/RESULT = {"GPIO0":{"52":"Led1"},"GPIO1":{"0":"None"},"GPIO2":{"92":"Button3n"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"127":"Button2in"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"126":"Button1in"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relay2"},"GPIO16":{"156":"ADE7953 IRQ"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
14:10:57 MQT: stat/sypialnia/sufit/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":263,"Free":248,"Rules":"on button1#state=3 do publish stat/sypialnia/sufit/HOLD1 1 endon on button1#state=2 do publish stat/sypialnia/sufit/DOUBLE1 1 endon on button2#state=3 do publish stat/sypialnia/sufit/HOLD2 1 endon on button2#state=2 do publish stat/sypialnia/sufit/DOUBLE2 1 endon"}
14:10:57 MQT: stat/sypialnia/sufit/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":5,"Free":506,"Rules":"reset"}
14:10:58 MQT: stat/sypialnia/sufit/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
14:11:22 MQT: stat/sypialnia/sufit/STATUS = {"Status":{"Module":0,"DeviceName":"Sypialnia","FriendlyName":["Sypialnia","Sypialnia2"],"Topic":"sypialnia/sufit","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,5,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:23:48","StartupUTC":"2020-05-19T11:47:34","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2020-03-25T07:49:22","SaveCount":736,"SaveAddress":"F4000"}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS2 = {"StatusFWR":{"Version":"8.3.1.1(d61345e-tasmota)","BuildDateTime":"2020-05-19T12:00:45","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"384/699"}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Aurora Borealis",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000880B","0A05C8000100060000005AFF000000000000","00000000","00000000"]}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS5 = {"StatusNET":{"Hostname":"sypialnia/sufit-1347","IPAddress":"192.168.1.95","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"24:62:AB:00:85:43","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_008543","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS7 = {"StatusTIM":{"UTC":"2020-05-19T12:11:22","Local":"2020-05-19T14:11:22","StartDST":"2020-10-25T02:00:00","EndDST":"2020-03-29T02:00:00","Timezone":99,"Sunrise":"04:33","Sunset":"20:29"}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS10 = {"StatusSNS":{"Time":"2020-05-19T14:11:22","ANALOG":{"Temperature":56.7},"ENERGY":{"TotalStartTime":"2019-06-29T18:35:16","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}}
14:11:22 MQT: stat/sypialnia/sufit/STATUS11 = {"StatusSTS":{"Time":"2020-05-19T14:11:22","Uptime":"0T00:23:48","UptimeSec":1428,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Aurora Borealis","BSSId":"EC:08:6B:27:31:29","Channel":6,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
14:11:48 CMD: weblog 4
14:11:48 MQT: stat/sypialnia/sufit/RESULT = {"WebLog":4}
14:11:48 ADE: 6166043, 4477, [1773, 30, 651, 118], [1773, -7, 651, -8]
14:11:48 ADE: U 6166043, C 4477, I 0 + 0 = 0, P 0 + 0 = 0
14:11:48 CFG: Saved to flash at FB, Count 737, Bytes 4096
14:11:49 ADE: 6167894, 4476, [1773, -64, 651, 59], [1773, 19, 651, -9]
14:11:49 ADE: U 6167894, C 4476, I 0 + 0 = 0, P 0 + 0 = 0
14:11:50 ADE: 6170016, 4477, [1773, 9, 652, 78], [1773, -5, 651, 9]
14:11:50 ADE: U 6170016, C 4477, I 0 + 0 = 0, P 0 + 0 = 0
14:11:51 ADE: 6165062, 4478, [1773, -25, 651, 23], [1773, 2, 651, 9]
14:11:51 ADE: U 6165062, C 4478, I 0 + 0 = 0, P 0 + 0 = 0
14:11:52 ADE: 6219207, 4478, [1773, -6, 657, 18], [1773, -37, 657, -8]
14:11:52 ADE: U 6219207, C 4478, I 0 + 0 = 0, P 0 + 0 = 0
14:11:53 ADE: 6218002, 4478, [1773, -81, 656, 24], [1773, -4, 656, 8]
14:11:53 ADE: U 6218002, C 4478, I 0 + 0 = 0, P 0 + 0 = 0
14:11:54 ADE: 6217132, 4478, [1773, 33, 656, 27], [1773, -5, 656, 7]
[... 131 lines cut ...]
14:12:30 ADE: 6215959, 4477, [1778, -51, 658, 53], [1773, 7, 656, -9]
14:12:30 ADE: U 6215959, C 4477, I 0 + 0 = 0, P 0 + 0 = 0
14:12:31 ADE: 6206567, 4476, [1778, -57, 658, 59], [1773, -3, 655, 25]
14:12:31 ADE: U 6206567, C 4476, I 0 + 0 = 0, P 0 + 0 = 0
14:12:32 ADE: 6214746, 4476, [1778, -75, 658, 58], [1773, -3, 656, -7]
14:12:32 ADE: U 6214746, C 4476, I 0 + 0 = 0, P 0 + 0 = 0

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
On shelly 2.5 setup double and long press.
Upgrade tasmota to 8.3.1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect that nothing changes - everything works as before.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-05-19:
Button handling has changed with 8.3.x

See https://tasmota.github.io/docs/Buttons-and-Switches/#multi-press-functions

arendst, 2020-05-19:
Also from the docs:

8.3.0

🚨 BREAKING CHANGE 🚨 Buttons have a new configurable multipress feature, to activate AP mode now requires 6 button presses.

closed 2020-05-19 by ascillato2