arendst/Tasmota #8489: Apostrophe in Device Name

danmed, 2020-05-19:
### PROBLEM DESCRIPTION
If you enter a device name with an apostrophe such as "Dan's Light" when you go back to the configuration > other page, the name is only displayed up to where the Apostrophe was.. "Dan"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [Y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [Y] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [Y] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [Y] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Any
- [ ] Tasmota binary firmware version number used: 8.3.1
  - [Y] Pre-compiled
  - [N] Self-compiled
    - [NA] IDE / Compiler used: _____
- [OTA] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:10:25 CMD: Backlog Template; Module; GPIO 255
22:10:25 MQT: stat/danslamp/RESULT = {"NAME":"Novostella B22","GPIO":[0,0,0,0,37,40,0,0,38,41,39,0,0],"FLAG":0,"BASE":18}
22:10:25 MQT: stat/danslamp/RESULT = {"Module":{"0":"Novostella B22"}}
22:10:25 MQT: stat/danslamp/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"40":"PWM4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"41":"PWM5"},"GPIO14":{"39":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
22:11:16 CMD: STATUS 0
22:11:16 MQT: stat/danslamp/STATUS = {"Status":{"Module":0,"DeviceName":"Dan's Lamp","FriendlyName":[" "],"Topic":"danslamp","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
22:11:16 MQT: stat/danslamp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"bedroomlights","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:49","StartupUTC":"2020-05-19T20:59:27","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-04-27T15:27:24","SaveCount":141,"SaveAddress":"F4000"}}
22:11:16 MQT: stat/danslamp/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"379/699"}}
22:11:16 MQT: stat/danslamp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DL-NET",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8029","2805C8000100060000005A00000000000000","00000200","00000008"]}}
22:11:16 MQT: stat/danslamp/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
22:11:16 MQT: stat/danslamp/STATUS5 = {"StatusNET":{"Hostname":"danslamp-5837","IPAddress":"192.168.2.211","Gateway":"192.168.2.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.6","Mac":"BC:DD:C2:1F:D6:CD","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
22:11:16 MQT: stat/danslamp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1FD6CD","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:11:16 MQT: stat/danslamp/STATUS7 = {"StatusTIM":{"UTC":"2020-05-19T21:11:16","Local":"2020-05-19T22:11:16","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:02","Sunset":"20:30"}}
22:11:16 MQT: stat/danslamp/STATUS10 = {"StatusSNS":{"Time":"2020-05-19T22:11:16"}}
22:11:16 MQT: stat/danslamp/STATUS11 = {"StatusSTS":{"Time":"2020-05-19T22:11:16","Uptime":"0T00:11:49","UptimeSec":709,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":59,"Color":"0,0,0,28,122","HSBColor":"0,0,0","White":59,"CT":435,"Channel":[0,0,0,11,48],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"DL-NET","BSSId":"D8:0D:17:10:B9:F4","Channel":11,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
N/A

```

### TO REPRODUCE
Set a device name with an apostrophe in it.

### EXPECTED BEHAVIOUR
The name should be reflected when you visit the Configuration > Other page again

### SCREENSHOTS
https://imgur.com/a/xB3L94L

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-05-19:
Just a cosmetic issue on text field. If you look on your device page and under HA the name is correctly saved and reported.
Anyway, I would avoid apostrophe since it could be misinterpreted by the parser.
[image]

danmed, 2020-05-19:
Hi effelle, thank you for your reply, i realise it's only cosmetic, however when you have a large amount of devices, that field is a comfort that you are on the correct device.. It's also nice when you go to that page to actually see the correct name, so when editing you know what you are changing it from.

I completely agree, avoiding them would be ideal.. however to change them now would be.. well.. a pain as i assume it will alter the identity in HA and then cause issues for automations / voice assistant integrations etc.. If i am incorrect please correct me. I need a high level of WAF and if i'm about to break things.. it would be good to know so i can plan accordingly!

effelle, 2020-05-19:
Since the ID of the device will not change when you update the name, HA should keep your actual configuration. Test with one, just to be sure but should be fine.

arendst, 2020-05-21:
Now don't start to use a quotation mark in a name...

danmed, 2020-05-21:
I know, i don't i just wondered if it would escape the apostrophe.. when it
didn't i removed it

closed 2020-05-22 by ascillato2