arendst/Tasmota #8490: Wemos D1 mini the I2C not working GPIO4 GPIO5

smokey1300, 2020-05-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

smokey1300, 2020-05-20:
Upgraded from Tasmota 6.3 now the GPIO4 GPIO5 don't work for I2C

ascillato, 2020-05-20:
Please, complete the troubleshooting template. We need that to troubleshoot your issue. Thanks.

blakadder, 2020-05-20:
You failed to read changes that happened between versions. Your sensors are not supported in tasmota.bin anymore, download tasmota-sensors.bin.

smokey1300, 2020-05-20:
> You failed to read changes that happened between versions. Your sensors are not supported in tasmota.bin anymore, download tasmota-sensors.bin.

Yeah okay I loaded this version it doesn't work either

smokey1300, 2020-05-20:
I loaded v8.3.0 tasmota-sensors.bin and everything works good.

meingraham, 2020-05-20:
@smokey1300 

Now that you have the device is working, can you try upgrading to the latest development version to see if the problem was with some corruption or settings mismatch? If the problem returns, then there may be an issue with 8.3.1 to investigate... and you can always downgrade to 8.3.0 to resolve it.

Thanks.

smokey1300, 2020-05-21:
> @smokey1300
>
> Now that you have the device is working, can you try upgrading to the latest development version to see if the problem was with some corruption or settings mismatch? If the problem [...]

@blakadder Already closed this issue with v8.3.1 although it is not resolved only that v8.3.0 works as intended on the Wemos with i2c.

meingraham, 2020-05-21:
@smokey1300 

Closed because you had "resolved" _your_ issue by downgrading... but I'm not certain that this confirms an issue with your flash/settings or with 8.3.1. Can you try upgrading?

Schrumpfschlauch, 2020-05-21:
@meingraham 

I had exact the same issue  -> upgrad to 8.3.1.1(sensors) solved it!

closed 2020-05-22 by ascillato2