arendst/Tasmota #8492: unishox makes scripting fail

blakadder, 2020-05-20:
```
/home/circleci/binaries/Tasmota/tasmota/xdrv_10_scripter.ino: In function 'bool Xdrv10(uint8_t)':
 
/home/circleci/binaries/Tasmota/tasmota/xdrv_10_scripter.ino:5443:112: error: 'unishox_decompress' was not declared in this scope
 
       unishox_decompress(Settings.rules[0], strlen(Settings.rules[0]), glob_script_mem.script_ram, UNISHOXRSIZE);
```
@gemu2015

blakadder, 2020-05-20:
also get on discord so we can pester you there ;)

gemu2015, 2020-05-20:
please undef USE_RULES_COMPRESSION when using scripter for the time being.
after conversion to a class unishox shows strange behavior .
i have to investigate.

blakadder, 2020-05-20:
will do

gemu2015, 2020-05-20:
will wait for @s-hadinger´s  fix and make a corrected pr then.
until then #undef USE_RULES_COMPRESSION to avoid this error

closed 2020-05-22 by ascillato2