arendst/Tasmota #8494: display Variable/Value in Webui

riker65, 2020-05-20:
Could there be the option to display variable/mem/any value in the webui.

**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Display a value in webui
-easy option: allow a Variable in webbuttonx name
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
virtual button in webui
**Additional context**  
_Add any other context or screenshots about the feature request here._
for faster  monitoring and debugging or info it would be good to be in  aposition to customize webui and display some more Information :
- rulestatus
-var/men
-sensorvalues
**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-05-20:
The Tasmota Scripting language allows what you want.

```
>W
The lines in this section are displayed in the web UI main page. Requires compiling with #define USE_SCRIPT_WEB_DISPLAY.
```

riker65, 2020-05-20:
ah ok this could be an option.
but would like to stay with upgradeable precompiled versions.

thanks

meingraham, 2020-05-20:
I'd like to have a McLaren Senna :wink:

For these kinds of customizations, I think scripting is going to be your only option for the foreseeable future.

Compiling your own is actually quite simple using [GitPod](https://tasmota.github.io/docs/Gitpod/). That article states that you need to create your own `user_config_override.h` file first. This is no longer needed as a template is now provided that you can edit directly.

riker65, 2020-05-20:
> I'd like to have a McLaren Senna 
> 
> For these kinds of customizations, I think scripting is going to be your only option for the foreseeable future.
> 
> Compiling your own is actually quite simple using [GitPod](https://tasmota.github.io/docs/Gitpod/). That article states that you need to create your own `user_config_override.h` file first. This is no longer needed as a template is now provided that you can edit directly.

ok thanks will check

riker65, 2020-05-20:
but I thought the option with `webbuttonx %var1% `would be not a McLaren Senna.....
but If you offer the Senna, I would go for it.

ascillato2, 2020-05-22:
> The Tasmota Scripting language allows what you want.

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-22 by ascillato2