arendst/Tasmota #8498: Physical dimmer toggle changing brightness but not sending updated state

jwoodard80, 2020-05-20:
### PROBLEM DESCRIPTION 
_A clear and concise description of what the problem is._
State msg not sent when using physical dimmer up/down buttons to change brightness

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Martin Jerry SD-01 Dimmer
- [x ] Tasmota binary firmware version number used: 8.3.1
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: OTA update
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:23:09 MQT: house/office-lights/stat/RESULT = {"NAME":"MJ-SD02","GPIO":[19,18,0,59,158,58,0,0,57,37,56,122,29],"FLAG":0,"BASE":73}
23:23:09 MQT: house/office-lights/stat/RESULT = {"Module":{"0":"MJ-SD02"}}
23:23:09 MQT: house/office-lights/stat/RESULT = {"GPIO0":{"19":"Button3"},"GPIO1":{"18":"Button2"},"GPIO2":{"0":"None"},"GPIO3":{"59":"Led4i"},"GPIO4":{"158":"LedLinki"},"GPIO5":{"58":"Led3i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"57":"Led2i"},"GPIO13":{"37":"PWM1"},"GPIO14":{"56":"Led1i"},"GPIO15":{"122":"Button1i"},"GPIO16":{"29":"Relay1i"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here: No rules

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:23:43 MQT: house/office-lights/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"office-lights","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:23:43 MQT: house/office-lights/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T03:40:47","StartupUTC":"2020-05-20T18:42:56","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-05-20T05:33:59","SaveCount":115,"SaveAddress":"F6000"}}
23:23:43 MQT: house/office-lights/stat/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":4,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"400/699"}}
23:23:43 MQT: house/office-lights/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHome",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","0505C8000100060000005A00000000000000","00000200","00000000"]}}
23:23:43 MQT: house/office-lights/stat/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:23:43 MQT: house/office-lights/stat/STATUS5 = {"StatusNET":{"Hostname":"Office-lights ","IPAddress":"10.20.90.39","Gateway":"10.20.90.1","Subnetmask":"255.255.255.0","DNSServer":"10.20.90.1","Mac":"84:0D:8E:91:3D:7C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:23:43 MQT: house/office-lights/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.20.90.5","MqttPort":1883,"MqttClientMask":"clientofficelights","MqttClient":"clientofficelights","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:23:43 MQT: house/office-lights/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-05-20T22:23:43","Local":"2020-05-20T23:23:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:01","Sunset":"20:31"}}
23:23:43 MQT: house/office-lights/stat/STATUS10 = {"StatusSNS":{"Time":"2020-05-20T23:23:43"}}
23:23:43 MQT: house/office-lights/stat/STATUS11 = {"StatusSTS":{"Time":"2020-05-20T23:23:43","Uptime":"0T03:40:47","UptimeSec":13247,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SmartHome","BSSId":"92:2A:A8:91:EC:F6","Channel":11,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here: No console output 

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Default install and configuration of the switch template. Watching console shows state message when changing dimmer VIA web or external (home assistant). Changing via physical toggle DOES change the brightness but does not send the new STATE msg

### EXPECTED BEHAVIOUR
pushing up/down buttons will change brightness and trigger a STATE message with the new brightness level

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I originally did a correct upgrade path from 6.x to latest. I have set SO59 as well as verifying with rules to see if anything was triggered dimmer#state. When that did not work, I factory reset and tried again. Still did not work. I tried downgrading to 8.2.0 and everything worked as expected. 

Happy to provide any other information as needed. Thanks
 

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-05-20:
Maybe PWMDimmer is not cooperating well with new button features? 
@pcdiem can you investigate?

pcdiem, 2020-05-21:
Fixed in [PR 8500](https://github.com/arendst/Tasmota/pull/8500). The new state was published when the dimmer buttons were held but not when they were tapped.

jwoodard80, 2020-05-21:
@pcdiem Thank you so much for your quick work on this! I look forward to this change.

closed 2020-05-21 by ascillato2