arendst/Tasmota #8502: Build in upgrade path and (maybe) WebUI main page

schrej-zz, 2020-05-21:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No problem, just ease of use
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Automatically append the build to the upgrade URL to ensure upgrade to the same build. E.g. tasmota-sensors.bin. Otherwise an upgrade might install the wrong build.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Always look into the information page first and cache upgrade path manually.
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-05-21:
Pls elaborate.

The OTA URL never changes so I don't see why an upgrade path needs to be updated manually.

schrej-zz, 2020-05-21:
@arendst Maybe I got something wrong, but my default update URL is always `http://thehackbox.org/tasmota/release/tasmota.bin`, no matter, if I have installed e.g. tasmota-sensors.bin. So starting an upgrade via default OTA Url would result in a wrong build afterwards.
Joerg

arendst, 2020-05-21:
You can change the OTA Url either in the GUI OR with command ``otaurl http://thehackbox.org/tasmota/release/tasmota-sensors.bin``.

Once done it stays there. A simple OTA upgrade is performed again using the GUI or with command ``upoad 1``.

If there is not enough room for a one-time-OTA it automagically loads the -minimal version and continues loading the final version. Just sitback and relax.

schrej-zz, 2020-05-21:
I know, but why not adding the right OTA Url right into the build?
So the OTA URL in the tasmota-sensors.bin should end with tasmota-sensors.bin.
Should not be a great issue and prevents wrong upgrades by accident.
As I said, it's not an issue but a matter of convenience.
Joerg

arendst, 2020-05-21:
For me it's a packaging issue.

So let's say you one time have to select the correct binary and you're finished for life! It's the chicken/egg issue too. In order to load the -sensors version you will definitly need to set your OTA to -sensors.

Unless ofcourse you sell devices with serial loaded -sensors binaries. Now you have to change them all by hand. In that case I should use an own compiled version with the OTAurl set in my_config_override.h ...

schrej-zz, 2020-05-21:
OK, got it. You are right. If I start using OTA right from the scratch, I have to change OTA URL anyway and then have the right path as long, as I stick with this binary.
Thanks, will close the issue.

closed 2020-05-21 by schrej-zz