arendst/Tasmota #8503: Wired ESP32 with PoE

nagyrobi, 2020-05-21:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
wESP32 is a very professional approach to the ESP family. https://wesp32.com/

Basically what it does is that it uses some of the ESP32-WROOM-32 pins connected to an Ethernet PHY module, which is also **Po**_ (supporting IEEE 802.3at standard and passive modes) not only for powering the ESP module itself, but also giving enough power to other accessories like sensors etc.

In the original firmware the module's _WiFi part can be still used as a redundant backup connection_ to the network when wired network fails.

**Describe the solution you'd like**  
Please add support for this wESP32 module.

**Describe alternatives you've considered**  
There are none so far.

**Additional context**  
[image]

s-hadinger, 2020-05-21:
The device costs 50€. That's quite high, especially compared to a regular esp32 and a POE splitter.

nagyrobi, 2020-05-21:
It's not just an esp32 and a POE splitter. It has an Ethernet PHY module on it, allowing wired LAN connections with the module. How many other such modules are out there?

Jason2866, 2020-05-21:
@s-hadinger  this board looks interesting https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

nagyrobi, 2020-05-21:
Good find!
[image]
Let's collect all such models in this issue.

ascillato2, 2020-05-21:
Have you tested if LAN connection works fine using Arduino core? Do you have any small sketch using that ethernet chip?

nagyrobi, 2020-05-21:
Look here: https://github.com/OLIMEX/ESP32-POE

s-hadinger, 2020-05-21:
Tasmota, esp32, ble, poe, ethernet, arduino compatible and affordable. I start to like it very much. Nice find!

hagenbuch, 2020-06-03:
I support this feature request strongly: This family of devices need some power supply, running from a battery is mostly impossible du to the relatively high power demand of the Wifi.

Therefor, it is an advantage if we can power it via one cable only, thus Ethernet/LAN.

Also, ESP32 is one of the rare chips able to handle HTTPS / SSL / TLS properly. This is a precondition to maiking things like electrical door locks and the like...

It would be awesome if you could add LAN support...

cg089, 2020-06-04:
+1 for ethernet support (esp32-poe works like a charm with ethernet and esphome)

dweston, 2020-06-05:
> The device costs 50€. That's quite high, especially compared to a regular esp32 and a POE splitter.

@s-hadinger US$55 for a 1 off wESP32 is expensive, I agree. However it can be purchased in volume with a price as low as US38/unit in quantities >100. At that latter price it is starting to become realistic as a LED driver/sensor board when you take into account the board can deliver circa 13W @ 12V. PoE driven LED lighting becomes feasible when driven from a variety of cost effective PoE switches. I'm using [Netgear GS728TP](https://www.netgear.com/support/product/GS728TP.aspx) switches, which have a 190W total PoE power budget supplied from 16 PoE 802.3af ports (15W capable) and 8 PoE+ 802.3at (38W capable) ports [[can be purchased on eBay for as low as US $169.00](https://www.ebay.com.au/itm/NetGear-ProSafe-GS728TP-100NAS-24-Ports-Rack-Mountable-Ethernet-Switch/333378110345?epid=168468271&hash=item4d9ee24389:g:ygsAAOxylh1SKjrb&frcectupt=true)]. Averaged out over many ports, taking into account the cost of conventional LED power supply components provision & installation, results in a more manageable cost overall. Furthermore the GS728TP can be monitored and controlled via SNMP, enabling innovation in sector lighting energy management. Quiescent  power consumption of many LED drivers can be as high as several watts even with the luminaire (LED) fully turned off. So being able to fully disable the power supply to multiple drivers based on ambient light and body movement is attractive as well.

The dilemma remains with which firmware to implement on these devices. ESPHome on the wESP32 or the Olimex ESP32 PoE boards currently appears to be the way to go, but the prospect that a Tasmota variant is really the ultimate Holy Grail.

s-hadinger, 2020-06-05:
Interesting perspectives. I believe anything above the $20 limit makes it harder for makers to hack and break. I'm still waiting my Olimex to arrive.

Tasmota on ESP32 is still early stage, but promising. Theo is very active on this front.

In order to drive LEDs at 13W, you will need both the on-board POE power supply to be sized for it (are they?) and some power drivers for the LEDs (MOSFET?). And a nice case to make it pretty.

nagyrobi, 2020-06-05:
Where do you take this $20 limit from? 
As soon as Theo provides a statement that wESP32 can run Tasmota, I'll be sure to order at least 2 pieces. I'm pretty fed up with unreliable chinese pieces, and WiFi coverage issuses in certain cases. wESP32 seems to be a premium quality and I am willing to wire up things as much as possible.
And I bet I'm not the only one thinking like that.

lbernstone, 2020-06-05:
Direct from the manufacturer:
https://www.mouser.com/new/espressif/espressif-esp32-ethernet-kit/

arendst, 2020-06-06:
Ordered the olimex iso one to play with...

hagenbuch, 2020-06-06:
I can send you one for free, we are in Germany, right?

arendst, 2020-06-06:
We are in the Netherlands, small province of Germany ;-)

hagenbuch, 2020-06-06:
Not yet :) Well I can send it to the Netzerlands too if you like.

arendst, 2020-06-06:
Pls mail me from the github mainpage

hagenbuch, 2020-06-06:
Done :)

TrDA-hab, 2020-06-07:
https://norvi.lk/industrial-esp32-norvi-iiot/

hagenbuch, 2020-06-07:
Yeah thanks, close but no LAN / Ethernet. Wifi inside a metal box is not exactly what I need :)

TrDA-hab, 2020-06-08:
@hagenbuch 
Sorry!
https://aliexpress.ru/item/4000016520686.html
https://aliexpress.ru/item/4000188817036.html
https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware
https://olimex.wordpress.com/2017/02/09/esp32-evb-our-new-board-for-iot-first-prototypes-are-ready/

hagenbuch, 2020-06-08:
Thank you. I knew most of the. Now the only question is: Are they all supported by tasmota and / or how do we make a configuration file for them if they are not yet supported?

nagyrobi, 2020-06-09:
Well, hopefully soon we will be able to make a config almost the same way as we would do with the ESP8266-based units.

Today I was in a place where I could have instantly suggested to use Tasmota for a small automation task, but was not able to because of the WiFi requirement of ESP8266. Here only Ethernet is available and deploying WiFis just for this reason is simply unacceptable.

Jason2866, 2020-06-09:
@nagyrobi @hagenbuch @TrDA-hab 
Since Theo ordered a Olimex ESP32 ISO he will work on (for sure).
Just give him a little time. It is done when it is done 😊

nagyrobi, 2020-06-09:
I don't mean to rush anyone. I was just emphasizing that the need of LAN without any WiFi is a real need.

arendst, 2020-06-15:
Currently only works on Olimex ESP32-PoE and I haven't tesetd the PoE functionality yet ;-)

BBBits, 2020-06-16:
loaded the new build on an Olimex ESP32-PoE & noticed a couple of things.
device rebooted repeatedly while connected via eth. It is still requiring the AP mode to be configured before it will stop restarting. can this be disabled by default ?

Lables for the different interfaces would make it clearer what is what in the information page.
I.E. eth_hostname ,eth_mac ,eth_ip  etc
[image]

arendst, 2020-06-16:
> It reboots repeatedly

Within seconds or every three minutes? If it's every 3 minutes you could have changed either ``wificonfig`` or disabled wifi al together with ``wifi 0``. You could also have configured the correct wifi credentials to make it stop starting the AP mode.

> Lables for the different interfaces would make it clearer what is what in the information page.

The top IP address is the ethernet IP address as can be seen from the Hostname ending in ``_eth``. The second IP address is the wifi address. The third one is an AP address although at that time the AP is not even running. A remnant from the ESP32 implementation.

arendst, 2020-06-16:
Were you able to use PoE?

BBBits, 2020-06-16:
POE working fine,
It rebooted every 3 mins perhaps , didn't time it. wasn't a short time.
I do see what ip info is where , but it would make sense to define them clearly.
there are  lables all saying IP Address without specifically associating it to an interface.

BBBits, 2020-06-16:
I have disabled wifi to stop it restating , 
In the big picture , I would like to have all the local esp82xx's connect to the esp32 AP.
can I leave the AP running without it rebooting ?

arendst, 2020-06-16:
Currently you can't as the implementation is still based on esp8266 initial installation. How many clients do you expect the ESP32 can service? I don't know but I'm not expecting that many.

kugelkopf123, 2020-06-16:
As far as i know there is a limit set to 10 devices on AP mode.

arendst, 2020-06-16:
Ok. This looks better.

[image]

Now also tested with a Ubiquiti PoE adapter GP-V480-032G. 
Another Ubiquiti PoE adapter GP-A240-050G fails to power my Olimex.

arendst, 2020-06-16:
With the latest commit you get three new commands to configure "any" ESP32 ethernet interface either based on LAN8720 (Olimex) or TLK110.

You will also need a template for the Olimex to function well so before upgrading first install this template.

```
{"NAME":"Olimex ESP32-PoE","GPIO":[65504,65504,65504,65504,65504,65504,0,0,5536,65504,65504,65504,65504,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,65504,65504,65504,65504,65504,0,0,65504],"FLAG":0,"BASE":1}
```

nagyrobi, 2020-06-16:
Theo, please be aware that according to the Ethernet standards a client device cannot have more than one default gateway set. Units operating with more gateways are the routers, which based on various (manual or automatic) routing tables are able to route traffic according to these.

I guess it's not the scope of ESP32 to implement an entire routing platform so may I suggest to draw a line here and accept a statement that a device can only have one single default gateway set. Things can get complicated when a user connects with a device to two different networks using DHCP for example. Which one will win? The last one which connected will always overwrite the default gateway address?

Perhaps a kind of priority setting would be helpful here, helping to decide which interface is allowed to set the default gateway using DHCP and when.
For example:
- Priority Ethernet: the default gateway is set from Ethernet when link established. When Ethernet not connected, default gateway can be set from WiFi.
- Priority WiFi (the other way around):  the default gateway is set from WiFi when connected. If wifi not connected, default gateway can be set from Ethernet.
These all above of course only when using DHCP or with static addresses and default gateway set. One could configure manually static address to the interface and not set a default gateway - that's the easiest case because you take the default gateway from the other one.

Whenever the default gateway address changes due to intermittent connectivity, network connections have to be reset.

nagyrobi, 2020-06-16:
> Another Ubiquiti PoE adapter GP-A240-050G (24V) fails to power my Olimex.
Note that this is not a standard 802.3af PoE adapter, this is a so-called passive, older Ubiquiti-proprietary adapter. This shouldn't work.

BBBits, 2020-06-17:
Thanks Theo, good having the isolation between the interfaces.
My usage idea is not huge.
although I have a heap of devices , 
there are only 6 mini's  that are hard wired to lights and also controlled by PIR's. the extra switch Input just sends out knx packets on some of them.
I was hoping to just connect those 6 devices to the ESP32  .
Haven't considered all the network issues yet but think an isolated one is probably less caos than a bridge. was considering changing the knx to broadcast instead of multicast on the isolated network.

arendst, 2020-06-17:
@nagyrobi thanks for the PoE adapter info. Makes sense to me. I've also readup on this and noticed there is PoE and PoE.

Regarding the "router" functionality it will indeed NOT be supported by Tasmota. The fact you saw the AP info in the information page was just a remnant from the ESP8266 to ESP32 conversion which is fixed now; the AP was never allive.

[image]

So for now let's close this issue as being fixed and see when other ethernet hardware turns up what is not supported by the current solution. At least PoE is transparent to the software solution as far as I can see.

closed 2020-06-17 by arendst

dweston, 2020-06-17:
@nagyrobi have you tried flashing a wESP32 with this latest version of Tasmota32 as yet?

I have access to one of these modules with an IoT dev group I'm working with. I'm in the process of retrieving it in order to try flashing it with this latest version. If I'm successful I'll report it here.

nagyrobi, 2020-06-17:
My wESP32 and Olimex are still on the way...

BBBits, 2020-06-17:
i think there is an issue with the interface used as gateway.
If I turn wifi off , timeserver is not contacted & we stay at 1970
Enable wifi & ntp update is ok

nagyrobi, 2020-06-17:
Yep, that's what I was talking about. Extra care should be taken when enabling/disabling interfaces: unless somebody deliberately wants to avoid using gateway addresses (static IP without gateway set), it should be ensured that **only one** gateway address is **always set**, no matter how interfaces go up or down.

arendst, 2020-06-17:
There is only one gateway address supported within Tasmota. There is also only one network supported within Tasmota. So both ethernet AND wifi have to be in the same network if enabled.

The latest commit makes sure that if ethernet only is configured the DHCP supplied gateway, netmask and DNS server are set correctly.

nagyrobi, 2020-06-17:
If both ethernet and wifi are in the same network, how does Tasmota decide which interface is actually used when sending packets? For example, when contacting the NTP server?

arendst, 2020-06-17:
I don't know. Does it matter as long as it's the same network? I know Tasmota does nothing with regards to network selection.

nagyrobi, 2020-06-17:
It does, as network switches handle packets based on MAC address in Layer2 networking (the same network). It will cause trouble if a packet goes out on one interface, but answer is coming back on the other. If we go one layer up, where the wifi interface and the ethernet interface have different IP addresses, communication will not establish if packet is sent out via one IP and the answer is coming back to the other.

With such an ESP32 having both LAN and WiFi we usually want redundant (backup) connections to achieve higher reliability, some kind of priority system would make the system more aware of what's happening there, like detect when ethernet link drops, and then automatically switch on to WiFi, and switch WiFi off as soon as Ethernet comes back etc

arendst, 2020-06-17:
Understand. Will need to dive deep in the espressif/arduino/esp32 core code of available features to find how to bind what to what.

If anyone feels the need too I'm open for any hints.

nagyrobi, 2020-06-17:
https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/components/modules/eth.c
see
```
  { "start",            SYSTEM_EVENT_ETH_START,           empty_arg     },
  { "stop",             SYSTEM_EVENT_ETH_STOP,            empty_arg     },
  { "connected",        SYSTEM_EVENT_ETH_CONNECTED,       empty_arg     },
  { "disconnected",     SYSTEM_EVENT_ETH_DISCONNECTED,    empty_arg     },
  { "got_ip",           SYSTEM_EVENT_ETH_GOT_IP,          eth_got_ip    },
```
also `#include "eth_phy/phy_ip101.h"` - this is another module type?

hagenbuch, 2020-06-17:
To Theo and all the others involved: Thank you very much for working so much on the ESP32 / Olimex version. Unfortunately, i am not able to contribute right now but as soon as I have more time, i will  :)

Jason2866, 2020-06-17:
Should be no problem, ethernet and Wifi -> there is no bridge

nagyrobi, 2020-06-17:
OK let's see it in action...

Jason2866, 2020-06-17:
and Lwip NAT is Arduino standard  ;-)
To get a bridge running is "some" work. It will not happen by accident ;-)

arendst, 2020-06-17:
@nagyrobi I see nothing binding related in the NodeMcu code. It's more or less a copy of the espressif code with an addition of type IP101.

See my xdrv_82_ethernet.ino also considering the callbacks you mention specifically, they are there too.

So for now indeed let's see when things get out of hands.

In the worst case I can always decide to support either ethernet OR wifi.

BBBits, 2020-06-18:
Still having issues with ntp.
had a sniff around support_rtc.ino
in RtcSecond() 
Rtc.ntp_time = sntp_get_current_timestamp();  Is Not called if global_state.wifi_down

changed 
if (!global_state.wifi_down) {
To
if ((!global_state.wifi_down)||(!global_state.eth_down)) {
And it now updates time.
So not an interface gateway problem.

arendst, 2020-06-18:
I'm afraid I didn't sync all my ethernet related changes correctly. NTP worked fine for me as I had this change locally. I'll revisit the differences as I was quite sure I changed most wifi_down to network_down already although not all must be changed.

My fault. Sorry.

arendst, 2020-06-18:
Verified to work too:

[image]

nagyrobi, 2020-06-19:
Trying tasmota32-sensors.bin from the [precompiled binaries](https://github.com/arendst/Tasmota/tree/firmware/firmware/tasmota32) but I get:
```
14:53:35 CMD: Ethernet
14:53:35 RSL: stat/tasmota_B90D58/RESULT = {"Command":"Unknown"}
```

Jason2866, 2020-06-19:
@nagyrobi  You need in your `user_config_override.h` following defines:

`#define USE_ETHERNET`

[52 more replies]