arendst/Tasmota #8505: Use of sonoff as bathroom fan controler (off delay timer + humidity)

radudorurus, 2020-05-21:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
The feature is not related to a problem
I searched for a while an did not find any idea how to implement or if it is possible using the existing features of the sonoff/tasmota.

**Describe the solution you'd like**  
I think most of us heard about bathroom fan timer controller.
Usually you turn the bathroom light on which signal is used by the timer module to start the fan.
After the light is turned off, the module will keep the fan running for some amount of time set. (1-15min).

Actually I have one of that kind of fan with timer and humidity feature.
If humidity is greater than the threshold, the controller will start the fan ignoring the input from the light.

So using the already present features in the tasmota software, the humidity can be read from the appropriate/supported sensor. (Sonoff TH10 or sonoff basic modded in sonoff TH)
The high voltage signal from light can be interfaced with one GPIO by an AC optocoupler.
The signal can come from a PIR sensor instead of the light.

Optional1:
A supplementary debounce for the activation signal can be introduced. 
What if the light is active just 1s? should the fan start and run for 3mins?
So a debounce of 2s should be required for starting the fan/relay.

Optional2:
A have seen one variant of bathroom fan timer which starts after the light is turned off and run 3-5mins. I think this is for avoid disturbing the person in the bathroom. 

**Describe alternatives you've considered**  
I tried to think of using the Pulsetime command but Pulsetime is started when the relay is ON or OFF.
In the scenario of the fan timer, the Pulsetime should start when the off command is issued (one GPIO port go from active to inactive).
I think "Optional2" is doable with Pulsetime.

Think of the bathroom light are hardwired (mechanical switches, high voltage), and it is quite difficult to replace these switches with push buttons.

I know there are other alternatives like using HomeAuromation with MQTT which get info/status from the light switch or humidity sensor and control the fan switch. (I mean smart switches)
But I think here as a standalone solution: sonoff/tasmota as humidity/timer for bathroom fan. 

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

rbswift, 2020-05-22:
Tasmota works fine for me as a bathroom fan timer & humidity sensor. I don't think any code changes are necessary. You can use PulseTime and a Rule:
PulseTime 580
Rule1 on BME280#Humidity>75 do Power on

You can also use PowerOnState 2 (toggle) and replace your FAN switch with a momentary NC push button and this will allow you to manually toggle the fan on and off using existing wiring but humidity will always override

ascillato2, 2020-05-22:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-22 by ascillato2

radudorurus, 2020-05-22:
The answer is wrong

ascillato, 2020-05-22:
Hi,

Yes, that behaviour can be achieved using the Rules' feature of Tasmota.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

radudorurus, 2020-05-22:
Thanks for the suggestion.