arendst/Tasmota #8508: Offline data collection (+ async data reporting via MQTT)

jensb, 2020-05-21:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I run a few Tasmota devices (eg. Gosund SP1) in areas where Wifi access is either unreliable or actively being switched off together with some other hardware to save energy during the night. This causes my Tasmota devices to stop reporting data (of course) but they *also* don't reset values at midnight (e.g. resetting ENERGY_Today, rolling over to ENERGY_Yesterday, etc) when they aren't online exactly at midnight.

**Describe the solution you'd like**  
I would like Tasmota to keep a buffer of values, at least internally, to be able to report values which rarely change (e.g. summary values like ENERGY_Yesterday) even when Wifi access isn't always available.
Ideally, I'd like Tasmota to buffer all MQTT messages when no Wifi is there and to send them out once Wifi returns - of course, with appropriate timestamps so a MQTT broker can see these are outdated messages and (probably) just for storage, and with a configurable maximum number of stored messages so internal memory isnt' exhausted.

**Describe alternatives you've considered**  
The easiest alternative would be to ensure consistent Wifi access in all locations. But this would cost me quite a bit, consume quite a bit of energy and Tasmota wouldn't benefit.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-05-22:
Hi,

There is not enough space on an ESP8266 chip for data logging.

Anyway, Tasmota supports SD card by using SCRIPT feature.

Please, check the docs for scripting to achieve the behaviour you want.

If you need help on that, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-22 by ascillato2

jensb, 2020-05-31:
OK, I understand.
But somewhere Tasmota stores the ENERGY_Today and ENERGY_Yesterday values because they are incremented constantly over the day.
Can we then agree on incrementing just *these* cumulative values even offline, so we have at least correct *daily* values, and resetting ENERGY_Today (and updating ENERGY_Yesterday) also when offline, so that these values are also correct even if the Wifi drops out sporadically?
Thank you!

ascillato, 2020-05-31:
They are saved into memory only at midnight. They will continue incrementing (in RAM), but saved into EEPROM at midnight. If you cut the power of the device before midnight, your data of the day will be lost.

It is saved that data only once a day to prevent flash wear. If it is saved more often your device will wear and stop working in few months.

jensb, 2020-06-02:
So ... Tasmota reboots every time it loses Wifi connectivity? Is this necessary? Because not rebooting every time Wifi connectivity drops out would also solve the problem.

arendst, 2020-06-02:
Learn how to configure tasmota for wifi loss without restart. Find info about ``wifimanager``.

jensb, 2020-06-08:
Thanks. I tried this and at least there is no more data loss for the _Today and _Yesterday values, which is sufficient for my purposes.