arendst/Tasmota #8509: PowerOnState 6 - POWER ON and after PulseTime POWER OFF

fufs, 2020-05-22:
**Have you looked for this feature in other issues and in the docs?**  

Yes, I couldn't find any information about it.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

I have a Sonoff Basic in an outside lamp. This lamp is connected to a regular switch. Right now I have set PowerOnState to 1 and SwitchMode to 13 with PulseTime1 set to 3h. This works ok. However, when someone at the house wants to turn off the light they go to reach for the switch instead of issuing a voice command. Or if the want to leave the light on for the whole, they can't, because it turns off after 3h. 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

I would like to have a PowerOnState option 6, which would behave as a SwitchMode 13 (1 is On, turn off after PulseTime), but have SwitchMode set to default.

This solution allows using the switch to turn the lamp on, which would turn off after x minutes (so nobody would hit the switch again to turn the lamp off), but if the lamp was turned on using either GUI, REST or MQTT it would stay on until someone (or something like automation) turns it off.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

I could probably make an automation in Home Assistant but would prefer it was implemented in Tasmota software. It would be so much better.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-05-22:
Hi, that behaviour can be achieved using rules.

Please, for Tasmota usage, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-22 by ascillato2

fufs, 2020-05-22:
I've managed to do it using this rule (PowerOnState = 0)
```
rule:
  ON Power1#Boot DO PulseTime1 700 ENDON 
  ON Power1#Boot DO POWER ON ENDON 
  ON Power1#Boot DO RuleTimer1 605 ENDON 
  ON Rules#Timer=1 DO PulseTime1 0 ENDON
```
It works fine however if I do a soft reboot or Tasmota issues one, then the light comes on, but with PowerOnState set to 1, turn it off it and do a soft reboot, then the light stays off. There probably is some complicated method to get it to work with rules however, I believe it would be a good feature to add to Tasmota.

ascillato, 2020-05-22:
If you want to use several commands in the same rule trigger you can use `backlog` as `ON Power1#Boot DO backlog PulseTime1 700;POWER ON;RuleTimer1 605 ENDON`

Anyway, It is not good for a sonoff basic to be connected differently than the manufacturer explains in the product instructions. If you switch off the energy of the device with a switch several times a day, will eventually degrade the internal power regulator. For the use case you have explained, you should use a Shelly 1, so you can connect the normal switch to the shelly (and put it inside the switch box) and let the shelly to control the light. In that way, you can control your light from your switch AND from MQTT. Shelly is designed to work that way ( Control by a SWITCH and MQTT ). A sonoff basic was made for the use cases where you need control ONLY over wifi.

fufs, 2020-05-22:
Or even better a rule trigger that only triggers after full reboot and not soft reboot (warm restart)

fufs, 2020-05-22:
> If you want to use several commands in the same rule trigger you can use `backlog` as `ON Power1#Boot DO backlog PulseTime1 700;POWER ON;RuleTimer1 605 ENDON`
>
> Anyway, It is not good for a sonoff [...]

I know, but all switches in my house are wired as Line in, Line out. There is no neutral in the switch box. The best option would be to short the wires in the switch box and place either a Zigbee or RF switch in front of it to cover the hole. But the Sonoff must do for now and since it's 5$ a piece I don't really care if the internal power regulator dies.