arendst/Tasmota #8511: 8.3.1-Dubble tap on Sonoff T1 2CH button 1 triggers relay 2

mainframecn, 2020-05-22:
### PROBLEM DESCRIPTION
Dubble tap on button 1 off a Sonoff T1 2CH, triggers relay 2. (tasmota.bin 8.3.1 firmware)
I did a full reset and only set the module type to "Sonoff T1 2CH (29)". 
After a restore to 8.2.0 (full reset and module to 29), the issue was gone. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [-] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [- ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff T1 2CH
- [ ] Tasmota binary firmware version number used: 8.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:21:46 CMD: Backlog Template; Module; GPIO 255
21:21:46 SRC: WebConsole from 192.168.0.110
21:21:46 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
21:21:46 SRC: Backlog
21:21:46 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
21:21:46 RSL: stat/tasmota_772A1F/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:21:46 SRC: Backlog
21:21:46 CMD: Group 0, Index 1, Command "MODULE", Data ""
21:21:46 RSL: stat/tasmota_772A1F/RESULT = {"Module":{"29":"Sonoff T1 2CH"}}
21:21:47 SRC: Backlog
21:21:47 CMD: Group 0, Index 1, Command "GPIO", Data "255"
21:21:47 RSL: stat/tasmota_772A1F/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
21:23:27 CMD: Backlog Rule1; Rule2; Rule3
21:23:27 SRC: WebConsole from 192.168.0.110
21:23:27 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
21:23:27 SRC: Backlog
21:23:27 CMD: Group 0, Index 1, Command "RULE", Data ""
21:23:27 RSL: stat/tasmota_772A1F/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
21:23:28 SRC: Backlog
21:23:28 CMD: Group 0, Index 2, Command "RULE", Data ""
21:23:28 RSL: stat/tasmota_772A1F/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
21:23:28 SRC: Backlog
21:23:28 CMD: Group 0, Index 3, Command "RULE", Data ""
21:23:28 RSL: stat/tasmota_772A1F/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:16:40 CMD: STATUS 0
21:16:40 SRC: WebConsole from 192.168.0.110
21:16:40 CMD: Group 0, Index 1, Command "STATUS", Data "0"
21:16:40 RSL: stat/tasmota_772A1F/STATUS = {"Status":{"Module":29,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_772A1F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:47","StartupUTC":"2020-05-22T20:12:53","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2020-05-22T21:12:57","SaveCount":20,"SaveAddress":"F8000"}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"339/699"}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MyWifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS5 = {"StatusNET":{"Hostname":"tasmota_772A1F-2591","IPAddress":"192.168.0.133","Gateway":"192.168.0.1","Subnetmask":"255.255.254.0","DNSServer":"185.228.168.168","Mac":"84:0D:8E:77:2A:1F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_772A1F","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS7 = {"StatusTIM":{"UTC":"2020-05-22T20:16:40","Local":"2020-05-22T21:16:40","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:59","Sunset":"20:34"}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS10 = {"StatusSNS":{"Time":"2020-05-22T21:16:40"}}
21:16:40 RSL: stat/tasmota_772A1F/STATUS11 = {"StatusSTS":{"Time":"2020-05-22T21:16:40","Uptime":"0T00:03:47","UptimeSec":227,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MyWifi","BSSId":"C4:AD:34:14:2C:D3","Channel":11,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
21:15:08 APP: Button1 multi-press 2
21:15:08 SRC: Button
21:15:08 RSL: stat/tasmota_772A1F/RESULT = {"POWER2":"ON"}
21:15:08 RSL: stat/tasmota_772A1F/POWER2 = ON
21:15:08 DBG: Relay2 found on GPIO5

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. flash 8.3.1 tasmota.bin
2. reset (5) in console
3. change module to 29 (using pull down menu in configuratio)
3. double tap button 1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
With the default settings, dubbel tap should not trigger any relay.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-22:
Hi,

Sorry, not an issue. That is a new feature. Please check the releases notes. Thanks.

closed 2020-05-22 by ascillato2

ascillato2, 2020-05-22:
A great explanation of this can be found at https://www.youtube.com/watch?v=jcdPRbWCGjE

mainframecn, 2020-05-23:
Thanks for the additional information. I will look into the feature a bit more.