arendst/Tasmota #8513: SO19 is not changing FullTopic anymore

araczkowski, 2020-05-23:
### PROBLEM DESCRIPTION
according to information in some issues and documentation SO19 1 changes full topic from %prefix%/%topic%/ to %topic%/%prefix%/
but this is not the case anymore

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
09:53:04 RSL: stat/tasmota_3B102F/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
09:53:04 RSL: stat/tasmota_3B102F/RESULT = {"Module":{"1":"Sonoff Basic"}}
09:53:05 RSL: stat/tasmota_3B102F/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
09:53:12 RSL: tele/tasmota_3B102F/HASS_STATE = {"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Module or Template":"Sonoff Basic","RestartReason":"Software/System restart","Uptime":"0T00:10:41","Hostname":"tasmota_3B102F-4143","IPAddress":"192.168.5.200","RSSI":"100","Signal (dBm)":"-46","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:07","MqttCount":0,"LoadAvg":19}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
-

```
- [ ] Provide the output of this command: `Status 0`:
```
09:53:48 RSL: stat/tasmota_3B102F/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_3B102F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:17","StartupUTC":"2020-05-23T08:42:31","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2020-05-23T09:42:31","SaveCount":7,"SaveAddress":"F5000"}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"341/699"}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ais-dom","8DB0839D"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS5 = {"StatusNET":{"Hostname":"tasmota_3B102F-4143","IPAddress":"192.168.5.200","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.5.1","Mac":"2C:3A:E8:3B:10:2F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B102F","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS7 = {"StatusTIM":{"UTC":"2020-05-23T08:53:48","Local":"2020-05-23T09:53:48","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:58","Sunset":"20:35"}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS10 = {"StatusSNS":{"Time":"2020-05-23T09:53:48"}}
09:53:48 RSL: stat/tasmota_3B102F/STATUS11 = {"StatusSTS":{"Time":"2020-05-23T09:53:48","Uptime":"0T00:11:17","UptimeSec":677,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ais-dom","BSSId":"B4:0F:3B:45:06:B9","Channel":6,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
-

```

### TO REPRODUCE
- program with new wersion
- reset settings Reset 5
- check FullTopic %prefix%/%topic%/ 
- SO19 1
- check FullTopic %prefix%/%topic%/ 

### EXPECTED BEHAVIOUR
the FullTopic should change after SO19 1 to %topic%/%prefix%/

### SCREENSHOTS
-

### ADDITIONAL CONTEXT
-

Jason2866, 2020-05-23:
Yes, as designed for the new release. Read the documentation
https://tasmota.github.io/docs/

araczkowski, 2020-05-23:
Fine by me. I just need to change the automatios in my project. I hope this will be all...
Can you point me to the  PR where this was done? I just need to know is there any other places I should check. Thanks.
Closing this since this is a feature not a bug.

effelle, 2020-05-23:
Or you can simply invert manually `%prefix%/%topic%/`. The discovery will not revert them. 
The inversion is already present on 8.2 master. And was introduced on dev channel in February. See https://github.com/arendst/Tasmota/pull/7680.

closed 2020-05-23 by effelle

araczkowski, 2020-05-23:
@effelle thanks for details. Now I see how this was done. Regards :+1: