arendst/Tasmota #8514: Pressing 5 times = PENTA and 6 times = PENTA

tiagofreire-pt, 2020-05-23:
### PROBLEM DESCRIPTION
With 8.3.1, on a Sonoff R3 RF, configured with the template for Sonoff R3, activating the SO73 flag ("1"), the clicks shows on console as:

1 click = SINGLE
2 clicks = DOUBLE
3 clicks = TRIPLE
4 clicks = QUAD
5 clicks = PENTA
6 clicks = PENTA

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff R3 RF
- [X] Tasmota binary firmware version number used: 8.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
14:46:45 MQT: stat/candeeiro_sala/RESULT = {"NAME":"Basic R3","GPIO":[17,255,0,255,255,0,0,0,21,56,255,0,255],"FLAG":0,"BASE":1}
14:46:46 MQT: stat/candeeiro_sala/RESULT = {"Module":{"0":"Basic R3"}}
14:46:46 MQT: stat/candeeiro_sala/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
N/A
```
- [X] Provide the output of this command: `Status 0`:
```
14:47:33 MQT: stat/candeeiro_sala/STATUS = {"Status":{"Module":0,"DeviceName":"Candeeiro Sala","FriendlyName":["Candeeiro Sala"],"Topic":"candeeiro_sala","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:47:33 MQT: stat/candeeiro_sala/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:04","StartupUTC":"2020-05-23T13:42:29","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-05-23T14:35:46","SaveCount":15,"SaveAddress":"F5000"}}
14:47:33 MQT: stat/candeeiro_sala/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"440/699"}}
14:47:33 MQT: stat/candeeiro_sala/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PA_WIFI","PSA_WIFI"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00800000","00000000"]}}
14:47:33 MQT: stat/candeeiro_sala/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
14:47:33 MQT: stat/candeeiro_sala/STATUS5 = {"StatusNET":{"Hostname":"candeeiro_sala-1440","IPAddress":"192.168.11.209","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"DC:4F:22:AA:85:A0","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
14:47:33 MQT: stat/candeeiro_sala/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.61","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA85A0","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:47:33 MQT: stat/candeeiro_sala/STATUS7 = {"StatusTIM":{"UTC":"2020-05-23T13:47:33","Local":"2020-05-23T14:47:33","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:58","Sunset":"20:35"}}
14:47:33 MQT: stat/candeeiro_sala/STATUS10 = {"StatusSNS":{"Time":"2020-05-23T14:47:33"}}
14:47:33 MQT: stat/candeeiro_sala/STATUS11 = {"StatusSTS":{"Time":"2020-05-23T14:47:33","Uptime":"0T00:05:04","UptimeSec":304,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":36,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"PSA_WIFI","BSSId":"2C:FD:A1:3E:31:58","Channel":11,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}
14:47:38 MQT: tele/candeeiro_sala/STATE = {"Time":"2020-05-23T14:47:38","Uptime":"0T00:05:09","UptimeSec":309,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"PSA_WIFI","BSSId":"2C:FD:A1:3E:31:58","Channel":11,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
14:48:31 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"SINGLE"}
14:48:43 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"DOUBLE"}
14:48:48 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"TRIPLE"}
14:48:52 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"QUAD"}
14:48:57 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"PENTA"}
14:49:02 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"PENTA"}

```

### TO REPRODUCE
Just use a plain valina Tasmota on Sonoff R3, with SO73 enabled.

### EXPECTED BEHAVIOUR
1 click = SINGLE
2 clicks = DOUBLE
3 clicks = TRIPLE
4 clicks = QUAD
5 clicks = PENTA
6 clicks = HEXA

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-23:
@effelle  This one if for you :-)

effelle, 2020-05-23:
No. 6 clicks will enable `wificonfig 2`, then is not a possible trigger hence you will not say an MQTT message. 
Can't reproduce anyway, please use `weblog 4` and enable `SetOption1`, doing so you will see how many clicks you did:
```
12:24:12 APP: Button1 multi-press 1
12:24:13 MQT: stat/tasmota_49A3BC/BUTTON1 = {"ACTION":"SINGLE"}
12:24:16 APP: Button1 multi-press 1
12:24:16 APP: Button1 multi-press 2
12:24:16 MQT: stat/tasmota_49A3BC/BUTTON1 = {"ACTION":"DOUBLE"}
12:24:18 APP: Button1 multi-press 1
12:24:18 APP: Button1 multi-press 2
12:24:18 APP: Button1 multi-press 3
12:24:19 MQT: stat/tasmota_49A3BC/BUTTON1 = {"ACTION":"TRIPLE"}
12:24:20 APP: Button1 multi-press 1
12:24:20 APP: Button1 multi-press 2
12:24:20 APP: Button1 multi-press 3
12:24:20 APP: Button1 multi-press 4
12:24:21 MQT: stat/tasmota_49A3BC/BUTTON1 = {"ACTION":"QUAD"}
12:24:22 APP: Button1 multi-press 1
12:24:22 APP: Button1 multi-press 2
12:24:22 APP: Button1 multi-press 3
12:24:23 APP: Button1 multi-press 4
12:24:23 APP: Button1 multi-press 5
12:24:23 MQT: stat/tasmota_49A3BC/BUTTON1 = {"ACTION":"PENTA"}
12:24:26 WIF: Checking connection...
12:24:27 APP: Button1 multi-press 1
12:24:27 APP: Button1 multi-press 2
12:24:28 APP: Button1 multi-press 3
12:24:28 APP: Button1 multi-press 4
12:24:29 APP: Button1 multi-press 5
12:24:29 APP: Button1 multi-press 6 <--here will start the wificonfig 2 when So1 is off.
```

tiagofreire-pt, 2020-05-23:
```
17:02:16 APP: Button1 multi-press 1
17:02:17 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"SINGLE"}
17:02:19 APP: Button1 multi-press 1
17:02:19 APP: Button1 multi-press 2
17:02:20 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"DOUBLE"}
17:02:21 APP: Button1 multi-press 1
17:02:21 APP: Button1 multi-press 2
17:02:22 APP: Button1 multi-press 3
17:02:22 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"TRIPLE"}
17:02:23 APP: Button1 multi-press 1
17:02:24 APP: Button1 multi-press 2
17:02:24 APP: Button1 multi-press 3
17:02:24 APP: Button1 multi-press 4
17:02:25 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"QUAD"}
17:02:26 APP: Button1 multi-press 1
17:02:26 APP: Button1 multi-press 2
17:02:26 APP: Button1 multi-press 3
17:02:26 APP: Button1 multi-press 4
17:02:27 APP: Button1 multi-press 5
17:02:27 MQT: stat/candeeiro_sala/BUTTON1 = {"ACTION":"PENTA"}
17:02:27 WIF: Checking connection...
17:02:28 APP: Button1 multi-press 1
17:02:29 APP: Button1 multi-press 2
17:02:29 APP: Button1 multi-press 3
17:02:29 APP: Button1 multi-press 4
17:02:29 APP: Button1 multi-press 5
17:02:29 APP: Button1 multi-press 6
```
Seems to be a problem with the button, at least, with this log option the 6 clicks doesn't trigger a "PENTA" response. Maybe I'm being too fast clicking on it and misses 1 count. :S

effelle, 2020-05-23:
Ok, then. Closing this issue as it has been answered.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-23 by effelle