arendst/Tasmota #8516: Original Sonoff sensor AM2301 (v1.0)

kiralikbeyin, 2020-05-23:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
New original sonoff sensor (v1.0) is different. (not am2301)
Chip is BB1 0 F 2 G A

Found this:
https://electronics.stackexchange.com/questions/443364/itead-sonoff-rf-wifi-433mhz-switch-identifying-an-unknown-chip-on-the-433mhz

https://au.mouser.com/datasheet/2/368/EFM8BB1-DataSheet-958779.pdf

**Is there any way to get temp-hum from new sensor?**

[image]

Old one is blue v0.03 which works - it is really am2301

[image]
[image]

**Describe the solution you'd like**  

**Describe alternatives you've considered**  

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-05-23:
Possibly related here, I've seen users having trouble with what was clearly marked as Sonoff AM2301 but instead had a SI7021 inside. Try configuring as that.

ascillato2, 2020-05-26:
When you have time, please, tell us if using your sensor as SI7021, solves your issue. Thanks

kiralikbeyin, 2020-05-27:
As you see this chip is not a sensor, it is like a atmega attiny which sends internal temp (also hum)

Sonoff uses another kind BB1 chip in RF Bridge for 433MHz reading-sending
@ascillato2 Please do not close the issue.

ascillato2, 2020-05-27:
Have you tried as DHT?
We are asking these tests, because from a dev side, it should be easier for iTead to "emulate" a working protocol rather than making a new one. 

Anyway, if you confirm that this don't work either, that means a new sensor type.

Reopening as requested as a feature request.

kiralikbeyin, 2020-05-27:
We have a new sensor type. This is not DHT.

arendst, 2020-05-27:
To advance this issue pls try all three available single wire devices and report if any works.

The fact that there is a bb1 chip present is a non-issue as this chip is to privide single wire protocol.

ascillato2, 2020-06-03:
@kiralikbeyin 

Hi, were you able to test as Theo has asked?

kiralikbeyin, 2020-06-03:
@arendst @ascillato2 

I bought a normal DS18B20 sensor and i will not use AM2013 SONOFF V1.0 sensor. I tried it some other firmwares and give up.

I am sure someone will ask this again in a few months whic needs a new protocol. Thank you for your interest.

closed 2020-06-03 by kiralikbeyin

ascillato, 2020-06-03:
Ok, in that case, someone having this sensor should reverse engineer the protocol in order to make a new driver.

MrAlyr, 2021-05-07:
I have same pair of sonoff AM2031
V1.0 have SI7021 sensor and working configuration for ESPHOME (sorry %): sensor:
  - platform: dht
    model: SI7021
    pin: GPIO14
    temperature:
      name: "Room Temperature"
    humidity:
      name: "Room Humidity"
    update_interval: 60s

Same 1-wire protocol. No changes. Only explicit model indication.

sfromis, 2021-05-07:
In that case, the reason for it working as SI7021 may be the lie Sonoff tells when the case says AM2301. Same well-known story as with Tasmota. However, for some of the devices, Sonoff seems to have put something else into the case labeled AM2301, in which case it may neither work configured as AM2301 or as SI7021.

In principle, someone having all 3 might be able to create a "Sonoff one-wire" driver able to detect which sensor is at the other end of the wire, and deal with the differences.

mox1, 2024-11-10:
Incase anyone stumbles upon this post trying to get their AM2301 from Sonoff working with a TH16/20. 

I have the same setup as OP. AM2301 was not working. I swapped over to SI7021, and started getting checksum errors:
`17:07:05.235 DHT: Pin14 cycles (80/80)  50 110 51 32 58 32 58 32 58 32 58 32 58 32 58 32 58 108 58 101 ..
17:07:05.237 DHT: Pin14 checksum failure 80D6806BC2 =? 41`

I was able to get it working using SI7021 and the following:
`
CpuFrequency 80
DhtDelay 500,40
`
(I'm pretty sure the CpuFrequency command isn't necessary, but I left it in because its working now).

Log:
`17:07:06.581 CMD: CpuFrequency 80
17:07:06.583 SRC: WebConsole from 192.168.9.9
17:07:06.586 CMD: Grp 0, Cmd 'CPUFREQUENCY', Idx 1, Len 2, Pld 80, Data '80'
17:07:06.591 RSL: RESULT = {"Command":"Unknown"}
17:07:09.221 DHT: Pin14 cycles (80/80)  50 110 52 32 58 32 58 32 58 32 58 32 58 32 58 31 58 109 57 102 ..
17:07:09.223 DHT: Pin14 checksum failure 80D6806BC2 =? 41
17:07:13.224 DHT: Pin14 cycles (80/80)  49 110 52 32 58 32 58 32 58 32 58 32 58 32 58 32 57 109 58 101 ..
17:07:13.226 DHT: Pin14 checksum failure 80D6806BC2 =? 41
17:07:17.228 DHT: Pin14 cycles (80/80)  50 110 52 32 58 32 58 32 58 32 58 32 58 32 58 31 58 109 58 101 ..
17:07:17.230 DHT: Pin14 checksum failure 80D7006BC3 =? C2
17:07:21.249 DHT: Pin14 cycles (80/80)  50 110 51 33 57 33 57 33 57 33 57 33 57 33 57 32 58 108 58 102 ..
17:07:21.252 DHT: Pin14 checksum failure 80D6806BC2 =? 41
17:07:22.520 WIF: Checking connection...
17:07:25.226 DHT: Pin14 cycles (80/80)  50 110 52 32 58 32 58 32 58 32 58 32 58 32 58 31 58 109 57 102 ..
17:07:25.228 DHT: Pin14 checksum failure 80D6806BC2 =? 41
17:07:29.237 DHT: Pin14 cycles (80/80)  50 110 51 33 57 33 57 33 57 33 57 33 57 33 57 32 58 108 58 101 ..
17:07:29.240 DHT: Pin14 checksum failure 80D6806BC2 =? 41
17:07:32.645 HTP: Main Menu
17:07:33.232 DHT: Pin14 cycles (80/80)  52 32 52 32 58 32 58 32 58 32 58 32 58 31 58 109 58 101 58 31 ..
17:07:33.234 DHT: Pin14 read 01AD00D785
17:07:34.232 HTP: Console
17:07:37.210 DHT: Pin14 cycles (80/80)  50 110 52 32 58 32 58 32 58 32 58 32 58 32 58 31 58 109 58 101 ..
17:07:37.212 DHT: Pin14 checksum failure 80D6006BC2 =? C1
17:07:41.218 DHT: Pin14 cycles (80/80)  50 110 52 32 58 32 58 32 58 32 58 32 58 32 58 31 58 109 58 101 ..
17:07:41.220 DHT: Pin14 checksum failure 80D6006BC2 =? C1
17:07:42.528 WIF: Checking connection...
17:07:43.133 CMD: DhtDelay 500,40
17:07:43.135 SRC: WebConsole from 192.168.9.9
17:07:43.138 CMD: Grp 0, Cmd 'DHTDELAY', Idx 1, Len 6, Pld 500, Data '500,40'
17:07:43.143 RSL: RESULT = {"DhtDelay1":[500,40]}
17:07:45.247 DHT: Pin14 cycles (80/80)  52 32 52 32 58 32 58 32 58 32 58 32 58 32 58 108 58 101 58 32 ..
17:07:45.249 DHT: Pin14 read 01AC00D784
17:07:49.246 DHT: Pin14 cycles (80/80)  52 32 52 32 58 32 58 32 58 32 58 32 58 32 58 108 58 101 58 32 ..
17:07:49.248 DHT: Pin14 read 01AC00D784`