arendst/Tasmota #8519: SGP30 Sensor Not Working - Adafruit_SGP30 Lib Needs Updated

peteakalad, 2020-05-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
SGP30 Sensor Not Working - Adafruit_SGP30 Lib Needs Updated

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [X] Tasmota binary firmware version number used: Sensors.bin - 8.3.1 and 7.2.0
  - [X] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VS Code
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
FROM FIXED DEVICE / BUILD SEE BELOW
0:29:41 MQT: stat/tasmota_D851A7/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,6,5,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:29:41 MQT: stat/tasmota_D851A7/RESULT = {"Module":{"0":"Generic"}}
20:29:41 MQT: stat/tasmota_D851A7/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
NO RULES
```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
FROM FIXED DEVICE / BUILD SEE BELOW
0:30:35 MQT: stat/tasmota_D851A7/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_D851A7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:28:23","StartupUTC":"2020-05-23T19:02:12","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-05-23T20:01:53","SaveCount":9,"SaveAddress":"FB000"}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS2 = {"StatusFWR":{"Version":"8.3.1.2(sensors)","BuildDateTime":"2020-05-23T19:59:23","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"354/699"}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BrambleFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS4 = {"StatusMEM":{"ProgramSize":657,"Free":344,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FFAA787","0415A3A5","B7FFBFCD","01DA9BC4","64367CC7","00000072"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,13,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67"}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS5 = {"StatusNET":{"Hostname":"tasmota_D851A7-4519","IPAddress":"192.168.1.45","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"F4:CF:A2:D8:51:A7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D851A7","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS7 = {"StatusTIM":{"UTC":"2020-05-23T19:30:35","Local":"2020-05-23T20:30:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:58","Sunset":"20:35"}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS10 = {"StatusSNS":{"Time":"2020-05-23T20:30:35","SGP30":{"eCO2":400,"TVOC":1}}}
20:30:35 MQT: stat/tasmota_D851A7/STATUS11 = {"StatusSTS":{"Time":"2020-05-23T20:30:35","Uptime":"0T00:28:23","UptimeSec":1703,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BrambleFi","BSSId":"00:1D:AA:12:8C:C8","Channel":1,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Nothing in console.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Buy a new SGP30 sensor and test it.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
SGP30 is read and details shown.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Newly purchased SGP30 sensor is not working with current builds of Tasmota.
* Wemos / Lolin D1 Mini
* Tested on sensors bin for 8.3.1 and 7.2.0
* Generic module configuration with GPIO4 = I2C SDA & GPIO = I2C SCL
* I2CScan could see the device with standard device address 0x58
* No information shown on main tasmota page
To diagnose and confirm the sensor was not broken I tested the SGP30 example sketch in the Arduino IDE. This worked fine.
I noticed the Adafruit_SGP30 library used by Tasmota is 1.0.3 vs 1.2.0 current version in manage libs of arduino ide.
I tested the example sketch with a downgraded version of Adafruit_SGP30 to 1.0.3. The sketch would not work and could not find the sensor.
Having confirmed that the old library version is incompatible/broken with my sensor I compiled a sensors bin of tasmota with Adafruit_SPG30 upgraded to 1.2.0 - the sensor then started working in Tasmota.
TLDR; In summary - the current Adafruit_SGP30 lib referenced in Tasmota is broken for some versions of the SGP30 device. The Adafruit_SGP30 library needs to be upgraded to latest 1.2.0.
SGP30 Sensor Purchased: https://www.aliexpress.com/item/4000575428246.html

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-05-23:
Checking up on the sensor, I noticed that the formatting of the absolute humidity reading in the web interface looks bad. Based on my superficial reading of the driver code, the conversion to the character format for the humidity only happens for JSON, not when putting the reading into the HTML.
https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_21_sgp30.ino#L117

peteakalad, 2020-05-23:
Fixed it altering the USE_WEBSERVER if section copying the JSON setup like so;

```
#ifdef USE_WEBSERVER
    } else {
      WSContentSend_PD(HTTP_SNS_SGP30, sgp.eCO2, sgp.TVOC);
      if (global_update && global_humidity>0 && global_temperature!=9999) {
        // has humidity + temperature
        dtostrfd(sgp30_abshum,4,abs_hum);
        WSContentSend_PD(HTTP_SNS_AHUM, abs_hum);
      }
#endif
    }
```

Jason2866, 2020-05-23:
A PR for updating the lib and the corrected code in the driver is welcome 😊

peteakalad, 2020-05-23:
Sure will see if I get some time..

sfromis, 2020-05-24:
@joselito1 reports a possible mixup of SGP30 and CCS811
https://discord.com/channels/479389167382691863/479389167382691865/714018133962260500

PR #8520 (2020-05-24): SGP30 Sensor Fixes #8519
## Description:
SGP30 Sensor Fixes:
- Upgrade Adafruit_SGP30 library from 1.0.3 to 1.2.0.  v1.0.3 was not working with recently purchased sensors.
- Fix for SGP30 absolute humidty value being blank on web ui but shown in MQTT json.  abs_hum string was only generated inside the JSON if.  Moved this outside this if.

**Related issue (if applicable):** fixes #8519 

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works  [... cut]
commit 43b99960 (2020-05-24): Merge pull request #8520 from peteakalad/peteakalad-SGP30_Sensor_Fixes

SGP30 Sensor Fixes #8519
closed 2020-05-24 by arendst