arendst/Tasmota #8522: Use device without WIFI to turn off after preconfigured PulseTime

sunsero, 2020-05-24:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
no

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
A friend of mine tends to forget to plug off the iron and has to go back to it or calls me to check if it is off. Therefore I want to preconfigure a Gosund SP111 to power off with 'PulseTime 1900' after half an hour. But I want this to work without a WiFi network available and without Tasmota setting up a hot spot if it finds no WiFi. The hotspot / WiFi should just turn on after long pressing the button or pressing the button several times.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Rely on WiFi. However this has several disadvantages:
- unnecessary WiFi waves
- making this solution depend on the router availability (does not work anymore if the router is exchanged)

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-24:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-24 by ascillato2

sunsero, 2020-05-24:
Thanks, I think what I would need is a new command WifiConfig 9:

9 = disable Wi-Fi Manager

so like 5 but without wait until an AP is available again

Then Wifi would be off until I press 6 times the button to start the AP for connecting a different WiFi if required.

Would be also required to be able to change the WiFi config option there in the WiFi configuration page however. Maybe offer WifiConfig in a combobox?

sunsero, 2020-05-24:
If the above is not implemented my best option is probably to:
- use my phone as a mobile hot spot and connect the plug to it
- enter my hot spot as the only AP in the WiFi configuration
- see which IP connected to the hot spot, open console and enter the following commands
- WifiConfig 5 to prevent the plug opening a Hot Spot if it does not see my mobile's hot spot anymore
- SetOption31 1 to disable blinking of the plug while looking for the AP if my mobile's hot spot is off or away. This unfortunately does not disable the permanent scanning for the AP however
- PulseTime 1900 to turn power off automatically after 30 minutes

ascillato, 2020-05-24:
The request for a command to **_turn off wifi_** has been requested before (that is why this issue is tagged as _duplicated_):

https://github.com/arendst/Tasmota/issues/5979
https://github.com/arendst/Tasmota/issues/4330
https://github.com/arendst/Tasmota/issues/6546

Please, check those discussions. As a summary, that feature was never implemented in Tasmota because it is outside of smart-home's objectives. If that device can't share information or be controlled, it isn't part of a smart home.

If you need help on configuring your plug to switch off power after a preconfigured Time ( may be using **_rules_** or `pulsetime` - depending on your setup or desired behaviour ), you can check the [Docs](https://tasmota.github.io/docs/#/) and ask for support on the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4).

If you want a new command to **_turn off wifi_** (or a new `wificonfig` command) please open a **new feature request** linking the above old issues and explaining why you need that and any other general use case you can think of. You need to convince somebody to program that and Theo to merge it into Tasmota. ;)

If you have concerns about wifi health issues, please check https://github.com/arendst/Tasmota/issues/1805

If you want to control your plug but reducing the TX power with a stable connection, please check https://github.com/arendst/Tasmota/issues/4971 and https://github.com/arendst/Tasmota/issues/3609

Thanks.

ascillato, 2020-06-16:
> The request for a command to turn off wifi has been requested before

Now it is available the command WIFI to turn ON and OFF the wifi from the Tasmota Console. Use latest development version.

thomas-lentz, 2021-06-26:
Dear Ascillato,
I now this issue is closed so feel free to move my statement to a better suiting place.
First let me say, that I'm very happy that "turn off WIFI" is now available. But let me also say, that there seems to be some misunderstanding what IoT or SMART home is:
IoT: Internet of things defines, that your thing is connected to some kind of network (e.g. the Internet). It does **not define** that it has **to be connected all the time**. I would say that only a few IoT devices are always on.
A lot of things are "on" just once a week or less. Some IoT devices (e.g using ISM band) are only allowed to be "on" a few times an hour to not block the frequencies. A lot of things are battery powered and need to take care of their energy consumption.
I would say TASMOTA evolved far beyond the initial alternate OS for Sonoff devices (also reflected in the name change of the binaries).
By broadening the device range TASMOTA evolved beyond the grid line powered devices. I observed that the/your arguments also changed from not being IoT if WIFI can be switched off, to: not be in compliance with the "smart-home's objectives".

Maybe one should adjust these objectives, since the smartest homes these days are all about energy saving and not energy wasting. Also not every home is only a house or an apartment with grid power available everywhere. Especially when there is no grid power and no Power over Ethernet you need sometimes to transmit data event based or on a scheduled basis. Having a device draw 110mA all the time, because somebody defines IoT or Smart Home is grid power based with always on WIFI, is not useful or smart!

So maybe one should think about that, when judging future feature requests. I for myself extended the TASMOTA libraries to support MAX17043 for LiPO monitoring for Solar powered TASMOTA devices, but it seems that this does not fit in the current  "smart-home's objectives" of TASMOTA.
No personal offense meant.

Kind regards
 Tom

ascillato, 2021-06-26:
Hi @thomas-lentz 

Thanks for sharing your thoughts.

> IoT ... It does not define that it has to be connected all the time.

Yes, we all agree on that. The definition of smart home is that the devices that are part of the home are interconnected, meaning that they share information. If they connect and disconnects is not contradicting to this as long as some information is being shared with other devices.

For example, If you have an automatic watering system for a garden, but that system is **_isolated_**, it is just an automatic system, not a smart system.

However, it is an smart system if the watering system can:
* Receive information from internet (or from a local weather station) about the weather forecast and self regulate,
* and/or can check if the user is present in the garden so as to not initiate the watering,
* and/or can regulate it's frequency or watering times using the state of the alarm system,
* and/or can share the information of temperature, soil humidity, failures, etc, to the user or to a hub for other actions to be performed,
* and/or any other interchange of information.

> I observed that the/your arguments also changed from not being IoT if WIFI can be switched off, to: not be in compliance with the "smart-home's objectives".

Not really what I wanted to express. I just wanted to make focus on the information interchange. It won't be a smart device a Wi-Fi capable device that has its Wi-Fi turned OFF **_all the time_**, as this issue was about. Check the following statement from the OP:
>I want this to work without a WiFi network available and without Tasmota setting up a hot spot if it finds no WiFi.
That is why, an automatic device was never the focus for Tasmota (meaning we were not going to spend our free volunteer time working on it). Tasmota is aimed to smart devices, meaning with interchange of information and mainly over MQTT. Now, as Tasmota supports the requested command from the OP in this old issue, that is why we post a message in this thread informing that news.

> ... these days are all about energy saving ...

Totally agree. We agree with that from the beginning. That is why, one of the first things we did was to make the Arduino core to support the dynamic sleep function to reduce power consumption while allowing the device to be connected at DTIM times. The actual Tasmota is not ON and not ONLINE all the time. Actually it sleeps and its radio also sleeps. That feature is working fine. In the first Wi-Fi connection, Tasmota talks to the router and they arrange the times where the Wi-Fi Beacon is going to be published from the router and in between Tasmota Sleeps. Those sleeps are very small in order to also be able to be responsive when the user push any button. So, for a smart plug, for example, Tasmota uses as small energy as possible and also it is very responsive from both, your home automation software and from the button. In this type of Setup, the device needs to be Online.

In order cases, like for example a battery powered plant moisture sensor, Tasmota allows you to use DEEPSLEEP. In this other case, being connected all the time is not needed. The user only needs the data from time to time. In this case, The device is not connected to the Wi-Fi Network. It wakes up from time to time to report its data and receive data if needed (MQTT retained commands for example). **_In this case, it is a Smart IoT that is not grid powered, that is not permanently connected to the Wi-Fi Network but is able to send and receive information every deepsleep interval._**

>So maybe one should think about that, when judging future feature requests. I for myself extended the TASMOTA libraries to support MAX17043 for LiPO monitoring for Solar powered TASMOTA devices, but it seems that this does not fit in the current "smart-home's objectives" of TASMOTA.

I think this was a misunderstanding and I did not express myself correctly. Tasmota supports a huge amount of types of setups. It already supports DEEPSLEEP, Some Batery powered devices, turn on and off its Wi-Fi radio, it supports dynamic sleeps and sleeps between Wi-Fi Beacons (and DTIM, meaning to skip some previously arranged amount of beacons), between a lot of features like Rules, Scripts, Berry, Timers, etc etc to make Tasmota behave as you need. It supports MQTT, KNX, Multicast DeviceGroups, Hue and Belkin emulation, MQTT autodiscovery, and several other to interconnect and share information. And much more. It supports to be used in a decentralized system(recommended) and also in a mixed or centralized system (for example all MQTT retained - not recommended and a lot to talk about it). So, it is an amazing firmware for smart home devices. If you want to use as a simple automatic device, you now can too.

So, as Tasmota supports all that, your _extend to allow MAX17043 for LiPO monitoring for Solar powered devices_ is in-line with being a Smart Home Device approach and with Tasmota, so I'm sorry if my old comment have made you to think otherwise.

>No personal offense meant.
No offense taken. In contrast, it is very good to put ideas on the table to polish them and reach a better and more precise understanding. This is the way.

ionultd, 2021-09-07:
@thomas-lentz  Would you be able to share your code for the Tasmota MAX17043 integration. I also want to use it to learn to do my own integrations.

Thank you