arendst/Tasmota #8528: Disabling DST?

WolfganP, 2020-05-24:
### PROBLEM DESCRIPTION
I've just flashed a new device, and while configuring the basics (ie changing my timezone to -3 and entering my lat/lon coordinates), I noticed the DST entries set and I couldn't find any reference in the docs https://tasmota.github.io/docs/Commands/ on how to disable them (my country abandoned DST years ago) and I just didn't want to "disabling" DST effects just by inputing 0s if not documented as workaround.

`12:00:43 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-05-24T15:00:43","Local":"2020-05-24T12:00:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"-03:00","Sunrise":"07:46","Sunset":"17:53"}}`

Any tip?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [*] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [*] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [*] Device used (e.g., Sonoff Basic): Sonoff POW R1
- [*] Tasmota binary firmware version number used: 8.3.1(tasmota)
  - [*] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [*] Flashing tools used: Tasmotizer
- [*] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
11:42:49 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:42:49 RSL: RESULT = {"Module":{"6":"Sonoff Pow"}}
11:42:50 RSL: RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"130":"HLWBL SEL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"132":"HLWBL CF1"},"GPIO14":{"133":"HLW8012 CF"},"GPIO15":{"52":"Led1"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [*] Provide the output of this command: `Status 0`:
```
12:02:13 RSL: STATUS = {"Status":{"Module":6,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1A656D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:02:13 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T01:15:17","StartupUTC":"2020-05-24T13:46:56","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2020-05-23T21:04:12","SaveCount":62,"SaveAddress":"FB000"}}
12:02:13 RSL: STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"409/699"}}
12:02:13 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["homewlesslan","homewlesslan2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00000000"]}}
12:02:13 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:02:13 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_1A656D-1389","IPAddress":"192.168.1.160","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:3A:E8:1A:65:6D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:02:13 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-05-24T15:02:13","Local":"2020-05-24T12:02:13","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"-03:00","Sunrise":"07:46","Sunset":"17:53"}}
12:02:13 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:02:13 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-05-24T12:02:13","ENERGY":{"TotalStartTime":"2020-05-23T21:04:12","Total":0.001,"Yesterday":0.001,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
12:02:13 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-05-24T12:02:13","Uptime":"0T01:15:17","UptimeSec":4517,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"homewlesslan","BSSId":"74:D0:2B:89:33:32","Channel":7,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
DST to be allowed being disabled if no Savings rules exist for the country.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-24:
To enable dst and std times you need to use command timezone 99. As you have set timezone -3, dst is not used.

arendst, 2020-05-24:
As long as timezone is NOT set to 99 it doesn't use DST. SO althought it is displayed it makes no use of it.

arendst, 2020-05-24:
Ah we just crossed.

closed 2020-05-24 by ascillato2