arendst/Tasmota #8533: rule > 511 char not uploaded corrcect via console

riker65, 2020-05-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

riker65, 2020-05-24:
see rule 1 

ending ;publish cmn..."}

rule should be longer

```
on systemboot do
	backlog teleperiod 10;VAR9 57;VAR8 47;MEM1 F_OFF;power0 0; publish tele/WP/%topic%/Act RB
endon 
on Power1#State=0 do
	backlog Power3 0;MEM1 F_OFF;publish tele/WP/%topic%/R1_F F_OFF
endon 
on Power1#State=1 do
	backlog MEM1 F_ON;publish tele/WP/%topic%/R1F_ON F_ON;
endon 
on Power2#State=1 do
	backlog power3 0; publish cmnd/TA_ESP0143/power1 1;publish tele/WP/%topic%/R2t2_1 p1T-St;RuleTimer2 600; 
endon 
on Rules#Timer=2 do
	backlog publish tele/WP/%topic%/R2T20 T2P10 ;power2 0;publish cmnd/TA_ESP0143/power1 0;
endon

so it is truncated

```

riker65, 2020-05-24:
in https://tasmota.github.io/docs/Rules/#rule-syntax
There are three separate rule sets called Rule1, Rule2 and Rule3. Each rule set can contain many rules which are dynamically compressed. Number of rules that can fit in a rule set varies. Expect at least 1000 characters available per rule set.

**can not 1000 chars per set !!!**

ascillato2, 2020-05-25:
From your Log:

```lua
"Length":540,"Free":232
```

That means that you still have space for more rules in the slot `RULE1`.

Compressed rules can't be fully seen in the Tasmota console. The buffer is not enough. That is why you can't write a very long rule at once and must be done by parts using:

`RULE1 ON ....`

and then

`RULE1 + ON .....`

and then

`RULE1 + ON .....`,

and so on, using chunks of less of 511 characters.

riker65, 2020-05-25:
Hi, thanks for that, but how do I verify or download the full rule?

meingraham, 2020-05-25:
From what I saw, @s-hadinger recently added that compressed rules are output to the Console in 500-byte "chunks" so the entire rule set can be inspected. Try a newer development version to get this newer feature.

closed 2020-05-26 by ascillato2