arendst/Tasmota #8534: ESP32-CAM crashing on image request over webUI

blakadder, 2020-05-24:
### PROBLEM DESCRIPTION
tasmota32-camera build crashes on AiThinker ESP32-CAM when requesting image over webUI

```
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400edcf5  PS      : 0x00060530  A0      : 0x800ee1d1  A1      : 0x3ffb1c80  
A2      : 0x00000005  A3      : 0x3ffd9984  A4      : 0x40172878  A5      : 0x3ffc1b9c  
A6      : 0x00000001  A7      : 0x3ffd525c  A8      : 0x800edcf5  A9      : 0x3ffb1c50  
A10     : 0x00000000  A11     : 0x00000005  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00989680  A15     : 0x3ffd5334  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400edcf5:0x3ffb1c80 0x400ee1ce:0x3ffb1d00 0x400ee207:0x3ffb1d20 0x400ee2a5:0x3ffb1d60 0x400deb43:0x3ffb1d80 0x400f8e72:0x3ffb1da0 0x4018b595:0x3ffb1e20 0x40103c32:0x3ffb1e40 0x40103ce9:0x3ffb1e60 0x40103d6a:0x3ffb1e90 0x40103ef3:0x3ffb1ee0 0x400d99af:0x3ffb1f30 0x400e9dfc:0x3ffb1f50 0x400deb43:0x3ffb1f70 0x400ff29f:0x3ffb1f90 0x40112451:0x3ffb1fb0 0x4008f0fd:0x3ffb1fd0
```

device is powered by 5V 2.7A supply

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-CAM
- [x] Tasmota binary firmware version number used: 8.3.1.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PIO
- [x] Flashing tools used: PIO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`: no :P

```
- [ ] Provide the output of this command: `Status 0`:
```
21:13:20 CMD: status 0
21:13:20 MQT: stat/tasmota_2AE930/STATUS = {"Status":{"Module":2,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_2AE930","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software reset CPU","Uptime":"0T00:04:56","StartupUTC":"2020-05-24T19:08:24","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2020-05-24T20:44:51","SaveCount":27}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS2 = {"StatusFWR":{"Version":"8.3.1.2(webcam)","BuildDateTime":"2020-05-24T20:43:56","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","Hardware":"ESP32","CR":"348/699"}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"10.1.1.2","LogPort":514,"SSId":["HTPC","HTPCext"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS4 = {"StatusMEM":{"ProgramSize":1172,"Free":1984,"Heap":111,"PsrMax":4095,"PsrFree":4095,ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000809","8FDAE787","001403A1","000000CD","010013C0","40000181","80000020"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,81","Sensors":"1,2,3,5,6"}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS5 = {"StatusNET":{"Hostname":"tasmota_2AE930-2352","IPAddress":"10.1.1.199","Gateway":"10.1.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.1","Mac":"24:0A:C4:2A:E9:30","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2AE930","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS7 = {"StatusTIM":{"UTC":"2020-05-24T19:13:20","Local":"2020-05-24T21:13:20","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:04","Sunset":"20:17"}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS10 = {"StatusSNS":{"Time":"2020-05-24T21:13:20"}}
21:13:20 MQT: stat/tasmota_2AE930/STATUS11 = {"StatusSTS":{"Time":"2020-05-24T21:13:20","Uptime":"0T00:04:56","UptimeSec":296,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"HTPCext","BSSId":"40:31:3C:93:F0:15","Channel":11,"RSSI":100,"Signal":-23,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400edcf5  PS      : 0x00060530  A0      : 0x800ee1d1  A1      : 0x3ffb1c80  
A2      : 0x00000005  A3      : 0x3ffd9984  A4      : 0x40172878  A5      : 0x3ffc1b9c  
A6      : 0x00000001  A7      : 0x3ffd525c  A8      : 0x800edcf5  A9      : 0x3ffb1c50  
A10     : 0x00000000  A11     : 0x00000005  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00989680  A15     : 0x3ffd5334  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400edcf5:0x3ffb1c80 0x400ee1ce:0x3ffb1d00 0x400ee207:0x3ffb1d20 0x400ee2a5:0x3ffb1d60 0x400deb43:0x3ffb1d80 0x400f8e72:0x3ffb1da0 0x4018b595:0x3ffb1e20 0x40103c32:0x3ffb1e40 0x40103ce9:0x3ffb1e60 0x40103d6a:0x3ffb1e90 0x40103ef3:0x3ffb1ee0 0x400d99af:0x3ffb1f30 0x400e9dfc:0x3ffb1f50 0x400deb43:0x3ffb1f70 0x400ff29f:0x3ffb1f90 0x40112451:0x3ffb1fb0 0x4008f0fd:0x3ffb1fd0

```

### TO REPRODUCE
try to access webUI

### EXPECTED BEHAVIOUR
not crashing :D

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-05-25:
please set the log level to debug to see the logs of the cams init routine

arendst, 2020-05-25:
Does it always fail? 

I have a bad camera which fails in the same way. Swapped with a functional one solved it. So I ordered new cameras waiting to arrive.

arendst, 2020-05-25:
It's a bug in the ESP32 webcam core code which fails to do checks resulting in this.

It's not only Tasmota that is alpha, it's espressif (the supplier of the webcam code mind you) too.

ascillato2, 2020-05-26:
@blakadder 

Hi, can you test latest dev version again? Theo have pushed a fix. Thanks

blakadder, 2020-05-26:
doesn't exception anymore... camera still not operational

seriallog 4 output:
```
16:56:16 HTP: Main Menu
16:56:16 CAM: Stream init
16:56:16 CAM: Default template
16:56:16 CAM: PSRAM found
[E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time!
16:56:21 CAM: Init failed to get the frame on time
```

gemu2015, 2020-05-26:
you are using the default template. this is Ai Thinker Cam
Do you have that Model ?
Otherwise you must specify the camera pins of your Model in Tasmota IOPin GUI.

blakadder, 2020-05-26:
Its using model 2 and I have the AiThinker ESP32-CAM device

closed 2020-05-26 by blakadder

teixeluis, 2020-09-19:
I'm also getting the same behavior but on a specific camera, which according to the seller it is also a OV2640 (described as "TTGO Camera Module OV2640").

With the OV2640 camera that came with the ESP32-CAM board, it runs well.

Similarly, I get the following log output:

```
18:56:21 HTP: Main Menu
18:56:21 CAM: Stream init
18:56:21 CAM: Default template
18:56:21 CAM: PSRAM found
18:56:26 CAM: Init failed to get the frame on time
18:56:26 QPC: Reset
18:56:27 HTP: Console
18:56:28 APP: Boot Count 6
```
[image]

(the camera that fails is the connected one)

Can it be the case that different pin assignments are needed, or different i2c addresses are being used by this particular camera?

If I run a scan, I get the error:

```
19:12:59 CMD: Group 0, Index 1, Command "I2CSCAN", Data ""
19:12:59 RSL: stat/tasmota_008F98/RESULT = {"Command":"Error"}
```

Thank you

Cheers

teixeluis, 2020-09-19:
Btw, the original (smaller) camera works with the module "ESP32 Cam AiThinker" setting. 

I tried the 3 settings, and the behavior varies between the two log outputs:

behavior 1:

```
19:22:37 HTP: Main Menu
19:22:37 CAM: Stream init
19:22:37 CAM: Default template
19:22:37 CAM: PSRAM found
19:22:41 CAM: Init failed to get the frame on time
19:22:42 HTP: Main Menu
19:22:43 QPC: Reset
19:22:43 HTP: Configuration
```

behavior 2:

```
19:20:48 HTP: Main Menu
19:20:48 CAM: Handle camserver
19:20:48 CAM: Create client
19:20:48 CAM: Start stream
19:20:49 HTP: Console
19:20:49 CAM: Client fail
19:20:49 CAM: Stream exit
19:20:52 WIF: Checking connection...
```
and also, the page shows a broken link to the stream:

[image]

teixeluis, 2020-09-19:
Yet another observation, with the working camera, the 3 different module settings all work.

teixeluis, 2020-09-19:
I have further attempted another test, and set with the template you provided here:

https://tasmota.github.io/docs/ESP32/#lilygo-ttgo-t-camera-ov2640_v05

and in this case I get a slightly different error:

```
20:23:10 CAM: Stream init
20:23:10 CAM: User template
20:23:10 CAM: PSRAM found
20:23:10 CAM: Init failed with error 0x20004
20:23:12 QPC: Reset
20:23:13 HTP: Configuration
20:23:14 HTP: Configure Module
```

The template data, after a readout:

```
20:23:30 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:23:30 RSL: stat/tasmota_008F98/RESULT = {"NAME":"TTGO_V05","GPIO":[1,1,1,1,5090,5088,1,1,5056,5024,5089,5091,1,1,5092,5184,0,640,608,5093,0,5152,4928,5120,0,0,0,0,4992,160,65,1,5094,0,0,5095],"FLAG":0,"BASE":2}
```

DanJunior78, 2021-01-07:
Same error here. I get also with the LilyGO TTGO T-Camera OV2640_V05 config same error... CAM: Init failed with error 0x20004. I have a TTGO Camera

DanJunior78, 2021-01-07:
Ok, there is a mistake in the template. GPIO35 in the docs is defined as USER...however it is not available on the TTGO Camera. I´ve changed to NONE and the cam was running.