arendst/Tasmota #8536: %04x in topic not working when publishing from Rule

martinren87, 2020-05-24:
### PROBLEM DESCRIPTION

Having configured the MQTT Topic as:
TOPIC: Tasmota/%04x

I created the following rule:
RULE1: ON Var1#State DO Publish %topic%/Var1 %value% ENDON

I was hoping to get the value published this way: 
TOPIC: Tasmota/603978/Var1 100

But instead I am getting this other way: 
TOPIC: Tasmota/%04x/Var1 100 

I guess the rule is not replacing %04d with the last digits of the MAC address. Is this intended to work this way, or it is a bug? 
In my case I need to publish this variable as any other in the same root topic that has the format "TASMOTA/MACADDRESS/VARIABLE"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Node MCU
- [x] Tasmota binary firmware version number used: 8.2.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Gitpod
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:50:51 CMD: Backlog Template; Module; GPIO 255
22:50:51 MQT: Tasmota/603978/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:50:51 MQT: Tasmota/603978/stat/RESULT = {"Module":{"0":"Generic"}}
22:50:52 MQT: Tasmota/603978/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"186":"Slave RST"},"GPIO3":{"0":"None"},"GPIO4":{"185":"Slave RX"},"GPIO5":{"184":"Slave TX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"7":"WS2812"},"GPIO13":{"0":"None"},"GPIO14":{"37":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
22:51:35 MQT: Tasmota/603978/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":363,"Rules":"ON Var1#State DO Publish %topic%/Var1 %value% ENDON"}
22:51:35 MQT: Tasmota/603978/stat/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":300,"Rules":"ON mqtt#disconnected DO color 255,255,0 ENDON ON mqtt#connected DO color 0,255,0 ENDON ON wifi#disconnected DO color 255,0,0 ENDON ON wifi#connected DO color 255,255,0 ENDON ON Power1#Boot DO color 255,0,0 ENDON"}
22:51:35 MQT: Tasmota/603978/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":428,"Rules":"ON Dimmer#State DO Publish Tasmota/603978/Dimmer %value% ENDON"}

```
- [ ] Provide the output of this command: `Status 0`:
```
22:55:12 MQT: Tasmota/603978/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"Tasmota/603978","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:55:12 MQT: Tasmota/603978/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:04:50","StartupUTC":"2020-05-24T21:50:22","Sleep":50,"CfgHolder":4617,"BootCount":35,"BCResetTime":"2020-05-09T21:12:05","SaveCount":502,"SaveAddress":"FB000"}}
22:55:12 MQT: Tasmota/603978/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(tasmota)","BuildDateTime":"2020-04-16T23:53:28","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"376/699"}}
22:55:12 MQT: Tasmota/603978/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":[" xxxxxxxxxxx ",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C8000181060000005A00000000000000","00000000","00000000"]}}
22:55:12 MQT: Tasmota/603978/stat/STATUS4 = {"StatusMEM":{"ProgramSize":585,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,31,35,37","Sensors":"1,2,3,4,5,6"}}
22:55:12 MQT: Tasmota/603978/stat/STATUS5 = {"StatusNET":{"Hostname":"Tasmota/60","IPAddress":"192.168.168.135","Gateway":"192.168.168.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.168.1","Mac":"B4:E6:2D:60:39:78","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:55:12 MQT: Tasmota/603978/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_603978","MqttUser":"xxxxxxxxxxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:55:12 MQT: Tasmota/603978/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-05-24T21:55:12","Local":"2020-05-24T22:55:12","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:57","Sunset":"20:36"}}
22:55:12 MQT: Tasmota/603978/stat/STATUS10 = {"StatusSNS":{"Time":"2020-05-24T22:55:12"}}
22:55:12 MQT: Tasmota/603978/stat/STATUS11 = {"StatusSTS":{"Time":"2020-05-24T22:55:12","Uptime":"0T00:04:50","UptimeSec":290,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"0,255,0","HSBColor":"120,100,100","Channel":[0,100,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":" xxxxxxxxxxx ","BSSId":"xxxxxxxxxxx","Channel":2,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

22:59:54 CMD: var1 20
22:59:54 MQT: Tasmota/603978/stat/RESULT = {"Var1":"20"}
22:59:54 RUL: VAR1#STATE performs "SlaveSend 20"
22:59:54 MQT: Tasmota/603978/stat/RESULT = {"SlaveSend":"Done"}
22:59:54 RUL: VAR1#STATE performs "Publish Tasmota/%04x/Var1 20"

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Change VAR1 value 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expected to be published as:

22:59:54 RUL: VAR1#STATE performs "Publish Tasmota/603978/Var1 20"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

martinren87, 2020-06-12:
Hi guys... any news about this? Is it possible to solve this problem?

ascillato, 2020-06-12:
Hi, so far it is not supported, that is why it has the label **feature request** to see if there is anyone interested on working on this to add this new variable.

Instead of `%04x` it should be a new rule's variable like `%mac%` in order to be more easy to read.

If you want, you can check how other variables are programmed (like %topic% for example) in the file https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_10_rules.ino and make a new one for the mac. And then, if you want, you can make a Pull Request. If you need help on any part of this, please just ask.

ascillato2, 2020-06-14:
@martinren87 

Hi, are you interested on working on this?

martinren87, 2020-06-15:
Hi Adrian. I am not an expert, but I´m interested, so I will do my best.

ascillato2, 2020-06-19:
Hi,

After checking the code, your assumption was right. This was a Bug.

Now it is fixed on PR #8731 

Thanks for reporting!!! :+1:

closed 2020-06-19 by ascillato2

martinren87, 2020-06-19:
Thank you Adrian! 

Regards,

PR #8731 (2020-06-19): Fix bug on rule variable %topic%
## Description:

**Related issue (if applicable):** fixes #8536

As explained in the referenced issue, if having a topic as:
TOPIC: `Tasmota/%04x`

And if a rule is created like the following:
RULE1: `ON Var1#State DO Publish %topic%/Var1 %value% ENDON`

We expect to have as result a MQTT message published to the topic:
`Tasmota/3978/Var1`

But instead we were getting this:
`Tasmota/%04x/Var1`

The `%topic%` variable didn't have the replacement of the mac last digits on %04x. This bug was because the rules' driver was using an incorrect internal topic variable  [... cut]