arendst/Tasmota #8537: Add a option for Millisecond timestamping?

multicron, 2020-05-25:
**Have you looked for this feature in other issues and in the docs?**  
Yes, and in the code.

**Is your feature request related to a problem? Please describe.**  
I have a remote sensor on a electical power meter that sends an MQTT message every time one Watt-hour of energy is used.  By accurately measuring the time between pulses, I can know the instantaneous power consumption value.  The problem is the most granular timestamping I can get in a rule appears to be 1 second.  I could have much more accurate results if I could have a timestamp with millisecond accuracy.

**Describe the solution you'd like**  
A SetOption to allow the default timestamps to have milliseconds added like so:  2020-05-25T03:23:18.432 or a Rule %timestamp%-like variable that provides time to the millisecond.

**Describe alternatives you've considered**  
I've searched the code and lots of documentation but I can't find anywhere in Tasmota where a timestamp with finer than 1 second granularity is available.

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-05-25:
To get the most out of your millis display you'll need to stop sleep as otherwise you'll still have a granularity of the sleep period.

multicron, 2020-05-25:
Wow!  That's super cool!  And super fast!  Thank you Theo!

multicron, 2020-05-31:
Sorry to be a pest, but I noticed a problem with this.  The milliseconds need to be zero-padded on the left to three digits.  Currently it outputs 1 millisecond returned from RtcMillis() as ".1" instead of ".001", which parses as ".100" when run through a date parsing function.

multicron, 2020-06-01:
I went to create a pull request for my fix for this, and I noticed that you already fixed it (soon after the original enhancement commit).  Thank you again for all your work!

closed 2020-06-01 by multicron

thavincinew, 2023-08-14:
I do realize the topic is closed, however I hope someone could point me in the direction to have ms timing in other areas other than just the console. 
As an example I would like a rule that can use %timestamp% , that must include ms.

sfromis, 2023-08-14:
With rules you basically have no feature for ms timing in general. In a few cases, you can play tricks with things like the Backlog inter-command delay to try to control specific delay timing, or 1/10 second timing tricks via PulseTime or BlinkTime, but such depends on you what you specifically try to achieve.

With ESP32 and the Berry language, you can handle ms resolution, but still not real time programming with time-critical exact controls.

thavincinew, 2023-08-15:
Thank you very much for the response! So essentially I am doing a Webquery on a contact close to log the event and needed the value of the event to ms.
It is a custom API i am calling so cannot use HA/MQTT/Syslog or anything that is a "external" system.
I did think of Berry as it is a ESP32, but too complex for me at the moment.

Would you perhaps know if i could simply expand the %timestamp% variable via the code and a recompile, and if so where I could find the section to modify? (I am comfortable building my own firmwares, however not sure how viable this approach would be for me)

sfromis, 2023-08-15:
I'd say that ESP32 + Berry would be the less complex way for millisecond processing....

If you want to do nothing with the time stamp in rules, other than getting a string, something like this can work:
`rule1 on Event#test do Backlog Time endon on Time do var1 %value% endon`
(Obviously, the Event trigger is merely for testing, not part of the technique)

`%timestamp%` is expanded [here](https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_10_rules.ino#L479).

thavincinew, 2023-08-15:
fromis: I cannot thank you enough!
The following worked perfectly as a POC. (have removed sensitive info)

Rule1 ON Power1#state=1 do Backlog0 Time endon on Time do WebQuery https://api.telegram.org/botxxxxxxxxx:xxxxxxxxxxxxxxxxx/sendMessage?chat_id=xxxxxxxxx&text=%value% GET ENDON

sfromis, 2023-08-15:
Beware that the trigger for `Time` may also fire at other points in time than just the command.