arendst/Tasmota #8538: PowerCal argument should not be limited to 32000

marcelm, 2020-05-25:
### PROBLEM DESCRIPTION

Summary: My smart plug needs `PowerCal 63770` in order to display correct power measurements, but the value of the argument is restricted to 32000.

I have a smart plug with energy metering function, which I calibrated using the `PowerSet`, `VoltageSet` and `CurrentSet` commands. Then I ran the commands `PowerCal`, `VoltageCal` and `CurrentCal` and noted down the reported values in order to be able to easily restore the calibration. The value reported for `PowerCal` was 63770. Then I did a `Reset 5` (for an unrelated reason), and when trying to restore the calibration, the command `PowerCal 63770` did not actually change the power calibration offset/coefficient.

I realize that the `PowerCal` command is documented to accept only values up to 32000, so this is an intentional restriction. However, it appears to me that it least for my device, this limitation is too low and should be increased.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Nedis WIFIP110
- [X] Tasmota binary firmware version number used: 8.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tuya-Convert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
{"NAME":"Nedis WIFIP110","GPIO":[17,0,0,0,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":49}
{"Module":{"0":"Nedis WIFIP110"}}
{"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"56":"Led1i"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
stat/xyz/STATUS = {"Status":{"Module":0,"xyz":"Xyz","FriendlyName":["Xyz"],"Topic":"xyz","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
stat/xyz/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T18:52:48","StartupUTC":"2020-05-24T12:45:28","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-05-24T13:30:46","SaveCount":57,"SaveAddress":"FB000"}}
stat/xyz/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"346/699"}}
stat/xyz/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xyz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
stat/xyz/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
stat/xyz/STATUS5 = {"StatusNET":{"Hostname":"xyz","IPAddress":"xyz","Gateway":"xyz","Subnetmask":"255.255.255.0","DNSServer":"xyz","Mac":"xyz","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
stat/xyz/STATUS6 = {"StatusMQT":{"MqttHost":"xyz","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_58C6A5","MqttUser":"DVES_USER","MqttCount":26,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
stat/xyz/STATUS7 = {"StatusTIM":{"UTC":"2020-05-25T07:38:17","Local":"2020-05-25T08:38:17","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"xy:z","Sunset":"xy:z"}}
stat/xyz/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
stat/xyz/STATUS10 = {"StatusSNS":{"Time":"2020-05-25T08:38:17","ENERGY":{"TotalStartTime":"2020-05-24T13:30:46","Total":0.012,"Yesterday":0.009,"Today":0.002,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
stat/xyz/STATUS11 = {"StatusSTS":{"Time":"2020-05-25T08:38:17","Uptime":"0T18:52:49","UptimeSec":67969,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":26,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xyz","BSSId":"xyz","Channel":10,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
08:43:01 CMD: PowerCal 63770
08:43:01 SRC: WebConsole from xyz
08:43:01 CMD: Group 0, Index 1, Command "POWERCAL", Data "63770"
08:43:01 MQT: stat/pump2/RESULT = {"PowerCal":65096}
```
(I got the current value of 65096 by using `PowerSet`, not with `PowerCal`)

### TO REPRODUCE

Try to run `PowerCal` with an argument above 32000

### EXPECTED BEHAVIOUR

PowerCal should accept values above 32000

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-25:
Yes it is in driver xdrv_03_energy.ino
```
void CmndPowerCal(void)
{
  Energy.command_code = CMND_POWERCAL;
  if (XnrgCall(FUNC_COMMAND)) {  // microseconds
    if ((XdrvMailbox.payload > 999) && (XdrvMailbox.payload < 32001)) {
      Settings.energy_power_calibration = XdrvMailbox.payload;
    }
    ResponseCmndNumber(Settings.energy_power_calibration);
  }
}
```

arendst, 2020-05-25:
And it is there because the max proper positive integer input value is 32767.

Supporting larger values would mean redesign of input values. It's better to change the offset values in your case. Again, I will need to redesign things here too.

arendst, 2020-05-25:
You might want to change GPIO04 from 134 to 133 and see what happens. This uses another offset range which might be better suited for your device.

marcelm, 2020-05-25:
Cool, this works! Just changing GPIO04 to 133 does indeed let me use more sensible values way below 32000. Thanks for this!

Just for completeness: Do I understand correctly that the BL0937 (134) and HLW8012 (133) chips actually are somewhat compatible with each other except for the offset ranges?

arendst, 2020-05-25:
Correct

marcelm, 2020-05-25:
Thanks again for the help. My problem is fixed, and I have submitted a documentation PR to possibly help others avoid it, and also a PR to fix the device template in the template repository.

I would now close the issue, but since you mentioned you would need to do some redesign, which you may want to track by using this issue, I’ll leave that to you.

arendst, 2020-05-25:
Thx. No redesign needed ;-)

closed 2020-05-25 by arendst