arendst/Tasmota #8539: update problem with TasmoAdmin

e9hack, 2020-05-25:
### PROBLEM DESCRIPTION
After commit a5130fa2e46825a1eeaa99a0e25159ce8b1220b5, update with TasmoAdmin does not longer work. The device with tasmota-minimal isn't able to connect to WiFi. Over the serial connection, I get the following debug output from the device:
```
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld
00:00:00 CFG: Loaded from flash at F6, Count 21996098
00:00:00 CFG: Text overflow by 295 char(s)
00:00:00 Project esp8266-test-board-1 Œ`
                                        "L
                                          F Version 8.3.1.2(minimal)-2_6_1
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP6 rB      rB
Â RRBÂ@ &
        r  Channel 133 BSSId 90:86:91:87:92:88 in mode 11N as B¢¬ÐõÀ...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
--- snip ---
00:00:08 APP: Boot Count 1537
00:00:09 WIF: Checking connection...
00:00:09 WIF: Connect failed as AP cannot be reached
00:00:09 WIF: Connecting to AP5 B¢¬ÐõÀ in mode 11N as B¢¬ÐõÀ...
00:00:10 WIF: Checking connection...
00:00:10 WIF: Attempting connection...
--- snip ---
00:00:18 WIF: Connect failed as AP cannot be reached
```
If I revert the changes from commit a5130fa2e46825a1eeaa99a0e25159ce8b1220b5 in support_tasmota.ino in tasmota-minimal only, update with TasmoAdmin works again. 

If I got this issue already and try to flash a reverted tasmota-minimal over the serial connection, it doesn't help. I must flash tasmota-sensor (maybe any other tasmota-xxxx with exception of tasmota-minimal) to get the device working again.

If I do the update directly with a browser and the embedded web server by flashing tasmota-minimal first and tasmota-sensor as second, I don't see this issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP12E
- [x] Tasmota binary firmware version number used: 8.3.1.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Visual Studio Code/PlatformIO
- [x] Flashing tools used: TasmoAdmin and esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-05-25:
- Your settings are corrupt as reported by the strange characters displayd and the message ``CFG: Text overflow by 295 char(s)``
- Your minimal version is too old as it uses a not supported core version of 2.6.1

As noted in the release notes Tasmota 8.3.1.x only supports core 2.7.1 and up.

e9hack, 2020-05-25:
>As noted in the release notes Tasmota 8.3.1.x only supports core 2.7.1 and up.

I forgot to update platformio_override.ini

closed 2020-05-25 by e9hack