arendst/Tasmota #8546: Measure AC signals (Vpp, Vrms, ...) on analog inputs or via ADC1115?

jensb, 2020-05-26:
**Have you looked for this feature in other issues and in the docs?**  
Yes. Google, Github, FAQ, Wiki, forums, StackOverflow.

**Is your feature request related to a problem? Please describe.**  
I use an ESP8266+ADC1115 to measure four voltages which are generated using contactless current clamps SCT-013-0x0 (see e.g. here: https://www.ebay.de/itm/232468468235). Of course these generate an AC signal. I could now extend my PCB with four active rectifiers (passive rectifiers with diodes have too much losses, tried that) but that would blow the whole PCB design out of proportion. But I just need Vpp since I know it's a sine wave at f=50Hz, and then convert that to Vrms.

**Describe the solution you'd like**  
1) Let the ADC1115 multisample each input for a configurable time (for f=50Hz, at least 20ms) and take the delta between largest and smallest value as Vpp. No rectifier needed. The ADC1115 can collect >800 samples per second which is more than enough.
2) Allow simple calculations and a user-configured unit to the figures collected by the ADC1115 to convert Vpp to Vrms, assuming a frequency and a sine wave.

**Describe alternatives you've considered**  
4 hardware based rectifiers require *at* *least* 16 additional elements on the PCB, introduce additional errors, and even then don't give you Vrms.

**Additional context**  
None so far.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-26:
Github issues are meant for Tasmota Software Bug Reporting. 
It is not a forum nor a place to ask general electronic questions

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

jensb, 2020-05-26:
I chose "Feature Request", not "Bug Report".
Is this not a feature request?

Frogmore42, 2020-06-12:
Take a look at the datasheet for the ADS115. Section 7.3 Recommended Operating Conditions
                                                   | MIN  | NOM | MAX | UNIT
V(AINx) Absolute input voltage | GND |           | VDD  | V

This means that the minimum voltage that can be on the analog input is 0V.  The absolute maximum in section 7.1 is -0.3V.  More than that will be shunted by the input diodes and if there is too much current it will blow the diode.

This means, at a minimum, there needs to be additional circuitry to bias the AC signal to a range that is valid for the input circuitry.

Doing that would at least keep the ADS1115 from getting destroyed by the input signal.

Assuming the software were changed to get all 860 samples/second on one channel, that would give 860/50=17.2 samples per cycle at 50Hz.  That is nowhere near an enough to get an accurate value for RMS voltage.

If you want to measure power, get one of the esp8266 devices that have a dedicated power measurement circuit.  The esp8266 is capable of measuring current accurately, but it needs a capable ADC to do it (and sophisticated software that is optimized for that task).
Check out this open source product:
http://iotawatt.com/

For more on the theory of this you can look here (they also have Arduino powered devices: 
https://learn.openenergymonitor.org/electricity-monitoring/ac-power-theory/arduino-maths

So, while this is a feature request, it is unlikely to ever be implemented in Tasmota.  It is not something simple to accomplish and is not really in harmony with the goals of Tasmota.

ascillato2, 2020-06-12:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-12 by ascillato2

jensb, 2020-06-12:
I expected this much after some more research. So Tasmota is probably not the tight tool for this special task. I started this primarily to gain some experience, not because I needed a polished end user product (like iotawatt), and so I have already implemented my own firmware.
Thank you!