arendst/Tasmota #8550: RF pairing not working with 4ch pro

daniel-lo, 2020-05-26:
### PROBLEM DESCRIPTION
8.3.1.2 (also with 8.3.1.1)

I have huge probs pairing my rf remote .. with my 4ch pro rev2. I double tap the physical button and nothing happens.
Also it seems that expecially Button 1 has problems when manually pressed .. 2-4 seem to work better. Via Web-IF everything works well.

Is there a limitation related to the RF use in newer tasmota version?
Is there a way to trigger RF pairing via console?

BR

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
12:09:51 CMD: status 0
12:09:51 MQT: stat/sonoff/STATUS = {"Status":{"Module":23,"DeviceName":"sonoff-garten1","FriendlyName":["sonoff-garten1","sonoff-garten2","sonoff-garten3","sonoff-garten4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:09:51 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:25:23","StartupUTC":"2020-05-26T09:44:28","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-05-05T16:39:37","SaveCount":54,"SaveAddress":"F9000"}}
12:09:51 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"8.3.1.2(9a7f252-tasmota)","BuildDateTime":"2020-05-26T10:00:47","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"412/699"}}
12:09:51 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
12:09:51 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:09:51 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-garten","IPAddress":"192.168.178.68","Gateway":"192.168.178.1","Subnetmask":"255.255.0.0","DNSServer":"192.168.178.1","Mac":"2C:F4:32:A7:F3:EA","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:09:51 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.115","MqttPort":1883,"MqttClientMask":"sonoff-garten","MqttClient":"sonoff-garten","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:09:51 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"2020-05-26T10:09:51","Local":"2020-05-26T12:09:51","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:55","Sunset":"21:38"}}
12:09:51 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2020-05-26T12:09:51"}}
12:09:51 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2020-05-26T12:09:51","Uptime":"0T00:25:23","UptimeSec":1523,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box","BSSId":"24:65:11:C8:5C:3E","Channel":1,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-05-26:
Tasmota is in no way involved in RF pairing.

ascillato2, 2020-05-26:
Closing this issue as it has been answered.
 
If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-26 by ascillato2