arendst/Tasmota #8556: WS2812 scheme settings does not honor SetOption20

reimerp, 2020-05-26:
**Have you looked for this feature in other issues and in the docs?**  
yes, but nothing found

**Is your feature request related to a problem? Please describe.**  
Setting Option20 to ON will not Power the device when a Color command is sent.
Tihs is ignored for a Scheme Command.
Steps to reproduce:
```
SO20 ON;
Color 1; 
Scheme 7;
Power;
{ POWER: ON }
``` 
**Describe the solution you'd like**  
The Device should stay OFF,
```
SO20 ON;
Color 1; 
Scheme 7;
Power;
{ POWER: OFF }
``` 

**Describe alternatives you've considered**  
It is not documented that Option20 effects Scheme changes 

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-27:
Hi,

Color command don't have any option over Scheme 7 that is a RGB pattern on  WS2812B led strip

If what you wanted is to set a color without turning the led strip on, you need to use COLOR2 command instead of COLOR command.

ascillato2, 2020-05-27:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-27 by ascillato2

reimerp, 2020-05-27:
Maybe you misundersttod:

I don't wand the device to be powered on if i issue a scheme command when Option20 is set

ascillato, 2020-05-27:
Setoption20 is not related to ws2812b as docs states. 

Making setoption20 to affect also ws2812 can brake some backwards compatibility with users that have already integrations with home automation softwares.

@arendst @effelle what do you think?

effelle, 2020-05-27:
Indeed the scheme is not vinculated to `SetOption20`. 
`Settings.flag.not_power_linked` uses the `light_state` to determine if an ON state can be omitted and I'm not sure if it will break something on WS setup.  @s-hadinger would probably be more helpful than me in this case.
Now the real problem is the HA integration: when SetOption19 is enabled it takes SetOption20 into account, but not as we intended.
If we look to the definition for `on_command_type` on HA docs:

```
Defines when on the payload_on is sent. Using last (the default) 
will send any style (brightness, color, etc) topics first and then a 
payload_on to the command_topic. Using first will send the 
payload_on and then any style topics. Using brightness will only
send brightness commands instead of the payload_on to turn the light on.
```
 
Now, on Discovery (L266) we have:
`strncpy_P(stemp3, Settings.flag.not_power_linked ? PSTR("last") : PSTR("brightness"),
 sizeof(stemp3));`
Those are the only two types we use, where `last` is used when SetOption20 is enabled.
In fact I don't see an option to let Home Assistant know I don't want to lit a light when I change the brightness. @emontnemery any idea?

emontnemery, 2020-05-27:
>  I don't see an option to let Home Assistant know I don't want to lit a light when I change the brightness. @emontnemery any idea

No, it's not possible.

effelle, 2020-05-28:
Thanks for the confirmation Erik.
Adrian in resume even if we add scheme to Setoption20 we will still have an `ON` state when changing Brightness (the only slider available when a light is off) from HAss. 
For me the change is not necessary and I suggest to let SetOption20 stay as it is.