arendst/Tasmota #8560: Sonoff Dual button not work

AlexTransit, 2020-05-27:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Flash last release. 8.3.0 - button not work
I remember what worked. made a downgrade
7.1 - not work
6.7 - not work
6.4.1 - worked (1 press toggle 1 relay, 2 press toggle 2 relay)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Dual R1
- [x ] Tasmota binary firmware version number used: 8.3.0.1
  - [x ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: platform io/arduino
- [x ] Flashing tools used: OTA/Serial
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  17:49:08 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:49:08 RSL: RESULT = {"Module":{"5":"Sonoff Dual"}}
17:49:08 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"4":"DS18x20"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
17:50:59 RSL: STATUS = {"Status":{"Module":5,"FriendlyName":["bfloor","bfloor1"],"Topic":"bathroom-floor","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:50:59 RSL: STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://mqtt.local/sonoff-6-4-1.bin","RestartReason":"Software/System restart","Uptime":"10T06:37:02","StartupUTC":"2020-05-17T06:13:57","Sleep":50,"CfgHolder":4617,"BootCount":52,"BCResetTime":"2020-05-17T11:14:07","SaveCount":202,"SaveAddress":"F7000"}}
17:50:59 RSL: STATUS2 = {"StatusFWR":{"Version":"8.3.0.1(tasmota)","BuildDateTime":"2020-05-16T10:11:22","Boot":4,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"345/699"}}
17:50:59 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"adn","LogPort":514,"SSId":["AlexHome","AlexHomeR"],"TelePeriod":120,"Resolution":"558180C0","SetOption":["04008000","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
17:50:59 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","05802782","041883A1","00000045","010000C0","00004000","00000420"],"Drivers":"1,2,3,4,8,9,10,16,20,21,39","Sensors":"1,2,3,5"}}
17:50:59 RSL: STATUS5 = {"StatusNET":{"Hostname":"bfloor","IPAddress":"192.168.1.113","Gateway":"192.168.1.3","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.3","Mac":"5C:CF:7F:E7:5E:87","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
17:50:59 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-05-27T12:50:59","Local":"2020-05-27T17:50:59","StartDST":"2020-03-30T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+05:00"}}
17:50:59 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-05-27T17:50:59","DS18B20":{"Id":"00000851A45C","Temperature":31.1},"TempUnit":"C"}}
17:50:59 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-05-27T17:50:59","Uptime":"10T06:37:02","UptimeSec":887822,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"AlexHome","BSSId":"B8:69:F4:B1:94:DE","Channel":6,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
button not work if firmware >6.4.1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
doesn't react when i press the button.
if press internal separated button (inside case)
relay switched. in log no data. in web page no change.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-27:
Hi,

Using latest Tasmota version, please, could you share the output of the console when pushing the button on 1 press and also on 2 presses? Please, set weblog to 4 before the test.

AlexTransit, 2020-05-27:
```STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,```
When you press the button in the log does not write anything.
If you set version 6.4.1 and press the button, then the log writes
```
08:57:13 APP: Button code 0404
08:57:13 APP: Button1 multi-press 1
```
I wrote about it
https://discordapp.com/channels/479389167382691863/490244847568158751/714691331703701556
I was recommended to open issues.

effelle, 2020-05-27:
Please test this experimental build and check in console the output, thanks.
[dual_experimental.zip](https://github.com/arendst/Tasmota/files/4689522/dual_experimental.zip)

AlexTransit, 2020-05-28:
i flashed this firmware
if press button then write log
```
00:01:31 APP: Button code 0000
00:01:31 APP: Button code 0004
00:01:31 APP: Button code 0400
00:01:31 APP: Button code 0400
```
if press again then write
```
00:01:35 APP: Button code 0000
00:01:35 APP: Button code 0004
00:01:35 APP: Button code 0404
00:01:35 APP: Button code 0404
```

arendst, 2020-05-31:
I have a dual and will research now.

effelle, 2020-05-31:
Ok, I didn't had the time to play with yet, but two more people confirmed the button isn't working since 6.5.

arendst, 2020-05-31:
Just verified it too. It digged up a broken Dual with 5.12.0c which works when powered from USB.

Just updated to 8.3.1.2 and indeed as dead as dead can be. Nice problem. Investigating.

effelle, 2020-05-31:
@AlexTransit I'm closing this issue as it has been fixed. Please add your comment here if you need more assistance.

closed 2020-05-31 by effelle

AlexTransit, 2020-06-01:
Thank you.
Checked. everything works.