arendst/Tasmota #8566: Adjustable LWT-topic

tobetobe66, 2020-05-28:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
I have an arbitrary number of  nodes (Gosunds SP1) communicating with each other via MQTT in order to do load management from the electricity grid. In order for the nodes to find out if any other node disconnected I would like to use the MQTT LWT feature. However as none of the nodes knows the device name of any other node I would like to set the LWT-Topic.
The standard seems to be: tele/DVES_XXXXXXX/LWT.
I would like to set it for example to: abc/LWT. 

**Describe the solution you'd like**  
 I would like to set the LWT-Topic via command.
The standard seems to be: tele/DVES_XXXXXXX/LWT.
I would like to set it for example to: abc/LWT. 

**Describe alternatives you've considered**  
I could implement a watch-dog on the mqtt-broker.

**Additional context**  
_._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-28:
Hi

LWT is Last Will an Testament and is being used by the MQTT broker as a trigger for ONLINE and OFFLINE. If the broker detects an unclean disconnection from a client, the MQTT broker fires an OFFLINE message that can be used for any home automation software or any device.

> I would like to set the LWT-Topic via command.
> The standard seems to be: tele/DVES_XXXXXXX/LWT.

The standard is not `tele/DVES_XXXXXXX/LWT`
The standard is `tele/[YOUR_DEFINED_TOPIC]/LWT` like `tele/tasmota/LWT`

And you can change the topic by `TOPIC` command in the console or by the MQTT menu in Tasmota WebUI, or by sending to your device the command TOPIC XXXX

> Describe alternatives you've considered
> I could implement a watch-dog on the mqtt-broker.

That is already implemented in all MQTT brokers by the use of LWT messages

So, most of what you want is already working on Tasmota and in all MQTT brokers as default.

If you want a device to MONITOR another device if it is ONLINE or not, you can make your Tasmota to subscribe to the LWT of the other device and make a rule to trigger and do what you want.
For subscribe to work, you need to enable that and compile by yourself.

See https://github.com/arendst/Tasmota/blob/fdc5ffa36bec2eda112205c5a0f976eb567df379/tasmota/my_user_config.h#L406

For rules' usage, please check the docs at rules and rules cookbook.

If you need further assistance on Tasmota usage, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-28 by ascillato2

tobetobe66, 2020-05-28:
First of all: Thank you.
Maybe I did not make myself clear enough.
I was aware of all you said. My problem is, the clients do not know each other.
If I could set the LWT of the clients to a top level topic like abc/LWT it would help tremendously.
If I just change %TOPIC% all nodes will respond i.e. to cmnd/%TOPIC%/.... I am loosing the possibility to address nodes individually.
Thx

meingraham, 2020-05-28:
Each device retains its unique %topic%. If you know in advance which devices to monitor, you can have each device [subscribe](https://tasmota.github.io/docs/Commands/#subscribe) to the LWT message for all of the devices of interest. This would be a "hard coded" list of devices, of course.

tobetobe66, 2020-05-28:
Agreed, but it is not hard coded. It' a self organizing network. The devices are monitoring themselves without knowing each others device name. Devices come and go (in the event of failure). If each device subscribes to topic abc/LWT every device can find out if another device goes offline. I can do that certainly also with say tele/abc/LWT but then every command message (say for setting up configuration etc...) is send to all devices as I cannot discriminate them anymore. I am not sure if I made myself clear enough?

tobetobe66, 2020-05-28:
I forgot to say, I am using scripting.

ascillato, 2020-05-28:
> The devices are monitoring themselves without knowing each others device name.

????

If they don't know their names or you have nothing to identify them, they don't have any topic to subscribe to.

You want just one LWT for the whole device group???

In that case, you need a non-standard approach. MQTT don't have that type of option.

The simplest approach to that general LWT is to make it by the use of rules and you can define the behaviour as you want.

lalo-uy, 2020-05-28:
Then subscribe to tele/+/LWT

tobetobe66, 2020-05-28:
> Then subscribe to tele/+/LWT El jue., 28 may. 2020 a las 14:44, Adrian Scillato (< notifications@github.com>) escribió:
> […](#)

Thx, this should work, aaargh, could have noticed myself.
Thx a lot.

tobetobe66, 2020-05-28:
Sorry for bothering you guys, special thx to lalo-uy

meingraham, 2020-05-28:
> Then subscribe to tele/+/LWT

@lalo-uy - You beat me to it :wink: I was coming on to say the exact same thing.

tobetobe66, 2020-05-28:
Yeah guys, here comes the sad part, it works say in mosquitto_sub, but not in tasmota, at least not in scripting. I do not use rules and am therefore not familiar with it, so I cannot say up to now, if it is a scripting issue. If I use i.e. +>subscribe r, tele/DVES_8D6B7A/LWT for a particular client I get the last will of that client. If I use +>subscribe r, tele/+/LWT, I get no message.

Is the /+/ feature implemented?
Thx

meingraham, 2020-05-28:
@tobetobe66 You'll have to pore through some commits to see who the author of the `Subscribe` command was.

Pretty sure that @gemu2015 is probably just leveraging that code w/in his scripting.

I wonder if the full blown wildcard `#` would make any difference?

tobetobe66, 2020-05-28:
I'll check. I'm in touch with gemu already because of other issues. Just checked #, it works, but obviously I get all sorts of messages. Maybe I get along with that.

tobetobe66, 2020-05-28:
Ok, here is what I will do.
as /+/ does not work in scripting, I'll use either /# or most likely I'll use the same topic tele/[YOUR_DEFINED_TOPIC]/LWT for all clients. In the latter case I will use the FALLBACK topic in case I have to address clients individually.

meingraham, 2020-05-28:
Have you tried `tele/#/LWT`?

tobetobe66, 2020-05-28:
Well just tried: mosquitto constantly disconnects the client. mosquitto_sub also does not accept it.
As far as I know you can only use it a the end of a topic, in between you have to use +.

meingraham, 2020-05-28:
I thought `#` allowed any number of levels to be wild carded and then look specifically for the ending bit. Learned something new today.