arendst/Tasmota #8568: Tasmota Mqtt Google IoT Core

xiskoct, 2020-05-28:
Hello, I am trying to connect tasmota to the iot core mqtt broker, for this I am following the same steps as AWS IOT Core mqtt.

I am getting the following error:

19:57:09 MQT: TLS connection error: 62
19:57:09 MQT: Connect failed to mqtt.googleapis.com:443, rc -2. Retry in 10 sec
19:57:20 MQT: Attempting connection ...

Error 62:
X509 not trusted, the server certificate is not signed by the CA (AWS IoT or LetsEncrypt)

In TLSKey1 I have assigned the value of private key.
In TLSKey2 I have uploaded the root certificate of google iot (wget pki.goog/roots.pem)

Is it possible to use tasmota to register the device without using a certificate, since in iot core it is not mandatory to use this?

I've tried commenting // #define USE_MQTT_TLS_CA_CERT and // #define USE_MQTT_AWS_IOT, but in this case I can't set the private key using the TLSKey command, it tells me the command doesn't exist.

Thank you.

Regards.

closed 2020-05-28 by close-issue-app[bot]

chrisfraser, 2020-07-23:
@xiskoct Did you win with this?

PavelGorbanj, 2020-12-30:
@chrisfraser, @xiskoct , have you succeeded in getting T to work with GCP/ Cloud IoT Core (with or without that certificate)?

WardenCommander, 2021-11-15:
anyone managed to connect this two (Google IoT Core and Tasmota) ?

PavelGorbanj, 2021-11-15:
@WardenCommander 
> anyone managed to connect this two (Google IoT Core and Tasmota) ?

not me (though I've tried long time ago). If nothing has changed since then, and if you by chance figure out relatively lightweight workaround (some sort of MQTT proxy/retranslator or even Node Red), can you please share? My recent check on respective Node Red node has shown it's not really supported and thus seems doesn't work with current IoT Core env. Thank you!