arendst/Tasmota #8576: Integrating HAA with Tasmota

MOJT17, 2020-05-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Is it possible for you to integrate HAA (Homekit Accessory Architect) into Tasmota natively so I could use the Tasmota functionalities while also being able to control it from apple devices at the same time through Homekit?

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Using HAA only

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-05-30:
Sorry no since it uses esp-open-rtos. 
Tasmota is running only with Arduino Esp8266 which is based on nonos-sdk.

effelle, 2020-05-30:
Closing this issue as it has been answered.

Support Information
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-05-30 by effelle