arendst/Tasmota #8577: Treatlife Dimmer DS01: LED stop working after power loss

filgr8, 2020-05-30:
#

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

After power loss Dimmer works as expected but LED light will stop working/responding to anything  starting Tasmota 8.2.0 and until latest release 8.3.1
8.1.0 LED works as expected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ok ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ok] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ok] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ok ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ok] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ok] Device used (e.g., Sonoff Basic): ___Treatlife Dimmer DS01
- [ok] Tasmota binary firmware version number used: ____8.3.1
  - [ok ] Pre-compiled 
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ok ] Flashing tools used: _____tuya convert ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

treatlife ds01 template only

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
can not as downgraded to working Tasmota 8.1.0

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1 apply Treatlife Dimmer template 
2 disconnect dimmer from power sours AC
3 connect dimmer to power sours AC
4 LED light on the dimmer stops working

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
LED light (on the switch) On when dimmer is Off) 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
"temp fix"
In order to fix (until next power loss)  have to change Tuya Tx gpio  at the template  to a different  gpio, save it, then disconnect it from power AC and then reconnect back . LED comes back to its expected behavior. Set Tuya Tx to its original gpio if want to use dimmer . 
If to disconnect switch/dimmer from AC power again - no LED.

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-05-30:
you have not provided any relevant output information nor the template applied to the device. please add the requested

filgr8, 2020-05-30:
I did mentioned template that I have applied it is treatlife dimmer ds01 template 
{"NAME":"DS02S Dimmer","GPIO":[0,107,0,108,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":54}

Nothing else was tampered or applied to the switch.
As mentioned before I have downgraded to 8.1.0 because neither 8.2.0 nor 8.3.1 LED working if loss of power happens.

blakadder, 2020-05-30:
Since the template doesn't define LED it is controlled by the MCU so the error you're experiencing is due to power loss reinitialising something in the MCU

are you saying that does not happen on Tasmota 8.1?

filgr8, 2020-05-30:
yes it does work on 8.1 but breaks at 8.2 and 8.3

broken, 2020-06-24:
@filgr8 Can you let me know the steps you took to downgrade? I've tried in the past, but Tasmota appears to only allow me to upgrade the switch..

filgr8, 2020-06-24:
Lets say you want downgrade to 8.1.0

Go to https://github.com/arendst/Tasmota/releases/tag/v8.1.0

Download 2 files 

tasmota-minimal.bin
and
tasmota.bin

Then go to your tasmota device and go to Upgrade section. Choose upgrade from file.

First choose and upload Tasmota-Minimal bin file, hit upgrade. Then when device restarts it would say it is “limited minimal bin file bla bla bla “ in red. 

Then do same for tasmota bin file and do upgrade. 

Wait until it restarts a minute or two and you are all done. You are downgraded to 8.1.0

joba-1, 2020-06-24:
this can work or fail. If the old versions persistent config is incompatible you could end up with no WLAN and no way to configure it in tasmota-minimal. Better be prepared  to flash via serial if you try this.

broken, 2020-06-26:
Thanks; that downgrade process worked for me. My problem before was I didn't upgrade using the minimal build first.

Unfortunately, this didn't fix my led issue, even after resetting the configuration. I'm not sure if it was because of how long it's been since I upgraded, or it was in some bad state when downgrading, but it was disappointing to say the least.

broken, 2020-06-26:
oh, I just read your fix above.. I'll have to try that this weekend when I'm able to switch off the circuit breaker.

ascillato2, 2020-07-09:
Please, when you have time, tell us if you were able to solve your issue. Thanks.

filgr8, 2020-07-09:
ascillato2 why did you closed it?
Nobody solved this issue yet, or can fix it. It is a bug in tasmota firmware starting 8.2 and up to present 8.3

x9p2vq, 2020-07-14:
I also have several of the same switch (TreatLife DS01) that exhibit the same behavior.  All of mine are running 8.3.1.  Let me know if there is anything you would like me to test.

mkfink, 2020-07-27:
I just got a pack of these and have been playing with them, and I ran across this issue. I don't think I'm experiencing it, but I want to make sure I'm not missing something or else maybe help out. Which LED is going out? I see the primary red and white LEDs along with the LEDs along the dimmer track. I have one with wire nuts connected to a power cord and I'm plugging/unplugging it repeatedly and I don't see a change in LED behavior. Running Tasmota 8.3.1 compiled with [this user_config_override.h](https://gist.github.com/mkfink/b12456543bba7e2905319af3331b9756) (sorry it's a bit unconventional to look at, I generate that with a script. I think all the relevant bits are at the end.)

broken, 2020-07-27:
It's the LED light at the bottom that turns on when the switch is off.

Do you experience it w/o your custom user_config_override compiled in?

mkfink, 2020-07-27:
Interesting. I just built stock tasmota with only my wifi info in user_config_override.h, pointed it at my broker, and set the module to 54. And I can definitely reproduce your issue with that, no more white LED. No change after running 'backlog TuyaMCU 21,2; DimmerRange 150,1000; Rule1 ON System#Boot DO SerialSend5 55AA00060005040400010114 ENDON; Rule1 1; Restart 1' or enabling SetOption54/TUYA_SETOPTION_20. 

I just flashed back to my version and the LED works again. It seems like something that I'm disabling is affecting the LED, but I don't think I've made any other changes to the defaults related to Tuya devices.

closed 2020-08-29 by stale[bot]

SylvainGa, 2020-10-17:
Is there a solution for this? I'm at 8.5.1 and white LED is never turning on. It should be on when the light is off and off when the light is on. Using TuyaMCU (54) config, TuyaMCU 21,2 and DimmerRange 150,1000 console commands. Everything else is stock.

I know this is a closed issue but I don't think it has been resolved.

shawnr13, 2021-03-10:
> I know this is a closed issue but I don't think it has been resolved.

This resolved for me with a combination of upgrading to 9.1 and then having a power outage a while later. Once the power was restored, the LED indicator magically started working like it did before.

filgr8, 2021-03-10:
The only solution to issue Is to roll back to 8.1.
All my treatlife dimmers are on 8.1 and work without issues. Interesting thing that with 9.1 those dimmers in HA are not responding to dimming sometimes but works fine on 8.1

shawnr13, 2022-04-10:
Just updating this as I have 4 switches on the same circuit (2, 3-way switches) and only one set of those are not working (power outages should have affected them equally). I upgraded to 11.0.0, went back down to 8.1 (and back to 11.0.0). I could not get it to work again and I fear flipping the breaker might fix the broken ones and break the working ones. I've never tried making my own using the user_config_override but I might try that when I get some time and see if that fixes it. My switches were all on 9.3.1 working fine until a power outage (of course). Surely with there is something to be done software side on this. I'm just not knowing enough to figure that part out. I'd be happy to attempt any suggestions.