arendst/Tasmota #8583: Adventures with Sonoff Zigbee Bridge

s-hadinger, 2020-05-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
This issue is opened to track progress for the support of Sonoff Zigbee Bridge.

**Describe the solution you'd like**  
Z2T full support, like with CC2530.

**Describe alternatives you've considered**  
CC2530

**Additional context**  
This issue is only to report about progress and issues on the Sonoff Zigbee Bridge with Tasmota. It shall not contain any speculation like in #8197. All speculations should stay in #8197.

s-hadinger, 2020-05-30:
The form factor is cool and similar to the Sonoff RF Bridge, in white.

[image]

Opening the box is super easy, remove the 4 rubber protections, and unswcrew.
[image]
[image]

As reported before, it is based on ESP8266EX so flashing Tasmota shouldn't be hard. 

[image]
[image]

The good news is that pads are easily accessible. Sonoff made available lots of GPIOs, although it looks only RX/TX (GPIO1/3) are used to communicate with the Zigbee module.

There is a `Z_RST` pad which should be the RESET for the Zigbee module. This may be useful since GPIO1/3 are connected to the module. If we want to flash Tasmota without interference, I guess we should keep `Z_RST` low.

Useful pads: GPIO1, GPIO3, GPIO0, 3.3V, GND, Z_RST
[image]

We'll however need to wait a little more to analyze the protocol, my logic analyzer is arriving next week.

s-hadinger, 2020-05-30:
First test with the Sonoff Zigbee Bridge in pairing mode, using another Z2T in device mode.

The Z2T device finds the Sonoff coordinator, connects to the network but then nothing happens, the device does not receive any more packets from the coordinator. It looks like Sonoff is restricting to its own devices, either filtering on IEEE Address, or waiting for a specific packet.

s-hadinger, 2020-05-31:
Listening to TX at boot, 115200 bauds, 8N1, here is what is printed to Serial:
(3 sequences)

```

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DOUT
SPI Flash Size & Map: 16Mbit(1024KB+1024KB)
jump to run user1 @ 1000
```

```
OS SDK ver: 2.0.0(e271380) compiled @ Mar 30 2018 18:54:06
```

```
phy ver: 1055_1, pp ver: 10.7

rf cal sector: 507
```

The MCU sends two identical frame burst at 172800 bauds (very uncommon bit rate):
```
FE 0D 01 80 02 00 00 00 00 01 01 04 74
```

s-hadinger, 2020-05-31:
Here is the complete sequence at start (115200 bauds, 8N1):

- ESP pulls down `Z_RST` for 10ms
- MCU signals it has booted with `FE 0D 01 80 02 00 00 00 00 01 01 04 74`
- later, ESP sends `FE 0A 01 01 02 00 00 00 43 B5 FE 0C 01 04 02 00 00 00 43 00 03 B5`
- MCU responds `FE 0D 01 41 02 00 00 00 43 01 01 04 F6`
- MCU also responds 8ms later `FE 15 01 44 02 00 00 00 43 00 03 00 FA A4 66 FE FF 14 2E 84 6B`

arendst, 2020-05-31:
So it starts with fe followed by the total byte count

s-hadinger, 2020-06-01:
Aha, listening to the UART port of the MCU:

First sequence:
```
Reset info: 0x04 (PWR)
Extended Reset info: 0x0401 (HV )
```

52ms later:
```
init pass
Build Time Apr 29 2020 17:12:29
```

```
EEPROM init: 0x00
```

```
EMBER_NETWORK_UP 0x0000
[Z3Coord_callbacks.c:emberAfStackStatusCallback]
status = 0x90
Send Data:
FE 0D 01 80 02 00 00 00 00 01 01 04 74 

Enter password:
```

Then after the ESP pulls down MCU Reset, we see a similar sequence again:
```
Reset info: 0x03 (EXT)
Extended Reset info: 0x0301 (PIN)
init pass
Build Time Apr 29 2020 17:12:29
EEPROM init: 0x00
EMBER_NETWORK_UP 0x0000
[Z3Coord_callbacks.c:emberAfStackStatusCallback]
status = 0x90
Send Data:
FE 0D 01 80 02 00 00 00 00 01 01 04 74 

Enter password:
```

And 175ms later:
```
InitNetworkEventHandler
PANID = D136
```

Hedda, 2020-06-01:
@s-hadinger It would be absolutely great if that SONOFF ZBBridge (SONOFF Zigbee Bridge) already have Ember firmware based on EmberZNet Zigbee Stack with their standard EZSP (EmberZNet Serial Protocol) serial protocol interface / API over UART bus, as it now looks to have.

If it already has Ember firmware with EZSP interfaces then will hopefully not need to worry about any firmware upgrade of in the first place and instead just focus on hacking its ESP8266 instead and maybe offering a serial server service (like Ser2Net) to offer remote access the Zigbee module.

However to answer the question. Yes, If the Zigbee SoC module used in the SONOFF ZBBridge (SONOFF Zigbee Bridge) is at all similar to other EFR32 modules, like the E180-ZG120B by Ebyte, then yes at least according to the Ebyte manual can flash the firmware on EFR32 module with J-Link

- http://www.ebyte.com/en/product-view-news.aspx?id=842
  - http://www.ebyte.com/en/downpdf.aspx?id=842
  - http://www.ebyte.com/en/pdf-down.aspx?id=1618

In the Ebyte E180-ZG120B manual, the burning program sais that GPIO pins for J-Link interface is:
* VCC (Positive Supply Voltage)
* SWCLK (Clock Pin)
* SWDIO (Serial Wire Debug IO)
* GND (Ground)

FYI, there's also a related discussion about flashing firmware on these type of EFR32 modules here:

- https://github.com/zigpy/bellows/issues/24

If still are serious about flashing that firmware then checkout a Silicon Labs document called AN136

- https://silabs-prod.adobecqms.net/documents/public/application-notes/AN136-production-programming-options.pdf

Maybe it is possible to just use the Command-Line Interface (CLI) of Simplicity Commander as it supports using Serial Wire Debug (SWD) or Joint Test Action Group (JTAG) as the target interface?

- https://www.silabs.com/documents/public/user-guides/ug162-simplicity-commander-reference-guide.pdf

AN136 otherwise refers to other documents for many different methods, like SWD, JTAG, and C2:

- AN0062 https://www.silabs.com/documents/public/application-notes/an0062.pdf
- AN105 https://www.silabs.com/documents/public/application-notes/an105.pdf
- AN127https://www.silabs.com/documents/public/application-notes/AN127.pdf

MarkDing who work at Silicon Labs summarized their Serial Wire Debug (SWD) in their forum here:

 - https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2014/10/21/serial_wire_debugs-qKCT

The same @MarkDing also maintains a wiki about development with Silicon Labs IoT MCU SoCs:

- https://github.com/MarkDing/IoT-Developer-Boot-Camp/wiki

Hedda, 2020-06-01:
All, by the way, here are a few of related discussions specifically about hacking SONOFF ZBBridge:

- https://github.com/arendst/Tasmota/issues/8583 (started by @s-hadinger  )
- https://github.com/zigpy/zigpy/issues/385
- https://github.com/xoseperez/espurna/issues/2224
- https://github.com/esphome/feature-requests/issues/688
- https://github.com/1technophile/OpenMQTTGateway/issues/205
- https://github.com/letscontrolit/ESPEasy/issues/3021

s-hadinger, 2020-06-01:
More data:

When entering Pairing mode, MCU sends:
```
FE 0B 02 01 02 F9 FF 00 04 37 C1
```

MCU responds:
```
FE 0B 02 41 02 F9 FF 00 04 00 B6 
```

MCU logs:
```
Recv Data:

FE 0B 02 01 02 F9 FF 00 04 37 C1 
pJoin for 55 sec: 0x00
Open network: 0x00
Send Data:
FE 0B 02 41 02 F9 FF 00 04 00 B6 
```

s-hadinger, 2020-06-01:
Aborting Pairing mode:
```
Recv Data:

FE 0B 02 01 02 F9 FF 00 03 00 F1 
pJoin for 0 sec: 0x00
Send Data:
FE 0B 02 41 02 F9 FF 00 03 00 B1 
```

s-hadinger, 2020-06-01:
Finally, when trying to join the network from a Z2T device, the MCU logs the following but does not report to the ESP:
```
[Z3Coord_callbacks.c:emberAfTrustCenterJoinCallback]
add dev:NodeID=4351 mac:4B:CB:9D:19:00:4B:12:00ParentOfNewNode=0, status = 1 decision=0
Adding device to the queue
Adding to queue at index 0
```

It looks like Ember layer did accept the connection, but the app layer dropped it; probably filtering based on IEEE address.

s-hadinger, 2020-06-01:
Connecting to the EFR32 with a JLink probe (SWD mode using `SWDIO` and `SWCLK`), it shows that the device is debug-locked but can be erased and should be possible to reflash with a custom firmware:
[image]

Or Sonoff is kind enough to give the unlock key?

digiblur, 2020-06-01:
Excellent work so far!  I see the have a new board revision listed on yours.  I'll have to compare to see what changed.

smdjeff, 2020-06-02:
There's no need to reverse engineer the protocol. EZSP was Ember's (now Sillabs) RPC protocol between network coprocessors and host microcontrollers. There's a SPI and USART flavor. It's well documented and the host side sources are easily ported to any host including ESP. https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

stevew817, 2020-06-02:
> Or Sonoff is kind enough to give the unlock key?

According to that screenshot, there is no unlock key on the device, so Sonoff wouldn't be able to give you one either way :)

Hedda, 2020-06-02:
Sounds as if don't need an unlock key to erase & flash firmware via JTAG interface or SWD interface:

- https://www.silabs.com/mcu/programming-options#programming
  - https://www.silabs.com/documents/public/application-notes/an958-mcu-stk-wstk-guide.pdf
    - https://www.silabs.com/documents/public/application-notes/an105.pdf
    - https://www.silabs.com/documents/public/application-notes/an0062.pdf

Simplicity Commander support CLI as the target for EFx32 but it is not open-source or for Arduino.

- https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2016/12/29/simplicity_commander-DqhR

"Simplicity Commander is a software package provided by Silicon Labs that provides a very powerful GUI and command-line utility for working with your EFR32 parts. With Simplicity Commander you can flash firmware (bootloaders and applications), update the WSTK firmware, reset locked parts and unlock debug access. It is the EFR32 version of the ISA3 Utilities."

Simplicity Commander GUI and CLI interfaces with software download for Linux, Mac, and Windows.

- https://www.silabs.com/documents/public/user-guides/ug162-simplicity-commander-reference-guide.pdf

Simplicity Studio also includes a simpler tool called "Flash Programmer" that allows the user to flash the device with a binary or hex file, supporting flashing to EFM8, EFM32, C8051, EZR32, and EFR32.

- https://www.silabs.com/products/development-tools/software/simplicity-studio

Hedda, 2020-06-02:
Will you brick the EFR32 if erasing application firmware also erases the bootloader firmware? 

That is if erasing the bootloader firmware is not optional, sounds strange if that is not optional?

If that is the case and EFR32 gets bricked then can you still unbrick it by flashing a new bootloader via the JTAG or SWD interface on EFR32 using just a Raspberry Pi / Arduino or will you need a purpose-built debugger for EFR32 or an official EFM32 Gecko Development Kit to unbrick it?

Gamester17, 2020-06-02:
> Or Sonoff is kind enough to give the unlock key?

ITead (Sonoff) as a company has benefited a lot from alternative firmware from the community in the past and there are today more aware of that fact so would not hurt for you to ask.

Suggest that you sent a mail to "Jerry Shi" <jerry.shi 'at' itead.cc> and "Stan Li" <stan.li 'at' itead.cc> and explain that you are an open-source developer from the Tasmota community that is looking into developing an alternative firmware for the Sonoff ZBBridge and are therefore wondering if they could share the key to disable the debug lock. Jerry Shi and Stan Li 

I have no affiliations to ITead/Sonoff myself however in the past I have sent a few questions or suggestions to them and got relatively quick replies from Jerry Shi and Stan Li. I think that Jerry is in marketing at ITead so he has to pass on the technical questions to the developers, however, I know he helped the community in the past and since he is in marketing I assume he would especially helpful if it directly or indirectly helps ITead sell more products, which is something Tasmota surely does.

stevew817, 2020-06-02:
Again, there is no secure boot key installed in the device, so there are no obstacles to mass-erasing and writing a new bootloader + application. The only thing you cannot do is reading out the current firmware.

Hedda, 2020-06-02:
Unlocking / disabling debug lock also erase all data to prevent reading or? 

That is, does it also do a full erase of bootloader + application firmware?

That part is still unclear for me even after trying to interpret that from these:

https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2016/11/04/unlock_a_brickedefr-GkC7

https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2018/06/07/erasing_the_lockbit-WVQL

https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2017/10/09/how_to_lock_or_disab-mb2d

https://silabs-prod.adobecqms.net/content/usergenerated/asi/cloud/content/siliconlabs/en/community/mcu/32-bit/knowledge-base/jcr:content/content/primary/blog/unlock_a_brickede-xkAm.social.$%7BstartIndex%7D.10.html

https://silabs-prod.adobecqms.net/content/usergenerated/asi/cloud/content/siliconlabs/en/community/mcu/32-bit/forum/jcr:content/content/primary/qna/bricked_efr32-1EjX.social.list-item.0.15.html

https://silabs-prod.adobecqms.net/content/usergenerated/asi/cloud/content/siliconlabs/en/community/mcu/32-bit/knowledge-base/jcr:content/content/primary/blog/debug_unlock_withj--QGXr.social.0.10.html

stevew817, 2020-06-02:
@Hedda correct. Unlocking a locked device, or disabling a debug lock that has been set, will result in an erase of main flash and RAM.

Some devices in EFR32 Series 1 (e.g. EFR32MG12) have a separate flash area where a bootloader optionally can reside.
This is not the case for Series 2 (e.g. EFR32MG21 as used here), where a bootloader always resides at the start of main flash, and is thus erased together with the rest of it during an unauthorised unlock sequence.

s-hadinger, 2020-06-02:
Bad day. I couldn't flash Tasmota on the ESP8266. The flash is corrupt and comparing by reading back the content, there are some random bit flips. Even reading the same location gives random bit flips. I changed the power supply and from another host (Raspberry Pi) but I get the same corruption.

I will check tomorrow the soldering, but this is a complete mystery. This should be the easy part. I'm stuck...

arendst, 2020-06-02:
Are you sure the serial interface is free of noise from the attached zigbee part? It might interfere with the firmware upload and download.

arendst, 2020-06-02:
Also flash while usb powered.

s-hadinger, 2020-06-02:
There must be noise. I force the zigbee reset to low to make sure the MCU doesn't interfere, no change. I was USB powered too, with 3 different ways (battery, laptop and charger).

I flash a 4KB block full of 0xFF, and read it back. There are ~4 bit flips in the block. And re-reading the block, some but flips remain, some change. Very weird...

I saw similar bit flips when flashing the entire Tasmota firmware.

Since the the firmware is sent in compressed form by esptool.py, I must not be the UART interface, otherwise the flash should be completely corrupt.

Only GPIO 6/7/8/11 are connected to the SPI Flash, so it's either DIO or DOUT.

On a side note, GPIO9/10 have Pads but are not connected to anything, no pull-ups/down. Could this be where the noise come from?

s-hadinger, 2020-06-03:
I'm almost certain there is an issue with the Flash or the ESP-Flash communication. Or with Flash erasing.

I flashed a 4KB file full of 0xFF. The file is transmitted in ZIP format by esptool.py and the checksum is locally tested. And it's always wrong.w

Then I flashed a 4KB file full of 0x00. Now the flash went ok.

kugelkopf123, 2020-06-03:
@s-hadinger Why not simply unsolder the flash chip and replace it with a flashed one from a Wemos or NodeMCU? At least you'll have Tasmota on it and you can continue testing. You could also solder the one from the zigbeebridge to the Wemos and test if the chip has a defect.

s-hadinger, 2020-06-03:
@kugelkopf123 That would be in last resort. But I don't have a heat gun nor spare flash chips, or I need to crack open a Wemos D1.

After doing more tests, it looks like erasing flash is broken. Some bits never go back to `1` or start flipping randomly.

I have a second Sonoff Zigbee bridge but I'm reluctant in potentially bricking it too.

kugelkopf123, 2020-06-03:
> @kugelkopf123 That would be in last resort. But I don't have a heat gun nor spare flash chips, or I need to crack open a Wemos D1.
>
> After doing more tests, it looks like erasing flash is broken. [...]

Never needed a heatgun. Had done it always this way: https://youtu.be/0KWkNDRQW5E
So, if you think you have already destroyed the device, go ahead!  nothing to lose.

s-hadinger, 2020-06-03:
Thanks. I will try. Do you know how to open a Esp8266mod case?

kugelkopf123, 2020-06-03:
> Thanks. I will try. Do you know how to open a Esp8266mod metallic case?

It's soldered on too. 
So it might be possible to disconnect it with the same method. I've never done it before, though. Don't you have a Shelly or Sonoff Basic? There's no metal shield covering the chips. Maybe you also have a Wemos d1 Mini v3. They're not covered there either.

Hedda, 2020-06-04:
WeMos D1 (all models?) should have the same type of "ESP8266EX" as Sonoff ZBBridge so could possibly be the perfect hardware for creating alternative development kit platforms for this project?

As for the Zigbee MCU SoC part it seems to currently be a bit harder to find inexpensive alternative hardware modules with Silicon Labs EFR32 Series 2 (EFR32MG21) MCU SoC if you do not want to by Silabs official EFR32xG21 Wireless Gecko Starter Kit (SLWSTK6006A) for $479 US-dollars on your own:

- https://www.silabs.com/products/development-tools/wireless/efr32xg21-wireless-starter-kit

Though three developers could co-purchase it as that kit contains three complete development boards and EFR32MG21 Zigbee radio modules, plus a bonus is that purchasing the starter kit also it gives your Silabs registered account access to the Zigbee stack which is not publicly available otherwise.

However, if it would be good enough to use a Silicon Labs EFR32 Series 1 (EFR32MG12) module in a DIY development kit instead, if so then check out this Ebyte E180-ZG120B module and the E180-ZG120B-TB (EFR32 Test Board) which can both be found relatively inexpensive from ex. Aliexpress or eBay.

- http://www.ebyte.com/en/product-view-news.aspx?id=842

- http://www.ebyte.com/en/product-view-news.aspx?id=896

At least Silicon Labs seem to use the same SDK and Zigbee stack for all EFR32; EFR32 Mighty Gecko Wireless SoC Series 1 = EFR32MG1/EFR32MG12/EFR32MG13/EFR32MG14, and EFR32 Wireless Gecko Series 2 =EFR32MG21 "Mighty Gecko". Same Zigbee API too however a newer version API.

- http://wless.ru/files/ZigBee/EFR32MG21/EFR32xG21_migration_guide.pdf

Hardware-wise all Wireless Gecko Series 1 MCUs obviously has less RAM and FLASH memory as well as being based on Arm Cortex-M4 instead of an Arm Cortex M33 CPU that Wireless Gecko Series 2 has. 

- https://www.cnx-software.com/2020/04/27/ebyte-e180-zg120b-tb-zigbee-3-0-evaluation-board-features-silicon-labs-efr32mg1b-zigbee-thread-soc/

FYI, alternative hardware platforms for development platforms was also suggested/discussed here:

- https://github.com/arendst/Tasmota/issues/8197

mtx512, 2020-06-04:
I'm currently working with a few EFR32 modules ((MGM111A256V2, MGM13P12F512GA) through my work and have access to the Zigbee stack and can generate a UART NCP build if required or help in any other way. May save you having to purchase the Silabs dev kit.

I reviewed the  protocol dumps by @s-hadinger and initially thought the serial protocol was based on EZSP SPI interface which also starts with a leading 0xFE however the rest of payload doesn't match the EZSP protocol definition. So it looks like its a custom serial protocol that is integrated with  the Ember NCP stack.

I suggest using Simplicity Commander to try to take a flash backup of the EFR32. Unfortunately I don't have a Sonoff Zigbee Bridge otherwise I would test it. 

@s-hadinger I suspect the EFR32 may be using the Gecko Bootloader, to access it you can connect a serial console 115200,8,N

1. Tx - Pin PB01 Rx Pin PB00  ( 2nd & 3rd pin from top on left hand side of module)
2  Ground PA00  (4 pin from top on left hand side of module)
3. Temporary ground the reset pin nRST to reset (top pin of right handside of module)

Hopefully the bootloader prompts appears.

grobasoz, 2020-06-05:
Looking at the messaging on the CLI ("MCU Logs" over ZigBee Usart) it seems like Sonoff are using a custom application on the EFR32 and not the Silabs NCP (ie EZSP protocol). It's not that common for the Silabs NCP to have a CLI interface either... the "Network Up" etc messages are normally posted on the Host side.
This would make sense due to the memory constraints of the ESP and the effort required to create the NCP support application in the ESP.
Sonoff also use a similar packet structure to the TI ZNP interface (though TI use a payload length and Sonoff a packet length in byte position 2).

Since there is already Tasmota to ZigBee firmware based on the TI ZNP protocol, it would make sense to build an EFR32 app that works in a similar fashion to the ZNP and uses the ZNP packet structure, perhaps 'upgraded' to ZigBee 3.0 with Green Power support.

As the EFR32 in the Sonoff product is probably not running the Silabs NCP firmware, I suspect they have also used the Gecko 'Application' Bootloader as opposed to the 'Standalone' XModem bootloader as suggested above. The EFR32 has enough flash to support internal storage of bootload firmware of about 512k... so you could potentially run the ZigBee/Bluetooth app and allow firmware updates over Bluetooth. 

[image]

The CLI is on the standard EFR32 USART0 port (TXD = PA5, RXD = PA6), but I can't make out the ESP serial pins from the pictures. They are probably disabled in the EFR32 app before updating the ESP.

Hope this helps... 
Gary.

Hedda, 2020-06-05:
@mtx512 if and when you do build compatible Zigbee coordinator firmware for the EFR32MG21 that is in Sonoff ZBBridge please consider uploading/hosting the firmware images in a GitHub repo?

That is, something similar to what @SillyDay does here https://github.com/SillyDay/EFR32

@SillyDay created that repo as part on the discussion in https://github.com/Koenkk/zigbee-herdsman/issues/168

Hedda, 2020-06-05:
> it seems like Sonoff are using a custom application on the EFR32 and not the Silabs NCP (ie EZSP protocol). It's not that common for the Silabs NCP to have a CLI interface either... the "Network Up" [...]

@grobasoz would it then as an alternative idea maybe even be possible to build EFR32 app(s) that implements the bellows library and maybe also the zigpy library on the EFR32 MCU so that Tosmata would instead interface with them instead of implementing support for Silabs EZSP protocol directly in the Tosmata firmware?

- https://github.com/zigpy/bellows

- https://github.com/zigpy/zigpy

I understand that bellows can run as a stand-alone application without zigpy and then you could choose if you want also would want to implement zigpy in an EFR32 app or in Tosmata or not at all?

Note! bellows don't yet support later versions of Silabs EZSP protocol so have to be updated. See:

- https://github.com/zigpy/bellows/issues/243

grobasoz, 2020-06-05:
@Hedda, Sorry I don't know anything about **zigpy/bellows** so will need to look at it first and maybe do some testing.

Hedda, 2020-06-05:
@grobasoz well for starters they are written in Python so the first question if Python can run on EFR32?

I'm not one of its developers (nor do I have the skill to be one myself) but as I understand it the purpose of the bellows started out as a high-level CLI wrapper that abstract Silabs EZSP interfaces as well as have give you ZDO and ZCL functionalities (with CLI) as well as include an application framework with device state persistence.

zigpy (which is a refactoring of bellows) gives you the option to split out the non-proprietary ZDO and ZCL functionalities (with CLI), as well as the application and device state management parts from bellows to the zigpy library so that your implementation can be made to support many "radio libraries" which then in turn each will only handle the proprietary interface protocols from different manufacturers.

So, in this case, you would really only need to use the bellows library in a stand-alone implementation and not the zigpy library, however, if Tasmota could run the zigpy library (and if an ESP8266 has enough resources) then it could potentially be less complicated to implement support in Tasmota for other Zigbee radios from different manufacturers.

As far as I know Home Assistant's "ZHA" integration component is the only implementation of zigpy

- https://www.home-assistant.io/integrations/zha/
  - https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha

However, I read that more projects do use bellows in a stand-alone implementation without zigpy.

s-hadinger, 2020-06-05:
zigpy/bellows is actually an EZSP client implementation.

@mtx512 thanks for the tip, I didn't know GPIO13 & GPIO15 had an influence when flashing. I will try but not before mid next week.

@grobasoz Thanks a lot for the pinout of the module.

So the complete EFR32 module pinout and ESP8266 connection is as follows:
- `PB01-TX` + `PB00-RX`, pads `ETX`+`ERX`, connected to `GPIO1+3`
- `PA00`, no pad, connected to `GPIO5`
- `PA01`, pad `SWCLK`, not connected to ESP
- `PA02`, pad `SWDIO`, not connected to ESP
- `PA05`, pad `Z_TX/PA05`, not connected to ESP (EFR32 console)
- `PA06`, pad `Z_RX/PA06`, not connected to ESP (EFR32 console)
- `nRST`, pad `Z_RST, connected to `GPIO4`
- no other EFR32 pin connected

Additional pads for ESP8266:
- `IO0` connected to `GPIO0`
- `LOG` connected to `GPIO13`
- `IO15` connected to `GPIO15`
- `SDA` connected to `GPIO12`, I2C connection to 512KB EEPROM
- `SCL` connected to `GPIO14`, I2C connection to 512KB EEPROM
- `E_RST` connected to `EXT_RSTB`
- `KEY` connected to (probably) `GPIO16` (external button)

grobasoz, 2020-06-05:
@Hedda - thanks for the zigpy/bellows summary... In short - Micropython on EFR32 is not easy - though has been done... 

@s-hadinger - thanks for the pinout summary. I have managed to cobble together a similar system here for development as my Sonoff ZigBee Bridges are still "on their way"...

mtx512, 2020-06-05:
@s-hadinger thanks for the pinout summary, so my instructions to flash ESP8288 need a correction ( I thought LOG was GPIO2):

1. Ensure (IO0) GPI00 & (IO15) GPIO15 are both low
2. Ensure GPI02 is high (I would assume its pull high by default)

mtx512, 2020-06-11:
Based on the pinout summary as a first step we could create the EZSP NCP firmware configured so its outputs to Z_TX/Z_RX since these aren't used by ESP. With this approach you can attach a usb to serial to Z_TX/Z_RX and access the NCP from a Linux machine. I can build a version of Z3GatewayHost that will allow command line access to the NCP this way you can test  the capabilities of the  EZSP NCP.

Since you have to erase the EFR given the debug lock, I would also need to generate a new gecko bootloader as that will be required. Once the bootloader is deployed it would possible to upload the NCP firmware through the bootloader if we enabled it through PA00 (GPIO15).

s-hadinger, 2020-06-11:
@mtx512 This sounds like a plan. I couldn't figure out the Flash size of the EFR32. Do you habe this informatio?

mtx512, 2020-06-11:
> @mtx512 This sounds like a plan. I couldn't figure out the Flash size of the EFR32. Do you habe this informatio?

Should be the EFR32MG21A020F768IM32 so 768kB flash and 64kB Ram

mtx512, 2020-06-11:
@s-hadinger created a bootloader and ncp if you want to test, see [here](https://github.com/mtx512/SonoffZigbeeBridge). 

Afer the EFR flash is wiped you can flash the bootloader with JFlashLite (JLink) or SimplicityCommander. Since there is no application the bootloader should jump to its upload menu (connect usb serial to  Z_TX/Z_RX).

`Gecko Bootloader v1.A.3
1. upload gbl
2. run
3. ebl info
BL >
`
If that works then you can flash the ncp-uart-sw.gbl using option 1 and it launch a xmodem session or flash ncp-uart-sw.s37 using JLink.

On the serial console you should see something like this in hex if successful :

'1a c2 02 8b c2 8a 7e'

s-hadinger, 2020-06-11:
Back to business, I flashed a second Sonoff device successfully. The first had definitely a faulty flash memory.

And booting the EFR32 with PA00 low, here is what I see on the console output:
```

Gecko Bootloader v1.9.1.04
1. upload gbl
2. run
3. ebl info
BL > 
```

Tada! There is a usable bootloader

s-hadinger, 2020-06-11:
@mtx512 Is it possible to use the existing Gecko bootloader?

mtx512, 2020-06-11:
> @mtx512 Is it possible to use the existing Gecko bootloader?

Yes select 1 and try the ncp-uart-sw.gbl with xmodem transfer .

s-hadinger, 2020-06-11:
Ok, sending 1: `310D0A` responds `begin upload` and then it sends `C` every second which I suppose is XMODEM protocol.

I don't see any option to make a backup of the ERF flash. Can XMODEM pull data from flash?

mtx512, 2020-06-11:
For XMODEM if your using minicom CRTL-A S (Send file) then select 'x-modem' and then you need to select the file to send. You need to complete all this before the bootloader times out andyou  need to select option 1 gain.

To backup the EFR flash see this [link](https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2017/08/01/dumping_uploadingfl-Ka8G). If your willing to sacrifice one device then you don't need to do it.

s-hadinger, 2020-06-11:
Thanks. I tried to backup already with JLink but it failed, probably because of the Debug Lock. It's still unclear to me how you choose to flash the bootloader or the application (there is only 1 option). Also I'm not familiar with S37 file format.

If the upload fails in the middle of the transfer, I guess the new app is not flashed, right?

mtx512, 2020-06-11:
That bootloader only allows an application (.gbl) to be upload hence why the single option. 
Yes if the transfer fails then its not replaced.

mtx512, 2020-06-11:
@s-hadinger let me check about upgrading the boot loader. It may be possible to send it a gbl file as a upgrade. However the boot loader mechanism has changed between Series 1 & 2 ERF devices.

MPM1107, 2020-06-11:
In the UG266 document there'a a whole section about the bootloader upgrade procedure, so it has to be possible somehow. My main question would be whether we even have to upgrade it in the first place. What's the benefit over the bootloader that's included on the device already?

s-hadinger, 2020-06-11:
You are right, we shouldn't need to update the bootloader after all. Only the application.

Hedda, 2020-06-12:
[Cross-posting]

> Maybe by flashing esp-link onto the esp8266 and using that as a serial over wifi bridge. Then you can just flash from your PC as if the EFR was connected to your PC over serial. This of course wouldn't be the way to do it in the final implementation, but maybe it's good enough for testing?

Serial Bridge functionality (via a transparent serial network server like example ser2net and socat) for serial access to the Zigbee MCU over the network would be awesome to get working as soon as possible.

Serial Bridge function would be great as that would allow those developers more interested in only deep-diving into Zigbee MCU part for and not so much have to worry about the ESP8266 part, leaving that part to those developers more interested in that.

As @MPM1107 mentioned, [esp-link by JeeLabs](https://github.com/jeelabs/esp-link/) is a firmware for ESP8266 that has "Serial Bridge" (a.k.a. "Serial over IP") as its main feature function, though many alternative firmware for ESP8266 does also have some kind of Serial Bridge implementation, including Tasmota.

https://github.com/jeelabs/esp-link

MPM1107, 2020-06-12:
The reason I recommended esp-link over anything else is that it has the option to reset the zigbee chip and pull down the bootloader pin every time a new telnet connection is made to a special port instead of the default one. That allows for easy firmware upgrades of the zigbee chip without the need for a swd probe.

mtx512, 2020-06-12:
Here is the test cli client [Z3GatewayHost](https://github.com/mtx512/SonoffZigbeeBridge/blob/master/efr32/linux/Z3GatewayHost) you can use once the ncp is flashed. Its compiled for linux and to run it 

`Z3GatewayHost -b 115200 -f x -p <serial port>`

eg

'Z3GatewayHost -b 115200 -f x -p /dev/ttyUSB0'

Once run you see:

`Reset info: 11 (SOFTWARE)`
`ezsp ver 0x08 stack type 0x02 stack ver. [6.7.6 GA build 327]`
`Ezsp Config: set security level to 0x0005:Success: set`
`Ezsp Config: set address table size to 0x0002:Success: set`
`Ezsp Config: set TC addr cache to 0x0002:Success: set`
`.....`
`Ezsp Config: set zll rssi threshold to 0xFFD8:Success: set`
`Ezsp Config: set transient key timeout to 0x00B4:Success: set`
`Ezsp Endpoint 1 added, profile 0x0104, in clusters: 8, out clusters 19`
`Ezsp Endpoint 242 added, profile 0xA1E0, in clusters: 0, out clusters 1`
`deactivate report event`
`Found 0 files`

s-hadinger, 2020-06-12:
@MPM1107 I would like to avoid messing with multiple firmwares. I developed a quick Serial to TCP bridge for Tasmota, that's easy to do, and I can have 2 parallel TCP connections: 1 for commands and monitoring and 1 for XMODEM.

I will try flashing this evening

@mtx512 I don't have a linux box here but a Mac. I can also use a Raspberry PI. Any other flavors of the client?

s-hadinger, 2020-06-12:
Hmmm. I triggered the XMODEM upload, it works until the last packet. The sender sends 0x04 (EOT) to signal the End of Transmission, but the Gecko bootloader answers with CAN (Cancel), like if it was not the right control character to mark the end of transfer.

I doubled checked with Wireshark, the XMODEM protocol from `lsx` is correct. Is possible that Sonoff changed the XMODEM protocol on purpose?

[557 more replies]

PR #8709 (2020-06-16): Add Zigbee initial support for EmberZNet protocol
## Description:

**[Experimental]** Initial support for EmberZNet protocol (Sonoff Zigbee bridge). It only allows to send/receive messages to further develop the stack. **Zigbee stack is not yet functional**.

Added `#define USE_ZIGBEE_ZNP` and `#define USE_ZIGBEE_EZSP` to choose at compile time the Zigbee chip. ZNP is CC2530, EZSP is EFR32.

Added `ZbEZSPSend <hex>` command to send raw commands to EFR32. <hex> is the logical payload. The command automatically adds: escaping of reserved bytes (stuffing), pseudo-randomization for DATA frames, CR [... cut]
PR #8825 (2020-06-30): Zigbee EZSP milestone 3
## Description:

Zigbee for Sonoff ZB Bridge. Pairing now works, as well as basic sensor reporting.

Also added `ColorTempMoveUp`, `ColorTempMoveDown` & `ColorTempMoveStop` to commands.

**Related issue (if applicable):** fixes #8583

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [ ] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github [... cut]
PR #8976 (2020-07-25): Zigbee EZSP improvements and reduced logging
## Description:

Reduced logging, all logs for features that are now known to work are moved to log_level 3.

Added `ZbRouteError` message when devices are unreachable (still not clear when these messages appear).

Added `ZbParent` reporting from routers.

Added specific logging when unhandled ZDO messages are received.

Added I2C GPIOs to the ZBBridge template.

Now a normal start should look like this:
```
14:44:24 ZIG: Resetting EZSP device
14:44:25 RSL: RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2} [... cut]
closed 2020-08-01 by s-hadinger