arendst/Tasmota #8584: help me shelly dimmer

claudiovillani58, 2020-05-30:
Good evening guys I follow you assiduously on the shelly dimmer problem but I am a novice unlike you that I have seen you have a lot of knowledge. I would like to use shelly dimmer with classic arduino sketch or with souliss firmware. Can you help me? Do you already have a working tasmota sketch? If yes, could you share it?
Thanks

closed 2020-05-30 by close-issue-app[bot]

ascillato, 2020-05-30:
Tasmota compiles in arduino.

Please, check the docs at http://tasmota.com

The support for shelly dimmer is not complete, so, no available driver for now.

For Tasmota usage, please read the docs first and then if you need further assistance, please, address that to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

claudiovillani58, 2020-05-31:
Thank you

Best regards

Claudio