arendst/Tasmota #8586: Shelly 2.5 24v DC reboots on relay toggle

sidey79, 2020-05-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Toggeling the relay (via WebUI Button Toggle 1 / Toggle 2 or via MQTT) causes reboot

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 2.5
- [x] Tasmota binary firmware version number used: 20200530 16:01 GMT+2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
Command produces no output

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Reserve-S20","Tasmota2"],"Topic":"Reserve-S20","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:08:36","StartupUTC":"2020-05-30T21:03:59","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2020-05-30T22:02:30","SaveCount":22,"SaveAddress":"F6000"}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS2 = {"StatusFWR":{"Version":"8.3.1.2(1ddf65e-tasmota)","BuildDateTime":"2020-05-30T16:01:10","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"402/699"}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Blausee",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":25,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS5 = {"StatusNET":{"Hostname":"Reserve-S20-4366","IPAddress":"10.2.11.74","Gateway":"10.2.11.1","Subnetmask":"255.255.255.0","DNSServer":"10.2.11.1","Mac":"C8:2B:96:10:D1:0E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS6 = {"StatusMQT":{"MqttHost":"10.2.11.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_10D10E","MqttUser":"svcconnectmqtt","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS7 = {"StatusTIM":{"UTC":"2020-05-30T21:12:35","Local":"2020-05-30T22:12:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:52","Sunset":"20:43"}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS10 = {"StatusSNS":{"Time":"2020-05-30T22:12:35","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":62.4},"ENERGY":{"TotalStartTime":"2020-05-30T22:02:30","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}}
22:12:35 RSL: stat/zuhause/garten/Reserve-S20/STATUS11 = {"StatusSTS":{"Time":"2020-05-30T22:12:35","Uptime":"0T00:08:36","UptimeSec":516,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Blausee","BSSId":"44:4E:6D:3B:10:8C","Channel":13,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

Weblog 4 produces a re reboot loop, 

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Flash firmware 8.3.x 
2. Apply Wifi ettings and MQTT settings via tasmotizer
3. power shelly 2.5 via 24v DC powersupply
3. Apply template https://templates.blakadder.com/shelly_25.html
4. Toggle relay1 or relay 2 in webui
5. Relay toggles on/off during device reboot 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Toggeling a relay should not cause reboots as it is in 8.2.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-05-30:
Are your power source giving enough power? What is its rating?

sidey79, 2020-05-30:
24V / 1A

It isn't very much, but enough for the application i use it.
Reboot occures alsow without anything connected to the relays and with version 8.2.0 there are no problems at all.

ascillato, 2020-05-30:
What is the restart reason right after it reboots when toggling your relay?

sidey79, 2020-05-30:
@ascillato 
external system :(

`23:07:56 MQT: stat/zuhause/garten/Magnetventil/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:00:10","StartupUTC":"2020-05-30T22:07:46","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2020-03-23T22:36:01","SaveCount":60,"SaveAddress":"FB000"}}
`

sidey79, 2020-05-30:
> @ascillato
> external system :(
>
> `23:07:56 MQT: stat/zuhause/garten/Magnetventil/STATUS1 = [...]

Also trying to flash tasmota-minimal.bin when using the devel firmware the system resets at ~25%

ascillato, 2020-05-31:
So, the problem is your power source. Sorry.

Not software related.

sidey79, 2020-05-31:
I tried another 3A psu but this has the problem also on 8.2 :(

But may you are correct. Not sure, why this problem occures with my 1A psu not on 8.2

ascillato2, 2020-06-01:
So, sorry but if the power source is not the source of your problem, as it is reported a restart reason as external (no more power is applied to the esp8266 chip) that means that your hardware has a faulty power regulator. Sorry.

daxy01, 2020-09-13:
Would it be possible to re-open this issue? I have the same issue, only mine crashes when I toggle both outputs to On (even without anything connected to the ports. My log mentions "RestartReason":"External System" as well while running either 8.4 or 8.5.

I believe this could be 24DC related. The inputs do measure 24.4VDC steadily even when both ports are enabled and tasmota reboots. I worked around this by ensuring the outputs are always Off after a reboot, but when I saw this thread I've downgraded to 8.2 and all is fine! My guess is this is a problem with Shelly 2.5 powered by 24VDC in combination with Tasmota 8.3 and later.

The default logs are clear and show nothing interesting (as posted by OP already). Any further debugging I can do (without serial if possible)?

ascillato, 2020-09-13:
We can reopen this issue, but if this is a hardware issue where there is not enough power for the esp8266 to work, there is nothing we can do from the software side. Sorry.

daxy01, 2020-09-13:
@ascillato, Thanks. Usually I would agree with you and definitely question the PSU or Shelly, but the weird thing all is fine in 8.2 and confirmed to be broken in 8.3/8.4/8.5. Consistently reproducible. Something did change between 8.2 and 8.3 which I believe introduced this issue.

I'm new to Tasmota so need guidance on troubleshooting this further. I did enable syslog in debug, but that seems to crash my Shelly 😀

daxy01, 2020-09-13:
I've reproduced the issue while running 8.3.1 and Web log level set to 4 - More Debug (which crashes the shelly and reboots it) and also while running 8.2.0 (succesfully) with the same debug logging. Hope this helps!

[Shelly-2.5-24VDC-issue.txt](https://github.com/arendst/Tasmota/files/5215002/Shelly-2.5-24VDC-issue.txt)

daxy01, 2020-09-15:
@ascillato, is there anything else I can do to collect at this moment? I'll put the Shelly and pumps connected into production soon which means it will be more difficult to test (not great wifi coverage yet and can't just start irrigation 😀). Thanks!

ascillato, 2020-09-15:
> is there anything else I can do to collect at this moment?

No, You have provided all the logs.

In the logs it states that the restart reason is external system. That message is managed by the esp8266 at hardware level, not Tasmota, so, for the chip itself it detects that the reset button has been pressed. That can't be executed from the software side. The software-resets have different messages. 

So, indeed is a weird issue. I have several shelly devices and I couldn't reproduce this. Sorry.

It is weird that in one version works fine and in another it get the reset button pressed.

Reopening issue to see if someone in the community has any other idea to test.

daxy01, 2020-09-15:
Thanks.
For completeness, I do have another Shelly 2.5 running on 220V AC which is just running fine. Hence my gut-feeling telling me this is 24V DC related.

I've dug a bit in the differences committed into 8.3.0 and noticed quite a number of commits on ESP32.
Specifically this commit might be of interest: https://github.com/arendst/Tasmota/commit/b70ca01231b9fc54d5ab23d458f3e124ce78acad#diff-9f49e1a469607e0614e5786eaa90f2a4
It is a workaround to disable Brownout detection, however most of it seems to be removed in this commit: https://github.com/arendst/Tasmota/commit/c37dde93c244d7e4e86189fde13d4c56f34863b1#diff-f86fe7273c788cc62fd806c2e340dd01

What is interesting is that one specific part (file tasmota/tasmota.ino lines 201 to 206) hasn't been removed and is still in the code to-date. I have no idea if this could be causing, but brownout detection while enabling 2 ports on 24V seems a good lead to me 😀. It might be worth digging into this (and other commits as there is also QuickPowerCycle added for example).

sidey79, 2020-09-15:
Same by me:
230 V Shelly works ans 24vDC Shelly has this problem.

I will flash stock to the shelly when weather will be bad.

daxy01, 2020-09-20:
I've been playing around a bit with building images and using different core/SDKs but no success.
For example, I've build 8.3.1 with espressif8266@2.3.3 for example to match 8.2.0 but no success, Shelly still crashes.

It was a long-shot as I'm extremely new to all of this and don't know what's responsible and used for what. I figured this would be easier than going through 7000+ commits in tasmota itself 😀 

Any other help would be highly appreciated as I'm sort of lost right now. Thanks!

Jason2866, 2020-10-11:
Have you tried latest release version? There was a fix for a unitialized GPIO pin.
This did result in needing more current as in earlier Tasmota versions.
Maybe the 24V power supply is two weak to handle this additional current.
See #7991

daxy01, 2020-10-11:
Excellent! This seems to work indeed for my Shelly 2.5 on DC. It now remains working when both O1 and O2 are enabled. \o/
@sidey79 Are you also able to confirm if v8.5.1 works for you?

Jason2866, 2020-10-11:
Great. Conclusion. The 24 DC power supply is designed bleeding edge...

closed 2020-10-22 by ascillato2

sidey79, 2020-11-01:
@daxy01 

Did a Upgrade from 8.2.0 to 8.5.0 and then to 8.5.1 (http://ota.tasmota.com/tasmota/release/tasmota.bin.gz)

The problem persists, shelly is rebooting after i switch both relays on, without any load on them .

Did you change anything else after the Upgrade?
I'm using this template
`{"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}`

daxy01, 2020-11-02:
I am using this template:
`{"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}`
My power supply is a 60W 2,5A 24DC LED Power Supply (waterproof).