arendst/Tasmota #8587: Chirp Moisture and Darkness values overlapping

thucar, 2020-05-31:
### PROBLEM DESCRIPTION
Chirp! moisture sensor uses Moisture variable for both, Moisture and Darkness

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R2, Sonoff 4CH Pro R2
- [x] Tasmota binary firmware version number used: 8.3.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Platformio
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  08:07:02 MQT: stat/greenhouse_mini/RESULT = {"NAME":"HW-655 PZEM","GPIO":[0,0,0,0,6,5,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
08:07:03 MQT: stat/greenhouse_mini/RESULT = {"Module":{"39":"Sonoff Dual R2"}}
08:07:03 MQT: stat/greenhouse_mini/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"6":"I2C SDA"},"GPIO2":{"0":"None"},"GPIO3":{"5":"I2C SCL"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"17":"Button1"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  08:07:47 MQT: stat/greenhouse_mini/STATUS = {"Status":{"Module":39,"DeviceName":"Greenhouse Mini","FriendlyName":["Tasmota","Tasmota2"],"Topic":"greenhouse_mini","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:07:47 MQT: stat/greenhouse_mini/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Power On","Uptime":"0T00:13:40","StartupUTC":"2020-05-31T04:54:07","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2020-05-19T05:39:21","SaveCount":74,"SaveAddress":"F4000"}}
08:07:47 MQT: stat/greenhouse_mini/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-31T04:57:00","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"390/699"}}
08:07:47 MQT: stat/greenhouse_mini/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Nutiseadmed",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
08:07:47 MQT: stat/greenhouse_mini/STATUS4 = {"StatusMEM":{"ProgramSize":604,"Free":396,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","87DAE797","043E83A1","000002CD","210013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,8,48"}}
08:07:47 MQT: stat/greenhouse_mini/STATUS5 = {"StatusNET":{"Hostname":"greenhouse_mini-6445","IPAddress":"10.10.10.25","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"60:01:94:AD:B9:2D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:07:47 MQT: stat/greenhouse_mini/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ADB92D","MqttUser":"Homey","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:07:47 MQT: stat/greenhouse_mini/STATUS7 = {"StatusTIM":{"UTC":"2020-05-31T05:07:47","Local":"2020-05-31T08:07:47","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+03:00","Sunrise":"06:51","Sunset":"22:44"}}
08:07:47 MQT: stat/greenhouse_mini/STATUS10 = {"StatusSNS":{"Time":"2020-05-31T08:07:47","CHIRP0":{"Moisture":247,"Temperature":25.3,"Darkness":0},"TempUnit":"C"}}
08:07:47 MQT: stat/greenhouse_mini/STATUS11 = {"StatusSTS":{"Time":"2020-05-31T08:07:47","Uptime":"0T00:13:40","UptimeSec":820,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":32,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"","Channel":1,"RSSI":100,"Signal":-11,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Build FW with Chirp enabled
Connect Chirp! moisture sensor
Place Chirp in a dark environment
Configure module and observe sensor output and note that Darkness is always 0% while Moisture value alternates between normal (200 - 500) and high (+60000) value every few minutes.

### EXPECTED BEHAVIOUR
Moisture and Darkness values are treated, stored and output separately

### SCREENSHOTS
[image]

thucar, 2020-05-31:
Oh wow, just realized the issue only occurs when I have the web interface open in a browser.
Red arrows indicate when I closed the browser window. Blue arrow indicates when I reopened it.
[image]

Staars, 2020-06-02:
Hi,
are you able to test this behaviour with an older version too?
Does this error still only occur with an open web browser (in continuous testing)?

I will try to test this myself ASAP (after finding my Chirp again).

thucar, 2020-06-10:
Just tested with clean 8.2.0 and seeing the exact same behavior. If web interface is open, the darkness value gets recorded as moisture.

TemoOoOo, 2020-06-10:
Yez im tamim Tayfour please let me know if you have any other questions مُرسل من هاتف Huawei الخاص بي-------- الرسالة الأصلية --------الموضوع: Re: [arendst/Tasmota] Chirp Moisture and Darkness values overlapping (#8587)من: thucar إلى: arendst/Tasmota نسخة إلى: Subscribed 
Just tested with clean 8.2.0 and seeing the exact same behavior. If web interface is open, the darkness value gets recorded as moisture.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arendst/Tasmota/issues/8587#issuecomment-641675939",
"url": "https://github.com/arendst/Tasmota/issues/8587#issuecomment-641675939",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Staars, 2020-06-11:
I hope this PR fixes it. The current driver was definitely broken.

https://github.com/arendst/Tasmota/pull/8661

ascillato2, 2020-06-19:
Closing as the PR has been merged. Thanks.

closed 2020-06-19 by ascillato2