arendst/Tasmota #8591: after a restart/powercycle rule sets switched off, with rule2 crashing system

gadget78, 2020-05-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
after a restart/power-cycle rule sets are switched off, 
and if you `rule2` from console in any way, that will cause a restart/reboot

so then i have to `rule2 delete` and reinstate rule set, (pasting them back in via web console
with all the rules back, all those rules work fine...
no problems atall,
but AFTER a restart/powercycle, rules are switch off again, and `rule2` or `rule2 1` invokes a crash/restart again

HAVE found a workaround... 
just split up rules, and span over 3 rule sets ... 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [yep ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ yep] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [no ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ yep, spoke to sfromis, and Blakadder] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [[MagicHome device](https://templates.blakadder.com/MagicHome_ESP-IR-B-v2_3.html) ] Device used (e.g., Sonoff Basic): _____
- [newest Version 8.3.1(tasmota)-2_7_1 ] Tasmota binary firmware version number used: _____
  - [yep] Pre-compiled
  - [nope] Self-compiled
    - [ ] IDE / Compiler used: _____
- [tasmotizer,and tried OTA too] Flashing tools used: tasmotizer, via CH314a
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:25:35 CMD: Backlog Template; Module; GPIO 255
23:25:35 SRC: WebConsole from 192.168.178.78
23:25:35 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
23:25:35 SRC: Backlog
23:25:35 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
23:25:35 RSL: RESULT = {"NAME":"MagicHome2.3IR","GPIO":[17,255,52,255,51,38,0,0,37,39,255,40,255],"FLAG":0,"BASE":34}
23:25:35 SRC: Backlog
23:25:35 CMD: Group 0, Index 1, Command "MODULE", Data ""
23:25:35 RSL: RESULT = {"Module":{"0":"MagicHome2.3IR"}}
23:25:35 SRC: Backlog
23:25:35 CMD: Group 0, Index 1, Command "GPIO", Data "255"
23:25:35 RSL: RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"52":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"51":"IRrecv"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"37":"PWM1"},"GPIO13":{"39":"PWM3"},"GPIO14":{"0":"None"},"GPIO15":{"40":"PWM4"},"GPIO16":{"0":"None"}}
23:25:35 UPP: Hue 3 response packets sent to 192.168.178.78:54086

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
this is of course from a working environment BEFORE the reboot... if i try this AFTER a reboot, it just makes it reboot again, so no output can be captured... just the log on restart, see below

23:28:25 CMD: Backlog Rule1; Rule2; Rule3
23:28:25 SRC: WebConsole from 192.168.178.78
23:28:25 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
23:28:26 SRC: Backlog
23:28:26 CMD: Group 0, Index 1, Command "RULE", Data ""
23:28:26 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":628,"Free":283,"Rules":"on IrReceived#Data=0x00FFB04F DO backlog power 1; power0 1 endon  on IrReceived#Data=0x00FFF807 DO power0 0 endon  on IrReceived#Data=0xFFFFFFFFFFFFFFFF DO backlog power2 1; white 100 endon on IrReceived#Data=0x00FF906F DO dimmer + endon  on IrReceived#Data=0x00FFB847 DO dimmer - endon  on IrReceived#Data=0x00FFB24D DO backlog blinkcount 0; blinktime 5; power 3 endon  on IrReceived#Data=0x00FF00FF DO backlog color 12; white 100; blinkcount 0; blinktime 2; power1 3; power2 3 endon  on IrReceived#Data=0x00FF5..."}
23:28:26 SRC: Backlog
23:28:26 CMD: Group 0, Index 2, Command "RULE", Data ""
23:28:26 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":802,"Free":278,"Rules":"on IrReceived#Data=0x00FF9867 DO color 1 endon on IrReceived#Data=0x00FFE817 DO color 4 endon on IrReceived#Data=0x00FF02FD DO color 7 endon on IrReceived#Data=0x00FF50AF DO hsbcolor 40,100,100 endon on IrReceived#Data=0x00FF38C7 DO color 10 endon on IrReceived#Data=0x00FFD827 DO color 2 endon on IrReceived#Data=0x00FF48B7 DO color 5 endon on IrReceived#Data=0x00FF32CD DO color 8 endon on IrReceived#Data=0x00FF7887 DO hsbcolor 145,100,47 endon on IrReceived#Data=0x00FF28D7 DO hsbcolor 160,100,50 endon on Ir..."}
23:28:26 SRC: Backlog
23:28:26 CMD: Group 0, Index 3, Command "RULE", Data ""
23:28:26 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":6,"Free":505,"Rules":"delete"}
23:28:30 UDP: Packet (125)

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

23:29:59 CMD: Group 0, Index 1, Command "STATUS", Data "0"
[... 67 lines cut ...]
on IrReceived#Data=0x00FF02FD DO color 7 endon
on IrReceived#Data=0x00FF50AF DO hsbcolor 40,100,100 endon
on IrReceived#Data=0x00FF38C7 DO color 10 endon
on IrReceived#Data=0x00FFD827 DO color 2 endon
on IrReceived#Data=0x00FF48B7 DO color 5 endon
on IrReceived#Data=0x00FF32CD DO color 8 endon
on IrReceived#Data=0x00FF7887 DO hsbcolor 145,100,47 endon
on IrReceived#Data=0x00FF28D7 DO hsbcolor 160,100,50 endon
on IrReceived#Data=0x00FF8877 DO color 3 endon
on IrReceived#Data=0x00FF6897 DO color 6 endon
on IrReceived#Data=0x00FF20DF DO color 9 endon
on IrReceived#Data=0x00FF708F DO hsbcolor 290,100,50 endon
on IrReceived#Data=0x00FFF00F DO color 11 endon
on IrReceived#Data=0x00FFA857 DO color 12 endon
```
but if its split in half, and you put half into rule3, then this will give no issues... 
Also FYI, if the newlines are replaced with spaces, also does EXACTLY the same... 

so using a character count, Rule1 is 609 characters
FULL Rule2 is 797 characters, but if reduced to say 543 character and move the other commands to rules3 alls good ... 

hope this is enough info to get some idea of fault, or this help others... 
or even the IR codes help !! (i will submit them separately once we know its not the codes at fault !
 
**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-06-01:
Please try the latest Development branch (8.3.1.2). There were some bugs in special cases for the compression algorithm used by long rules. The fact that splitting the long Rule into two rule sets seems to point to this possibility.

gadget78, 2020-06-01:
yep that solves that issue :) didnt realise the development builds were compiled and easily available..
 [http://thehackbox.org/tasmota/](http://thehackbox.org/tasmota/)
it upgraded fine with OTA in webgui to via the .gz version (just pasted link in)
hopefully this post at least helps one person !...

closed 2020-06-01 by gadget78