arendst/Tasmota #8594: Add MCP9808 i2c temperature sensor support

m-hume, 2020-06-01:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
MCP9808 not currently supported
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
MCP9808 support added
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Use of different temperature sensors
**Additional context**  
_Add any other context or screenshots about the feature request here._
http://ww1.microchip.com/downloads/en/DeviceDoc/25095A.pdf
These little sensors are cheep, accurate and you can run up to 8 of them on one i2c bus
looking at others pull requests for adding i2c sensors there seems to quite a bit of code necessary to add one. 
How difficult is it to add a new sensor? 
Thanks
**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-06-01:
New sensors are usually added when someone interested volunteers to create a driver and submit a PR.

m-hume, 2020-06-01:
That's kind of what i was expecting. This is not really my in my area of expertise but i'll have a play with gitpod.io and see if i can come up with something

device111, 2020-06-01:
@m-hume 
please test this bin file and test what happens..:-) 

It should support 8 sensors.

[tasmota.zip](https://github.com/arendst/Tasmota/files/4713324/tasmota.zip)

m-hume, 2020-06-01:
@device111 thank you!
Only tried with 2 sensors so far but it seems to work as described 

[image]

Is the bin missing anything or is it the standard build just with this sensor added?
I'm intrigued to see what was involved to get this working if you get chance to submit a pr
Thanks again for your time
m-hume

device111, 2020-06-03:
@m-hume 
the PR is now in dev. Can you test it if all works? the resolution is set to 0.125 Celsius.
thanks!

m-hume, 2020-06-03:
@device111 
I ran into a small issue regarding missing sensors
please see https://github.com/arendst/Tasmota/pull/8611

ascillato2, 2020-06-03:
Closing this issue as the driver is merged.

Thanks :+1:

closed 2020-06-03 by ascillato2

m-hume, 2020-06-03:
@device111 
I don't want to sound ungrateful at all. Without your work getting the sensor implemented i'd still be scratching my head. Thank you for your hard work. It's appreciated :+1:

device111, 2020-06-03:
@m-hume 
greatest thanks go to Theo @arendst , who keeps the Show on running and always have a look above the big code in Tasmota.
Sometimes i think, he is never sleep...;-)