arendst/Tasmota #8597: Ghost Switching on Sesoo WIFI-EU-SK3-02

Hustensaft92, 2020-06-01:
# PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The Light switch switch automatically on off. 
Two light switches, same issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
Sesoo WIFI-EU-SK3-02     {"NAME":"Sesoo SK3-02","GPIO":[0,0,57,0,22,0,0,0,17,21,18,0,52],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
no rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:42:28 CMD: status0
23:42:28 MQT: stat/tasmota_AB92F2/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_AB92F2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
23:42:50 CMD: status 0
23:42:50 MQT: stat/tasmota_AB92F2/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_AB92F2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"https://github.com/arendst/Tasmota/releases/download/v7.2.0/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T22:12:19","StartupUTC":"2020-06-01T00:30:31","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":230,"SaveAddress":"FA000"}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS2 = {"StatusFWR":{"Version":"7.2.0(tasmota)","BuildDateTime":"2019.12.22 11:45:04","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"356/1151"}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 7580 YB",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008068","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS5 = {"StatusNET":{"Hostname":"LichtschalterSZ","IPAddress":"192.168.178.211","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"A4:CF:12:AB:92:F2","Webserver":2,"WifiConfig":5}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.40","MqttPort":1883,"MqttClientMask":"LichtschalterSZ","MqttClient":"LichtschalterSZ","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS7 = {"StatusTIM":{"UTC":"Mon Jun 01 22:42:50 2020","Local":"Mon Jun 01 23:42:50 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"04:50","Sunset":"20:45"}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS10 = {"StatusSNS":{"Time":"2020-06-01T23:42:50"}}
23:42:50 MQT: stat/tasmota_AB92F2/STATUS11 = {"StatusSTS":{"Time":"2020-06-01T23:42:50","Uptime":"0T22:12:19","UptimeSec":79939,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box 7580 YB","BSSId":"E0:28:6D:48:7A:68","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
23:59:15 APP: Knopf1 Mehrfachdruck 1
23:59:16 SRC: Button
23:59:16 MQT: stat/tasmota_AB92F2/RESULT = {"POWER1":"ON"}
23:59:16 MQT: stat/tasmota_AB92F2/POWER1 = ON (beibehalten)
23:59:20 HTP: Konsole

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I test it with the actually version and I downgraded it to. Nothing change.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-02:
`Version":"7.2.0(tasmota)`

Please update to latest. Too many changes from that version to actual.

Hustensaft92, 2020-06-02:
Same Issue with 8.3.1 

01:09:16 CMD: status 0
01:09:16 SRC: WebConsole from 192.168.178.34
01:09:16 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
01:09:16 MQT: stat/tasmota_AB92F2/STATUS = {"Status":{"Module":0,"DeviceName":"LichtschalterSZ","FriendlyName":["LichtschalterSZ","Tasmota2"],"Topic":"tasmota_AB92F2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
01:09:16 MQT: stat/tasmota_AB92F2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/arendst/Tasmota/releases/download/v7.2.0/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:13","StartupUTC":"2020-06-02T00:09:03","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2020-05-30T17:50:45","SaveCount":238,"SaveAddress":"FA000"}}
01:09:16 MQT: stat/tasmota_AB92F2/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020.05.18 15:45:12","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"431/699"}}
01:09:16 MQT: stat/tasmota_AB92F2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 7580 YB",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008028","2805C8000100060000005A00000000000000","00000000","00000000"]}}
01:09:16 MQT: stat/tasmota_AB92F2/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000407","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
01:09:17 MQT: stat/tasmota_AB92F2/STATUS5 = {"StatusNET":{"Hostname":"LichtschalterSZ","IPAddress":"192.168.178.211","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"A4:CF:12:AB:92:F2","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
01:09:17 MQT: stat/tasmota_AB92F2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.40","MqttPort":1883,"MqttClientMask":"LichtschalterSZ","MqttClient":"LichtschalterSZ","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
01:09:17 MQT: stat/tasmota_AB92F2/STATUS7 = {"StatusTIM":{"UTC":"2020-06-02T00:09:17","Local":"2020-06-02T01:09:17","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:50","Sunset":"20:45"}}
01:09:17 MQT: stat/tasmota_AB92F2/STATUS10 = {"StatusSNS":{"Time":"2020-06-02T01:09:17"}}
01:09:17 MQT: stat/tasmota_AB92F2/STATUS11 = {"StatusSTS":{"Time":"2020-06-02T01:09:17","Uptime":"0T00:00:14","UptimeSec":14,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box 7580 YB","BSSId":"E0:28:6D:48:7A:68","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}

ADDITIONAL CONTEXT

Add any other context about the problem here.
I test it with the actually version and I downgraded it to. Nothing change.

curzon01, 2020-06-02:
Looks like classic retained MQTT msg problem, see [Retained MQTT Messages](https://tasmota.github.io/docs/MQTT/#retained-mqtt-messages) and related links and why its the main reason of ghost issues

Hustensaft92, 2020-06-02:
I take a look at this video. I used this commands: 
#Tasmota Console Command: SwitchRetain Off
#Tasmota Console Command: ButtonRetain On
#Tasmota Console Command: ButtonRetain Off
#Tasmota Console Command: PowerRetain On

but I don't know where I can check this: 

  - platform: mqtt
    name: "Tasmota Switch"
    state_topic: "stat/device_name/POWER"
    command_topic: "cmnd/device_name/POWER"
    availability_topic: "tele/device_name/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    payload_on: "ON"
    payload_off: "OFF"
    retain: false

can someone help please :-)

blakadder, 2020-06-03:
You need to read the link given and clear retained messages from your broker

Hustensaft92, 2020-06-04:
First of all I want to say Thanks for Supporting! 

Now I use iobroker with the sonoff application. I can’t find my devices with the Tasmota Device Manager. I follow the instructions but can’t find what I made wrong 😩

curzon01, 2020-06-04:
Seems issue is not an issue, your problems are more common. Keep in mind nobody is able to help even if they want to, when your informations consists of general places - _I follow the instructions_ is an example.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

<hr>

#### Support Information
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://tasmota.github.io/docs/FAQ/) for common questions/answers and [links](https://tasmota.github.io/docs/FAQ/#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

Hustensaft92, 2020-06-04:
https://discordapp.com/channels/479389167382691863/490244847568158751/718230705112481812

I hope anyone can try to help :-)

closed 2020-06-09 by ascillato2