arendst/Tasmota #8599: option to disable MQTT_CLEAN_SESSION (for deep_sleep and low power TuyaMCU)

bartowl, 2020-06-02:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
no

**Describe the solution you'd like**  
I would like to have an option, that tasmota do not connect always with  MQTT_CLEAN_SESSION. This would allow to queue commands on mqtt brokwer for battery powered devices, that are permanently off. Now, all commands send to cmnd/dev/* topic get ignored when the device boots. Without MQTT_CLEAN_SESSION they would be queued on the broker, and delivered when the device boots up. This would allow for Example to triffer an OTA upgrade od change some parameters wile the device is powered down. It would be nice to have sucha aswitch per setoption to be able to change this behaviour.

**Describe alternatives you've considered**  
Currently the only option is to wait for LWT to change to online and trigger the commands upon this event. This requires queueing the commands on the home automation side. Of course if such a command needs to be removed it is way easier, than to remove mqtt queued message for a particular client, but this requires extra coding here.

**Additional context**  
could be primarily usefull for battery powered devices, where ESP is powered down for most of the time and wakes up only on event or to report battery status.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-06-02:
Compile your own and change define MQTT_CLEAN_SESSION form 1 to 0

bartowl, 2020-06-02:
This is another way to do things... But then I have to build every single new version, maintain all those files etc... BTW I think this ist the best software for ESP today and am really thankful for it!
If you do not consider this as an worth-to-implement option, then well... I already have a workaround in place as described above.
Just thought it might make sense to have this switchable.

blakadder, 2020-06-02:
i could add the flag to the battery build in tasmota/binaries

arendst, 2020-06-02:
Yes pls do.

closed 2020-06-02 by ascillato2