arendst/Tasmota #8602: sleep 0-2 kills wireless connection

MGKQB, 2020-06-02:
### PROBLEM DESCRIPTION
I changed the sleep to 0 since that is reocmmended for use with KNX. 
Right after that my test device (a sonoff basic) looses the wireless connection. Neither sonoff nor wireless AP (Cisco 3700 running IOS) report anything useful. 
I connected a serial cable and chenaged the sleep back to 50, device connects back to wireless.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Basic
- [X] Tasmota binary firmware version number used: 8.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: NodeMCUflasher
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
15:58:37 CMD: Backlog Template; Module; GPIO 255
15:58:37 RSL: stat/tasmota_0F3050/RESULT = {„NAME“:“Generic“,“GPIO“:[255,255,255,255,255,255,255,255,255,255,255,255,255],“FLAG“:15,“BASE“:18}
15:58:38 RSL: stat/tasmota_0F3050/RESULT = {„Module“:{„1“:“Sonoff Basic“}}
15:58:38 RSL: stat/tasmota_0F3050/RESULT = {„GPIO0“:{„17“:“Button1“},“GPIO1“:{„0“:“None“},“GPIO2“:{„0“:“None“},“GPIO3“:{„0“:“None“},“GPIO4“:{„0“:“None“},“GPIO5“:{„0“:“None“},“GPIO9“:{„0“:“None“},“GPIO10“:{„0“:“None“},“GPIO12“:{„21“:“Relay1“},“GPIO13“:{„56“:“Led1i“},“GPIO14“:{„0“:“None“},“GPIO15“:{„0“:“None“},“GPIO16“:{„0“:“None“}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
no rules in use (yet)

```
- [X] Provide the output of this command: `Status 0`:

while connected:
```
5:59:25 CMD: status 0
15:59:25 RSL: stat/tasmota_0F3050/STATUS = {„Status“:{„Module“:1,“DeviceName“:“Tasmota“,“FriendlyName“:[„Tasmota“],“Topic“:“tasmota_0F3050“,“ButtonTopic“:“0“,“Power“:0,“PowerOnState“:3,“LedState“:1,“LedMask“:“FFFF“,“SaveData“:1,“SaveState“:1,“SwitchTopic“:“0“,“SwitchMode“:[0,0,0,0,0,0,0,0],“ButtonRetain“:0,“SwitchRetain“:0,“SensorRetain“:0,“PowerRetain“:0}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS1 = {„StatusPRM“:{„Baudrate“:115200,“SerialConfig“:“8N1“,“GroupTopic“:“tasmotas“,“OtaUrl“:“http://thehackbox.org/tasmota/release/tasmota.bin“,“RestartReason“:“Software/System restart“,“Uptime“:“0T01:21:47“,“StartupUTC“:“2020-06-02T12:37:38“,“Sleep“:50,“CfgHolder“:4617,“BootCount“:1,“BCResetTime“:“2020-06-02T13:37:46“,“SaveCount“:47,“SaveAddress“:“F5000“}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS2 = {„StatusFWR“:{„Version“:“8.3.1(knx)“,“BuildDateTime“:“2020-05-18T15:40:57“,“Boot“:31,“Core“:“2_7_1“,“SDK“:“2.2.2-dev(38a443e)“,“Hardware“:“ESP8266EX“,“CR“:“337/699“}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS3 = {„StatusLOG“:{„SerialLog“:2,“WebLog“:2,“MqttLog“:0,“SysLog“:0,“LogHost“:““,“LogPort“:514,“SSId“:[„MnM-Haus“,““],“TelePeriod“:300,“Resolution“:“558180C0“,“SetOption“:[„00008009“,“2805C8000100060000005A00000000000000“,“00000000“,“00000000“]}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS4 = {„StatusMEM“:{„ProgramSize“:583,“Free“:420,“Heap“:22,“ProgramFlashSize“:1024,“FlashSize“:1024,“FlashChipId“:“146085“,“FlashMode“:3,“Features“:[„00000809“,“9FDAE597“,“042683B0“,“000000CD“,“010013C0“,“8000F181“,“00000024“],“Drivers“:“1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,27,30,35,37“,“Sensors“:“1,2,3,4,5,6“}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS5 = {„StatusNET“:{„Hostname“:“tasmota_0F3050-4176“,“IPAddress“:“192.168.21.33“,“Gateway“:“192.168.21.1“,“Subnetmask“:“255.255.255.0“,“DNSServer“:“192.168.21.1“,“Mac“:“BC:DD:C2:0F:30:50“,“Webserver“:2,“WifiConfig“:4,“WifiPower“:17.0}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS6 = {„StatusMQT“:{„MqttHost“:““,“MqttPort“:1883,“MqttClientMask“:“DVES_%06X“,“MqttClient“:“DVES_0F3050“,“MqttUser“:“DVES_USER“,“MqttCount“:0,“MAX_PACKET_SIZE“:1200,“KEEPALIVE“:30}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS7 = {„StatusTIM“:{„UTC“:“2020-06-02T13:59:25“,“Local“:“2020-06-02T15:59:25“,“StartDST“:“2020-03-29T02:00:00“,“EndDST“:“2020-10-25T03:00:00“,“Timezone“:99,“Sunrise“:“04:54“,“Sunset“:“21:40“}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS10 = {„StatusSNS“:{„Time“:“2020-06-02T15:59:25“}}
15:59:25 RSL: stat/tasmota_0F3050/STATUS11 = {„StatusSTS“:{„Time“:“2020-06-02T15:59:25“,“Uptime“:“0T01:21:47“,“UptimeSec“:4907,“Heap“:22,“SleepMode“:“Dynamic“,“Sleep“:50,“LoadAvg“:19,“MqttCount“:0,“POWER“:“OFF“,“Wifi“:{„AP“:1,“SSId“:“MnM-Haus“,“BSSId“:“F0:7F:06:83:B1:52“,“Channel“:13,“RSSI“:24,“Signal“:-88,“LinkCount“:8,“Downtime“:“0T00:05:00“}}}
```

disconnected with sleep 0
```
16:01:12 CMD: 
status 0
16:01:15 CMD: status 0
16:01:15 RSL: stat/tasmota_0F3050/STATUS = {„Status“:{„Module“:1,“DeviceName“:“Tasmota“,“FriendlyName“:[„Tasmota“],“Topic“:“tasmota_0F3050“,“ButtonTopic“:“0“,“Power“:0,“PowerOnState“:3,“LedState“:1,“LedMask“:“FFFF“,“SaveData“:1,“SaveState“:1,“SwitchTopic“:“0“,“SwitchMode“:[0,0,0,0,0,0,0,0],“ButtonRetain“:0,“SwitchRetain“:0,“SensorRetain“:0,“PowerRetain“:0}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS1 = {„StatusPRM“:{„Baudrate“:115200,“SerialConfig“:“8N1“,“GroupTopic“:“tasmotas“,“OtaUrl“:“http://thehackbox.org/tasmota/release/tasmota.bin“,“RestartReason“:“Software/System restart“,“Uptime“:“0T01:23:37“,“StartupUTC“:“2020-06-02T12:37:38“,“Sleep“:0,“CfgHolder“:4617,“BootCount“:1,“BCResetTime“:“2020-06-02T13:37:46“,“SaveCount“:49,“SaveAddress“:“FB000“}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS2 = {„StatusFWR“:{„Version“:“8.3.1(knx)“,“BuildDateTime“:“2020-05-18T15:40:57“,“Boot“:31,“Core“:“2_7_1“,“SDK“:“2.2.2-dev(38a443e)“,“Hardware“:“ESP8266EX“,“CR“:“337/699“}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS3 = {„StatusLOG“:{„SerialLog“:2,“WebLog“:2,“MqttLog“:0,“SysLog“:0,“LogHost“:““,“LogPort“:514,“SSId“:[„MnM-Haus“,““],“TelePeriod“:300,“Resolution“:“558180C0“,“SetOption“:[„00008009“,“2805C8000100060000005A00000000000000“,“00000000“,“00000000“]}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS4 = {„StatusMEM“:{„ProgramSize“:583,“Free“:420,“Heap“:23,“ProgramFlashSize“:1024,“FlashSize“:1024,“FlashChipId“:“146085“,“FlashMode“:3,“Features“:[„00000809“,“9FDAE597“,“042683B0“,“000000CD“,“010013C0“,“8000F181“,“00000024“],“Drivers“:“1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,27,30,35,37“,“Sensors“:“1,2,3,4,5,6“}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS5 = {„StatusNET“:{„Hostname“:“tasmota_0F3050-4176“,“IPAddress“:“(IP unset)“,“Gateway“:“192.168.21.1“,“Subnetmask“:“255.255.255.0“,“DNSServer“:“192.168.21.1“,“Mac“:“BC:DD:C2:0F:30:50“,“Webserver“:2,“WifiConfig“:4,“WifiPower“:17.0}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS6 = {„StatusMQT“:{„MqttHost“:““,“MqttPort“:1883,“MqttClientMask“:“DVES_%06X“,“MqttClient“:“DVES_0F3050“,“MqttUser“:“DVES_USER“,“MqttCount“:0,“MAX_PACKET_SIZE“:1200,“KEEPALIVE“:30}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS7 = {„StatusTIM“:{„UTC“:“2020-06-02T14:01:15“,“Local“:“2020-06-02T16:01:15“,“StartDST“:“2020-03-29T02:00:00“,“EndDST“:“2020-10-25T03:00:00“,“Timezone“:99,“Sunrise“:“04:54“,“Sunset“:“21:40“}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS10 = {„StatusSNS“:{„Time“:“2020-06-02T16:01:15“}}
16:01:15 RSL: stat/tasmota_0F3050/STATUS11 = {„StatusSTS“:{„Time“:“2020-06-02T16:01:15“,“Uptime“:“0T01:23:37“,“UptimeSec“:5017,“Heap“:23,“SleepMode“:“Dynamic“,“Sleep“:0,“LoadAvg“:135,“MqttCount“:0,“POWER“:“OFF“,“Wifi“:{„AP“:1,“SSId“:“MnM-Haus“,“BSSId“:“F0:7F:06:83:B1:52“,“Channel“:13,“RSSI“:100,“Signal“:31,“LinkCount“:8,“Downtime“:“0T00:05:00“}}}
```

- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_

[... 216 lines cut ...]
16:27:04 WIF: Attempting connection…
16:27:05 WIF: Checking connection…
16:27:05 WIF: Connected
16:27:06 CFG: Saved to flash at F9, Count 91, Bytes 4096

```

### TO REPRODUCE
sleep between 0-2 breaks the wireless connection, sleep >2 restores it. LED starts blinking about 5 seconds after the sleep command.

### EXPECTED BEHAVIOUR
wireless should work with any sleep value, sleep 0,1,2 don't work, sleep >2 work reliably.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
this messages show up on the AP:

Jun  2 16:12:21 CEST: %DOT11-4-MAXRETRIES: Packet to client bcdd.c20f.3050 reached max retries, removing the client

and ONCE i got.
Jun  2 16:13:27 CEST: %DOT11-7-AUTH_FAILED: Station bcdd.c20f.3050 Authentication failed

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-06-02:
Hi,

Some other issue going on with your device. As Sonoff Basics are known to have a weak power supply, the energy used while sleep is 0, could make it fail to provide enough for wifi TX.

Have you tried to restart your router? Not all routers supports that a device will change itself from a sleeping behaviour to not.

At this moment the recommendation for sleep 0 is not needed anymore for KNX. That was for older arduino cores. I will delete that from the docs.

So, please, try to reset your router and try again, if still fails, your device can not be used without sleep. You can use another device if you want sleep 0.

But if sleep 0 was only because of KNX you can set sleep to the default 50 that is OK. You shouldn't have any issue. I have all my devices using KNX and with the default sleep. No issue at all.

MGKQB, 2020-06-02:
KNX doesn't require sleep 0 anymore? Great news, i'll close this then.

closed 2020-06-02 by MGKQB

s-hadinger, 2020-06-02:
This sleep 0 thing is a mystery. My Lohas bulbs also lose Wifi connectivity with Sleep 0, but it was reported by users that some device require Sleep 0 for better wifi connectivity.

When fading or light is on, the minimum sleep is defined in 
```
#define PWM_MAX_SLEEP          10                // Sleep will be lowered to this value when light is on, to avoid flickering
```

I use 25 rather than 10 in my bulbs.

ascillato, 2020-06-02:
If the power regulator of a device is weak, like some batches of sonoff basics, a `sleep 0` increases a lot the energy consumption on every wifi TX, making the device to fail. Sometimes reboots, watchdogs, exception 0 or 29, or just unable to connect to a wifi network. In this case is recommended the default (`sleep 50`), use Tasmota-lite.bin or use another device.

If the wifi router, where a Tasmota device is connecting, has issues with the ARP table, or it is unable to handle sleeping devices, your Tasmota can't connect, or disconnects and reconnects very often, or you are not able to reach the webUI after some time that it is connected. If you ping the device (the ARP table is refreshed on the router) and you can connect again to the webUI.
In this case, Sleep 0 can be a workaround to this issue but the solution is on the router's side, like in Mikrotik enabling Multicast-Helper=full
Another workaround to this ARP issue is enabling gratuitous ARP using `setoption41`

MGKQB, 2020-06-03:
I tested with other devices and you are right. No wireless issues there with sleep 0, so it’s a Sonoff Basic Problem, not tasmota.
Weird, I was actually powering the basic from USB and the voltage was fine. 

Since KNX doesn’t need sleep 0 anymore i set all devices back to the default sleep and have no issues with wireless! Thanks!