arendst/Tasmota #8607: Deep Sleep question/issue?

combatistor, 2020-06-03:
### PROBLEM DESCRIPTION
Deep Sleep issue, according to the documentation, the device should return to deep sleep (After a deep sleep wake) at the first TelePeriod. In my case my teleperiod is 300 (5 minutes) but the device returns imediately in deep sleep after a wake up. 
Actually one TelePeriod is trigger at the wakeup but according to #7298 it should wait for the 2nd TelePeriod, but it doesn't.
I also get this error message every time I trigger a deepSleepTime other than 0: 
DSL: Reset wrong settings wakeup: 0, slip 9785
I figured out the deep sleep is kinda unstable, waking up at random intervals, some times don't go back to deep sleep. It's strange behaviour.

By the way I connected GPIO16 to Reset according to the docs.

I was wondering if it's normal / I'm just confused by the documentation / thedocumentation is outdated?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: Tasmota 8.3.1 SENSORS
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:02:55 MQT: stat/wemosD1/balcony/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:02:56 MQT: stat/wemosD1/balcony/RESULT = {"Module":{"18":"Generic"}}
11:02:56 MQT: stat/wemosD1/balcony/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"5":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"6":"I2C SDA"},"GPIO4":{"9":"Switch1"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:04:01 MQT: stat/wemosD1/balcony/STATUS = {"Status":{"Module":18,"DeviceName":"Wemos-balcony","FriendlyName":["Wemos-balcony"],"Topic":"wemosD1/balcony","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-sensors.bin","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:04:05","StartupUTC":"2020-06-03T01:59:56","Sleep":50,"CfgHolder":4617,"BootCount":474,"BCResetTime":"2020-04-18T16:26:25","SaveCount":842,"SaveAddress":"F4000"}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS2 = {"StatusFWR":{"Version":"8.3.1(sensors)","BuildDateTime":"2020-05-18T15:41:26","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"408/699"}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UniFi-Bat-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000000060000005A00000000000000","00000000","00000000"]}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000809","8FDAA787","0415A3A4","B7FFBFCD","01DA9BC4","64367CC7","00000072"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67"}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS5 = {"StatusNET":{"Hostname":"wemosD1/balcony-4154","IPAddress":"theIPAddress","Gateway":"TheGateway","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"98:F4:AB:DA:B0:3A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.16","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DAB03A","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS7 = {"StatusTIM":{"UTC":"2020-06-03T02:04:01","Local":"2020-06-03T11:04:01","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+09:00","Sunrise":"12:49","Sunset":"04:46"}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS10 = {"StatusSNS":{"Time":"2020-06-03T11:04:01","Switch1":"OFF","BME280":{"Temperature":27.6,"Humidity":50.4,"DewPoint":16.4,"Pressure":1002.9},"PressureUnit":"hPa","TempUnit":"C"}}
11:04:01 MQT: stat/wemosD1/balcony/STATUS11 = {"StatusSTS":{"Time":"2020-06-03T11:04:01","Uptime":"0T00:04:05","UptimeSec":245,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"my-ssid","BSSId":"myBSSId","Channel":2,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

11:09:55 CMD: deepSleepTime 60
11:09:55 SRC: WebConsole from ********
11:09:55 CMD: Group 0, Index 1, Command "DEEPSLEEPTIME", Data "60"
11:09:55 MQT: stat/wemosD1/balcony/RESULT = {"DeepSleepTime":60}
11:09:56 CFG: Saved to flash at F7, Count 847, Bytes 4096
11:09:56 DSL: Reset wrong settings wakeup: 0, slip 9785
11:09:56 MQT: stat/wemosD1/balcony/RESULT = {"DeepSleep":{"Time":"2020-06-03T11:11:00","Epoch":1591150260}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
I expect the device to wait the TelePeriod interval before going back to deep sleep (As the documentation says)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-03:
As designed, see https://github.com/arendst/Tasmota/pull/7298

combatistor, 2020-06-03:
Oh! I'm sorry I haven't seen that it was for teleperiod from 10 to 300.
Another question is it normal to get the error message everytime? 
`Reset wrong settings wakeup: 0, slip -----`

Jason2866, 2020-06-03:
No it is exactly 10 or 300 only for teleperiod to go sleep asap.
The second question i cant answer. @stefanbode can you please answer? Thx

combatistor, 2020-06-03:
Ah okay, sorry for the misunderstanding,
It would be great indeed to add this to the documentation

closed 2020-06-09 by ascillato2