arendst/Tasmota #8609: Cannot configure CCT light correctly

grizewald, 2020-06-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I just bought some Deltaco CCT light bulbs, specifically the SH-LE14W and the SH-LE27W. Both have templates available on templates.blakadder.com.
I flashed one of the E14 bulbs with tuya-convert, updated it to the latest firmware and applied the correct template:

{"NAME":"SH-LE14W","GPIO":[0,0,0,0,0,140,0,0,38,0,37,142,141],"FLAG":0,"BASE":18}

The problem is that instead of creating a bulb with two sliders (one for colour temperature and one for the dimmer), it creates one with four sliders:

[image]

The two sliders that I expected control the bulb correctly - the dimmer slider varies the light intensity and the colour temperature slider adjusts the colour temperature from cold to warm. There are only two PWM channels specified, so according to the documentation, that should create a two channel device, not a four channel one.

I tried using other templates and changing the module, but none of my alternative configurations control the bulb correctly.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tuya-convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
07:30:10 CMD: Backlog Template; Module; GPIO 255
07:30:10 RSL: stat/tasmota_DE0A11/RESULT = {"NAME":"SH-LE14W","GPIO":[0,0,0,0,0,140,0,0,38,0,37,142,141],"FLAG":0,"BASE":18}
07:30:10 RSL: stat/tasmota_DE0A11/RESULT = {"Module":{"0":"SH-LE14W"}}
07:30:11 RSL: stat/tasmota_DE0A11/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"140":"SM16716 CLK"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"0":"None"},"GPIO14":{"37":"PWM1"},"GPIO15":{"142":"SM16716 PWR"},"GPIO16":{"141":"SM16716 DAT"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
07:30:46 CMD: status 0
07:30:46 RSL: stat/tasmota_DE0A11/STATUS = {"Status":{"Module":0,"DeviceName":"bedsidelamp","FriendlyName":["Bedside Lamp"],"Topic":"tasmota_DE0A11","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:14","StartupUTC":"2020-06-03T06:25:32","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-06-02T23:25:00","SaveCount":35,"SaveAddress":"F9000"}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"375/699"}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Cam24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS5 = {"StatusNET":{"Hostname":"tasmota_DE0A11-2577","IPAddress":"192.168.1.97","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:F4:32:DE:0A:11","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DE0A11","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS7 = {"StatusTIM":{"UTC":"2020-06-03T06:30:46","Local":"2020-06-03T07:30:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:49","Sunset":"20:46"}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS10 = {"StatusSNS":{"Time":"2020-06-03T07:30:46"}}
07:30:46 RSL: stat/tasmota_DE0A11/STATUS11 = {"StatusSTS":{"Time":"2020-06-03T07:30:46","Uptime":"0T00:05:14","UptimeSec":314,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":15,"Color":"0000002303","HSBColor":"0,0,0","White":15,"CT":177,"Channel":[0,0,0,14,1],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Cam24","BSSId":"3C:37:86:FD:9D:94","Channel":13,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Apply template.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A light with only the applicable controls should be created.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-03:
You have defined 2 PWM and one light controller SM16716. So there has to be more sliders than two
So the template is not the correct one for your bulb

blakadder, 2020-06-03:
Try removing the non pwm components from the template

grizewald, 2020-06-03:
Removing the lighting controller and just leaving the two PWM channels fixes the problem, thanks.

I'm just creating pull requests to the blkadder repo to correct the templates.

closed 2020-06-03 by grizewald